betaflight/betaflight #11487: LEDS not working properly on newer FURYF4OSD Versions, only Legacy Versions working correctly

thedvxl, 2022-03-24:
### Describe the bug

So my Problem is, that my LEDs only work on the Legacy Version of the FURYF4OSD for my Diatone Mamba F405 MK2. When I flash the other Versions (yes, I tried every version), only the first LED is permanently red, the third LED is flashing in seemingly random colors and the rest of the LEDs are completely off. But I really would like to use the newer Versions but still want to use the leds. And yes, my wiring is absolutely correct, I connected the led strip to 5V and GND and also to the BF LED Header since I am using a regular WS2812B LED Strip thats not connected to the motors in any way.

### To Reproduce

Buy a Mamba F405 MK2 FC and flash any FURYF4OSD firmware except the Legacy Versions, connect addressable LED Strip to 5V, GND and BF LED, make some settings in the LED_STRIP Tab and see that it will not working properly

### Expected behavior

Instead of seeing just two of my 6 LEDs light up, I expected the LEDS to light up the way I told them to do in the BF configurator. Not only in the FURYF4OSD Legacy Versions.

### Flight controller configuration

```
# diff

# version
# Betaflight / FURYF4OSD (FY4O) 4.1.1 Nov 15 2019 / 12:26:24 (1e5e3d369) MSP API: 1.42

# start the command batch
batch start

board_name FURYF4OSD

# feature
feature GPS
feature LED_STRIP

# map
map TAER1234

# serial
serial 0 0 115200 57600 0 115200
serial 2 2 115200 115200 0 115200
serial 5 64 115200 57600 0 115200

# led
led 0 0,0:ND:CTOW:3
led 1 1,0:ND:CTOW:4
led 2 2,0:ND:CTOW:5
led 3 3,0:ND:CTOW:7
led 4 4,0:ND:CTOW:9
led 5 5,0:ND:CTOW:10
led 6 6,0:ND:CTOW:11
led 7 7,0:ND:COW:12

# color
color 1 359,255,255

# mode_color
mode_color 3 0 2
mode_color 7 0 8

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 46 2 1700 2100 0 0
aux 3 15 3 1700 2100 0 0
aux 4 37 3 1300 1700 0 0

# master
set acc_calibration = -105,2,-139
set rc_smoothing_derivative_type = PT1
set serialrx_provider = CRSF
set dshot_idle_value = 500
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 5
set failsafe_procedure = GPS-RESCUE
set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 340
set current_meter = NONE
set ibata_scale = 100
set small_angle = 180
set gps_provider = UBLOX
set pid_process_denom = 1

profile 0

# profile 0
set p_yaw = 45
set angle_level_strength = 30

rateprofile 1

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: LED_STRIP
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: CAMERA_CONTROL
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: USB_DETECT
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D02: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: LED_STRIP
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 1
    CH2: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: LED_STRIP
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

Diatone Mamba F405 MK2

### Other components

Firmware: FURYF4OSD
Motors: Emax Eco II 2807 1700kv
ESC: Diatone F40 40A BlHeli_S Dshot600
RX: Crossfire Nano RX
VTX: AKK x2 Ultimate
Camera: Runcam Swift 2 RotorRiot Edition
GPS: TBS M8.2

### How are the different components wired up

BF LED to DIN
GND to GND
5V to 5V

spanciu, 2022-04-10:
I've been searching and trying out so many LED strips (mostly WS2812b), thinking that they are busted or that I did something wrong. This issue occurs to me as well, the only thing is that I cannot get any LEDs to light up at all.

klutvott123, 2022-04-11:
Probably caused by https://github.com/betaflight/unified-targets/pull/462 where motor 3 was moved to dma 1 stream 2 which was used by LED_STRIP

HarisLab, 2022-05-07:
I have the same problem. Only on legacy FW leds are fully working.

HarisLab, 2022-05-14:
Found a quick fix for it.

B01(MOTOR_3) pin and A00(LED_STRIP) pin on versions above 4.1.1 moved to same "DMA 1 Stream 2" and "AF2" timer. Maybe somehow they started to interfere. To fix it type in CLI:

timer B01 AF1
dma pin B01 0
save

spanciu, 2022-05-14:
@HarisLab Thank you! Will give it a try soon. Would there be any downsides expected after remapping?

HarisLab, 2022-05-14:
@spanciu, after remapping I gave it a quick test flight and I haven't noticed any differences. These commands remap MOTOR_3 pin to same timer and dma like it was on 4.1.1 and below versions. I don't think that there could be any downsides because it is not a new fix, more people used it in the past and no downsides were reported. However, I am not a Betaflight software expert, so keep an eye on that. I hope that finally it will be fixed in future releases.

Report back how this fix worked for you.

HarisLab, 2022-06-16:
Looks like that in BF4.3 this problem is gone

dentex, 2022-06-16:
> Looks like that in BF4.3 this problem is gone

It was exactly BF 4.3

closed 2022-07-24 by github-actions[bot]