betaflight/betaflight #11489: Frsky SPI RX can not saving changes with the lua-scripts

ss8688, 2022-03-26:
### Describe the bug

SPI RX can not saving changes, but can read setting. Does this scripts suport SPI RX?

### To Reproduce

Save it

### Expected behavior

Saving ... Retrying
But not saved

### Flight controller configuration

```
  # version
# Betaflight / STM32F411 (S411) 4.2.11 Nov  9 2021 / 20:28:23 (948ba6339) MSP API: 1.43
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FR, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

board_name CRAZYBEEF4FR
manufacturer_id HAMO

# name: mobula6

# feature
feature -RX_PARALLEL_PWM
feature TELEMETRY

# serial
serial 1 2048 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 900 1200 0 0
aux 2 47 1 1200 1700 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels OFF MIN MAX

# master
set acc_calibration = 76,-18,7,1
set mag_hardware = NONE
set baro_hardware = NONE
set dshot_bidir = ON
set motor_poles = 12
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 2
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5865
set frsky_spi_tx_id = 114,63
set frsky_spi_offset = -1
set frsky_spi_bind_hop_data = 2,39,76,113,150,187,224,26,63,100,137,174,211,13,50,87,124,161,198,1,37,74,111,148,185,222,24,61,98,135,172,209,11,48,85,122,159,196,233,35,72,109,146,183,221,22,59,0,0,0
set gyro_1_align_yaw = 900
set name = mobula6

profile 0

rateprofile 0

# end the command batch
batch end

```
```
  # resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 1
TIM3: FREE
TIM4:
    CH1: MOTOR 2
    CH2: MOTOR 3
    CH3: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 2
DMA1 Stream 1: MOTOR 1
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 3
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 4
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

Crazybee f4fr

### Other components

Drone - mobula6 frsky spi
Betaflight 4.2.11
TX - tx16s
edgetx 2.6.0
betaflight-tx-lua-scripts 1.5.0

ss8688, 2022-03-26:
also see https://github.com/betaflight/betaflight-tx-lua-scripts/issues/304

klutvott123, 2022-03-26:
has it ever worked, or is it just with 4.2.11? is your tx module firmware up to date? 
There's nothing in the scripts themselves that would prevent saving from working.

ss8688, 2022-03-26:
It can work with my another drone，which has a F.port rx and bf 4.2.1 .
This mobula6 is newly bought ,so I have only tested 4.2.11 
It seems that eta75 Pro 2 [Betaflight 4.1.2] also have same problem. see see https://github.com/betaflight/betaflight-tx-lua-scripts/issues/304

ss8688, 2022-03-26:
And the [README.md](https://github.com/betaflight/betaflight-tx-lua-scripts#readme) seems have no informaion about SPI RX suport.

ss8688, 2022-03-26:
My module firmware is 1.3.3.7

klutvott123, 2022-03-26:
are you using D8 or D16 mode?

ss8688, 2022-03-26:
D16

klutvott123, 2022-03-26:
do all pages fail to save, or just some of them?

ss8688, 2022-03-26:
all pages

klutvott123, 2022-03-26:
all right, could be something wrong in the firmware then. Maybe try BF 4.3 as there has been some changes since 4.2.11. 
I don't have any hardware with frsky spi rx so I won't be able to reproduce or troubleshoot this. The frsky spi and bf lua has always been a little hit or miss.

dududu76, 2022-03-26:
It doesn't work neither using D8 with the internal FrSky transmitter, a on Taranis X9D+. ( Tested on BF4.2.11 ).
It gets stuck at "Initilizing...". I can't even read values.

klutvott123, 2022-03-26:
you have to use D16. enable telemetry and "discover new sensors"

ss8688, 2022-06-01:
This bug still not fixed in 4.3.0 rc7

ss8688, 2022-06-01:
Though ELRS SPI  eeprom write and reboot for msp over telemetry fixxed,I still can not save changes using frsky spi.
They are not the same issue

ss8688, 2022-06-01:
I think we should re-open this issue

haslinghuis, 2022-06-02:
Did you try:
> you have to use D16. enable telemetry and "discover new sensors"

Only using CRSF and ELRS so unable to test the issue.

ss8688, 2022-06-03:
> Did you try:
> 
> > you have to use D16. enable telemetry and "discover new sensors"
> 
> Only using CRSF and ELRS so unable to test the issue.

@haslinghuis yes,I did that

ss8688, 2022-07-03:
Is anyone test it ?

closed 2022-08-11 by github-actions[bot]