betaflight/betaflight #11491: OSD lagging

trnnwnnhe, 2022-03-26:
### Describe the bug

OSD lagging firmware #2644 and newer. And CPU load +10% => firmware #2643 58% , #2644 70%

https://youtu.be/0uvn8yxY-2k

### To Reproduce

stm32f405  firmware #2644 and newer.

### Expected behavior

fix

### Flight controller configuration

```
  # 

# diff

# version
# Betaflight / STM32F405 (S405) 4.3.0 Feb 27 2022 / 15:02:58 (ccd9508fc) MSP API: 1.44
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name OMNIBUSF4
manufacturer_id AIRB

# resources
resource MOTOR 1 A08
resource MOTOR 4 B00
resource MOTOR 6 NONE
resource SONAR_ECHO 1 NONE

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature ESC_SENSOR

# serial
serial 0 64 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200

# mixer
mmix 0  1.000 -0.875  1.000 -1.000
mmix 1  1.000 -0.875 -1.000  1.000
mmix 2  1.000  0.875  1.000  1.000
mmix 3  1.000  0.875 -1.000 -1.000

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 1 1300 2100 0 0
aux 2 1 5 1300 1700 0 0
aux 3 13 2 1700 2100 0 0
aux 4 35 2 1700 2100 0 0

# adjrange
adjrange 0 0 7 900 1075 6 6 0 0
adjrange 1 0 7 1375 1625 7 6 0 0
adjrange 2 0 7 1900 2100 8 6 0 0
adjrange 3 0 3 900 1200 29 3 0 0
adjrange 4 0 3 1350 1650 29 3 0 0
adjrange 5 0 3 1775 2100 29 3 0 0

# rxfail
rxfail 11 s 1500
rxfail 12 s 1500

# master
set gyro_lpf1_static_hz = 0
set dyn_notch_count = 5
set dyn_notch_max_hz = 500
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_calibration = -61,142,-32,1
set mag_hardware = NONE
set serialrx_provider = CRSF
set blackbox_device = NONE
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set vbat_max_cell_voltage = 435
set current_meter = ESC
set battery_meter = ESC
set yaw_motors_reversed = ON
set mixer_type = LINEAR
set small_angle = 180
set simplified_gyro_filter = OFF
set report_cell_voltage = ON
set telemetry_disabled_acc_x = ON
set telemetry_disabled_acc_y = ON
set telemetry_disabled_acc_z = ON
set telemetry_disabled_pitch = ON
set telemetry_disabled_roll = ON
set telemetry_disabled_heading = ON
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_vbat_pos = 14391
set osd_rssi_pos = 14401
set osd_link_quality_pos = 14369
set osd_tim_2_pos = 14711
set osd_flymode_pos = 14465
set osd_anti_gravity_pos = 70
set osd_throttle_pos = 14679
set osd_current_pos = 10315
set osd_mah_drawn_pos = 14380
set osd_gps_speed_pos = 2199
set osd_gps_lon_pos = 2433
set osd_gps_lat_pos = 2448
set osd_gps_sats_pos = 2167
set osd_home_dir_pos = 2412
set osd_home_dist_pos = 2413
set osd_altitude_pos = 2231
set osd_pid_roll_pos = 8579
set osd_pid_pitch_pos = 8547
set osd_pid_yaw_pos = 261
set osd_debug_pos = 289
set osd_power_pos = 10336
set osd_warnings_pos = 14441
set osd_avg_cell_voltage_pos = 14423
set osd_flip_arrow_pos = 14684
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set cpu_overclock = 216MHZ
set gyro_1_sensor_align = CW270
set gyro_1_align_yaw = 2700

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 80
set dterm_lpf1_dyn_max_hz = 140
set dterm_lpf1_type = BIQUAD
set dterm_lpf2_static_hz = 0
set anti_gravity_gain = 4000
set iterm_relax_cutoff = 11
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set yaw_lowpass_hz = 0
set p_pitch = 70
set i_pitch = 65
set d_pitch = 59
set f_pitch = 0
set p_roll = 70
set i_roll = 65
set d_roll = 59
set f_roll = 0
set p_yaw = 65
set f_yaw = 0
set d_min_roll = 35
set d_min_pitch = 35
set auto_profile_cell_count = 4
set feedforward_averaging = 4_POINT
set feedforward_boost = 0
set simplified_pids_mode = OFF
set simplified_dterm_filter = OFF

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

# end the command batch
batch end

# 
```
```
  ## 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 1
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 4
B01: MOTOR 2
B02: FREE
B03: PREINIT 2
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: FREE
B11: SERIAL_RX 3
B12: FREE
B13: FREE
B14: FREE
B15: FREE
C00: INVERTER 1
C01: FREE
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: PREINIT 3
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 1
    CH2 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Flight controller

airbot omnibus f4

haslinghuis, 2022-04-02:
We are continuous updating firmware, please test with version below to see if the issue still exists:
[image]

trnnwnnhe, 2022-04-06:
> We are continuous updating firmware, please test with version below to see if the issue still exists: [image]

Tried #2663   firmware, it looks like problem is fixed.

trnnwnnhe, 2022-04-06:
Firmware #2659 and later not have discribed bug.