betaflight/betaflight #11492: [Minor Bug] MSP_SET_VTXTABLE_POWERLEVEL

JyeSmith, 2022-03-26:
### Describe the bug

This condition should use VTX_TABLE_MAX_POWER_LEVELS and not vtxTableConfig()->powerLevels.

If already user set or cleared to 0, then MSP can not add power levels above vtxTableConfig()->powerLevels.

https://github.com/betaflight/betaflight/blob/61f43fea9ea5dbd19d378b06ce8cf37b78d50ea6/src/main/msp/msp.c#L3273

### To Reproduce

Try and set a power level above the user set max number.

### Expected behavior

MSP is able to set power level up to VTX_TABLE_MAX_POWER_LEVELS.

### Flight controller configuration

Any

### Flight controller

Any

### Other components

Ill make a PR later if this can be confirmed by other.

JyeSmith, 2022-03-27:
Again here it should be VTX_TABLE_MAX_BANDS.

https://github.com/betaflight/betaflight/blob/61f43fea9ea5dbd19d378b06ce8cf37b78d50ea6/src/main/msp/msp.c#L3237

And a guard here for VTX_TABLE_MAX_CHANNELS.

https://github.com/betaflight/betaflight/blob/61f43fea9ea5dbd19d378b06ce8cf37b78d50ea6/src/main/msp/msp.c#L3230

closed 2022-03-27 by JyeSmith

etracer65, 2022-03-27:
No, it is correct as is. The `MSP_SET_VTX_CONFIG` message is sent first and that updates the max configured power level, band, and channel counts. Then the `MSP_SET_VTXTABLE_POWERLEVEL` and `MSP_SET_VTXTABLE_BAND` messages are sent to update the detail entries.

JyeSmith, 2022-03-27:
> No, it is correct as is. The `MSP_SET_VTX_CONFIG` message is sent first and that updates the max configured power level, band, and channel counts. Then the `MSP_SET_VTXTABLE_POWERLEVEL` and `MSP_SET_VTXTABLE_BAND` messages are sent to update the detail entries.

Thanks Bruce.  I ended up piecing it all together and finding that, so I closed the issue.

JyeSmith, 2022-03-27:
Here's a quick demo of a vtx setting up the table.

https://streamable.com/55z0el

Asizon, 2022-03-27:
Wow @JyeSmith these are fantastic improvements, want to try this openvtx changes now! 

> Here's a quick demo of a vtx setting up the table.
> 
> https://streamable.com/55z0el

Wow @JyeSmith these are fantastic improvements, want to try this openvtx changes now!