betaflight/betaflight #11496: clracingf4 goes into DFU mode when setting set_displayport_msp_serial

Kevc1, 2022-03-30:
### Describe the bug

For BF 4.3 RC 3, BF constantly goes into DFU mode. 
Target is clracingf4. I have an HDZero TX5S.1 VTX  and wired to tx4/rx4 on the FC.

### To Reproduce

In ports tab turned I turn on MSP for UART 4.  I
n CLI and set osd_displayport_device = MSP, save and all is ok. 
Go back into CLI and set displayport_msp_serial = 3. Save.
After doing this, whenever I connect to betaflight is goes into DFU mode. If I disable DFU mode it goes to com3 temporarily then immediately goes back into DFU mode.  Only way I can get around this is to re-flash firmware.  I back leveled to 4.2.11 and all works just fine.

### Expected behavior

Able to connect normally without going into DFU mode constantly.

### Flight controller configuration

```
I cant do a diff because it keeps going into DFU mode. I have flashed  to 4.2.11, so this is a diff for 4.2.11 that is working fine and has all the same settings:
# version
# Betaflight / STM32F405 (S405) 4.2.11 Nov  9 2021 / 20:27:49 (948ba6339) MSP API: 1.43
# config: manufacturer_id: CLRA, board_name: CLRACINGF4, version: 14929e39, date: 2020-08-31T10:05:16Z

# start the command batch
batch start

board_name CLRACINGF4
manufacturer_id CLRA

# name: DQuadO HDZ-Zoom

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature CHANNEL_FORWARDING

# beeper
beeper -BLACKBOX_ERASE

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 3 1 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 0 900 1200 0 0
aux 1 1 1 900 1200 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 900 1225 0 0
aux 4 35 4 1675 2100 0 0
aux 5 49 5 1725 2100 0 0

# adjrange
adjrange 0 0 3 900 2100 12 3 0 0

# master
set gyro_lowpass2_hz = 275
set dyn_lpf_gyro_min_hz = 220
set dyn_lpf_gyro_max_hz = 550
set acc_calibration = -60,13,-32,1
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 16
set serialrx_provider = CRSF
set dshot_bidir = ON
set vbat_min_cell_voltage = 355
set vbat_warning_cell_voltage = 360
set vbat_scale = 160
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 2
set yaw_deadband = 2
set osd_units = IMPERIAL
set osd_warn_batt_not_full = OFF
set osd_warn_core_temp = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_vbat_pos = 10
set osd_rssi_pos = 2048
set osd_tim_1_pos = 2552
set osd_tim_2_pos = 2520
set osd_flymode_pos = 2080
set osd_vtx_channel_pos = 21
set osd_craft_name_pos = 2536
set osd_pidrate_profile_pos = 2112
set osd_warnings_pos = 2505
set osd_avg_cell_voltage_pos = 2528
set osd_disarmed_pos = 2442
set osd_flip_arrow_pos = 2350
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_displayport_device = MSP
set vcd_video_system = PAL
set displayport_msp_serial = 3
set name = DQuadO HDZ-Zoom

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 77
set dyn_lpf_dterm_max_hz = 187
set dterm_lowpass_hz = 165
set dterm_lowpass2_hz = 165
set launch_control_mode = PITCHONLY
set launch_angle_limit = 45

rateprofile 0

# rateprofile 0
set thr_expo = 15
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 77
set roll_expo = 15
set pitch_expo = 15
set yaw_expo = 5
set roll_srate = 66
set pitch_srate = 66
set yaw_srate = 67
set throttle_limit_type = SCALE
set throttle_limit_percent = 99

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: PINIO 1
A15: PREINIT 4
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: PREINIT 3
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: FREE
B09: CAMERA_CONTROL
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: BEEPER
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

### Flight controller

clracingf4

### Other components

happymodel EP2 elrs receiver
HDzero tx5s.1

### How are the different components wired up

receiver connected to uart 6 (tx to rx, rx to tx)
vtx connected to uart 4. tx to rx, rx to tx. power connected to vtx+,vtx- pads. Jumper is set for 7.2v.

### Add any other context about the problem that you think might be relevant here

think I covered it all.
If you really need me to reflash to 4.3 RC3 and do commands without setting displayport_msp_serial = 3 but it may be a day or two before I have time. Send email to kevincyr01@gmail.com if you need this.  I'm just re[porting this to try and help you to make firmware better.

mrRobot62, 2022-03-31:
Question: did you check this strange behavior without connecting HDZero vTX, ESC, ExpressLRS?

If not, to avoid any circumstances around an issue with vTX, or ExpressLRS or ESC please remove this external devices.

1. Start with a naked FC, if this will work, than reconnect ESC - test again - 
2. connect ExpressLRS - check - if it works as design, configure ExpressLRS -> check again (reboot, ....) than 3
3. connect HDZero - check - 

I know - lot of work - but such kind of problems are hard to identify and a good approach is to start step by step. Possible for you?

Thanks

justinflipflops, 2022-04-01:
I have a similar issue. Only goes into DFU when you hook up via USB? If you power via battery first it boots fine and USB communication works as expected? Different board but same VTX. If so I'm going to dig into it some tomorrow and glad I'm not the only one.

haslinghuis, 2022-04-25:
Any updates? Please check RC6

Kevc1, 2022-04-25:
Sorry, been too busy building, flying and running races now that GQ season is underway. When I get a chance I'll try RC6

haslinghuis, 2022-05-09:
What is the ELRS firmware version being used? https://github.com/ExpressLRS/ExpressLRS/releases/tag/2.3.0 seems to implement ELRS / HDzero integration as 2.4.0 is the latest release now.

haslinghuis, 2022-05-13:
Please try `set osd_displayport_device = AUTO` instead of `MSP`.

matyas-brenner, 2022-05-23:
My Mamba mk3 H743 doesn't forget config, but goes to DFU by itself repeatedly. - BF 4.2.11 osd_displayport_device = MSP, not auto

haslinghuis, 2022-05-26:
Closing as duplicate of https://github.com/betaflight/betaflight/issues/11516 as that issue seems more active.

closed 2022-05-26 by haslinghuis

Kevc1, 2022-10-11:
Hmm,Not sure when I could get to this. Camping all weekend and I hope to get some flying in next week.  I also see someone else has the problem, maybe they can make more progress. I'll let you know if I do get to it.Kevc

Baggsz, 2023-01-24:
Damn I have this same problem! As soon as I changed Displayport to MSP it keeps going into DFU Mode! So annoying! I Have a BetaFPV F411 1S ELRS SX1280 board with an HDZero Whoop Lite VTX!
Any news on a fix? Thanks.

automatt1c, 2023-03-04:
I have the exact same issue. If it set "Set osd_displayport_device = AUTO" then i dont have the issue but the MSP doesn't work on the walksnail. It has something to do with the displayport device set to MSP. I also dont have the issue when the battery is plugged in from what i can tell.