betaflight/betaflight #11499: Beeper is not working with Mamba MK4 H743

matyas-brenner, 2022-04-02:
### Describe the bug

Beeper soldered to BUZ+ and BUZ- pads doesn't beep on any trigger.

I have betaflight 4.2.11 set to beep on a switch. When i flip the switch, blue gyro led blinks as it should, so I thing that betaflight is set correctly (Beeper mode is set and turns on a flip of a switch, all the switches in beeper configuration are turned on). The buzzer doesn't beep on any of the situation (USB connected, ready, switch ...).

When I measure voltage between buz+ and buz- pads, it is always about 0,7V. I would expect it to be 5V when buzzer should beep and 0-isch when it should not.
```
   Communication with Diatone:   
   Yes   
  
  We can confirm
  
  As long as FC bec and Mcu working
  
  The buzzer hardware function should be ok
  except the code has some error
  
   BR

    ------------------------------------------------------------------
    发件人：Matyáš Brenner <matyas.brenner@post.cz>
    发送时间：2022年4月2日(星期六) 17:35
    收件人：服务与支持 <[service.support](mailto:<a)@diatone.hk>
    主　题：Re: 回复：Buzzer not working?

    Hello,
    can you please confirm, that it is beta flight reason for the beeper? Is there guthub bug/issue for it?

    I can't see a relation between bidirectional dshot and dedicated beeper. I'm not talking about dshot beeper.

    Matyáš Brenner
    2. 4. 2022, 10:51, "服务与支持" <[service.support](http://service.support/)@diatone.hk target=_blank>service.support@diatone.hk> napsal/a:
    Dear

    It's a beta flight code reason

    Not hardware ,we now busy for solving the Bidirection function under BF4.3

    You can try this testing firmware

    BR
    ------------------------------------------------------------------
    发件人：Matyáš Brenner <mailer@zotabox.com>
    发送时间：2022年4月2日(星期六) 16:16
    收件人：服务与支持 <Service.Support@diatone.hk>
    主　题：Buzzer not working?

    Hello,
    I have Mamba MK4 743 from you. I can't get buzzer to work.
    The buzzer itself works. When I connect it to 5V, it beeps.

    I have betaflight 4.2.11 set to beep on a switch. When i flip the switch, blue gyro led blinks as it should as with your previous FCs, so I thing that betaflight is set correctly (Beeper mode is set and turns on a flip of a switch, all the switches in beeper configuration are turned on). The buzzer doesn't beep on any of the situation (USB connected, ready, switch ...).

    When I measure voltage between buz+ and buz- pads, it is always about 0,7V. I would expect it to be 5V when buzzer should beep and 0-isch when it should not.

    I also checked CLI:
    # get beeper_frequency
    beeper_frequency
```

### To Reproduce

1. Setup Beeper mode
2. Turn on everything in beeper configuration section
3. Activate the beeper by any trigger (usb connected, mode activation, ...), beeper won't beep

### Expected behavior

Beeper should beep when any trigger is fired.

### Flight controller configuration

```
# 

# diff all

# version
# Betaflight / STM32H743 (SH74) 4.2.11 Nov  9 2021 / 20:29:57 (948ba6339) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: MAMBAH743, version: e4b51680, date: 2021-10-04T00:52:01Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAH743
manufacturer_id DIAT
mcu_id 004e002f3230510a33343930
signature 

# resources
resource MOTOR 1 A03
resource MOTOR 2 A02
resource MOTOR 3 A01
resource MOTOR 4 A00

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature GPS
feature ESC_SENSOR

# beeper
beeper -RX_SET

# serial
serial 2 1 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200
serial 7 2048 115200 57600 0 115200

# aux
aux 0 0 2 1350 2100 0 0
aux 1 1 0 1750 2100 0 0
aux 2 2 0 1300 1700 0 0
aux 3 13 3 1700 2100 0 0
aux 4 28 2 1700 2100 0 0
aux 5 35 3 1350 1700 0 0
aux 6 36 1 1700 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY    0    0    0    0    0    0    0    0
vtxtable band 2 BOSCAM_B B FACTORY    0    0    0    0    0    0    0    0
vtxtable band 3 BOSCAM_E E FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY    0 5760    0 5800    0    0    0    0
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I FACTORY    0    0    0    0    0    0    0    0
vtxtable powerlevels 5
vtxtable powervalues 14 23 27 30 0
vtxtable powerlabels 25 200 500 MAX 0

# master
set acc_calibration = -53,-23,93,1
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set osd_displayport_device = MSP
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 1
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5880
set displayport_msp_serial = 2

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

# rateprofile 1
set roll_rc_rate = 127
set pitch_rc_rate = 127
set yaw_rc_rate = 127
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 40
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 72
set tpa_breakpoint = 1750

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 1

# save configuration
save
# 
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 4
A01: MOTOR 3
A02: MOTOR 2
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: FREE
B01: FREE
B02: SPI_MOSI 3
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: PREINIT 4
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: PINIO 1
C06: FREE
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: SERIAL_RX 4
D01: SERIAL_TX 4
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: SERIAL_TX 3
D09: SERIAL_RX 3
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: SERIAL_TX 8
E02: FREE
E03: BEEPER
E04: LED 2
E05: LED 1
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: GYRO_CS 2
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: MOTOR 4
    CH2: MOTOR 3
    CH3: MOTOR 2
    CH4: MOTOR 1
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 4
DMA1 Stream 1: MOTOR 3
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: MOTOR 1
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Flight controller

Diatone Mamba MK4 H743

### Other components

TBS Tracer
HD-Zero Freestyle VTX
Diatone Mammba M22-GPS (which is probably Beitian BN-220)
Diatone F55 128K ESC

### Add any other context about the problem that you think might be relevant here

Motors are intentionally reordered. Stack is mounted "backwards" (180° yaw).

uralm1, 2022-04-03:
Can you test it with
```
set beeper_frequency=0
set beeper_inversion=on
set beeper_od=off
```
in cli? Mamba MK4 H743 config had this settings wrong, I got buzzer to life on 4.3rc3. Btw, it seems we should use at least latest master branch version for the H743 FC.

matyas-brenner, 2022-04-03:
Thank you @uralm1.
after setting
```
set beeper_inversion=on
set beeper_od=off
```
beeper started to work.
`set beeper_frequency=0` was on by default. The two mentioned settings are changed.

I'm leaving this open, so that the values are set as default for Mamba MK4 H743.

sstteevvee, 2022-04-05:
Thanks for the settings to use! I can confirm that on my Mamba MK4 H743 (firmware from the manufacturer was v4.3.0? dated 1 Nov 2021) I had success as well: beeper_inversion=on and beeper_od=off got the beeper to work.

For anyone else troubleshooting their setup: I also found that I had to have my particular model of beeper plugged in the right way around (polarity matters), and to have the battery connected. Having the Mamba powered by USB alone is not sufficient.

closed 2022-04-05 by blckmn