betaflight/betaflight #11506: Kakute H7 Mini 4.3-RC4 locks up when performing certain actions

thujones, 2022-04-07:
### Describe the bug

When using Betaflight 4.3-RC4, Flight controller appears to lockup after saving most configuration settings. Flash pattern appears to be 4/5 rapid flashes followed by 5 or so slow flashes. Non responsive until power cycled.

Note, however, settings that were applied do appear to stick

### To Reproduce

- Backed up 4.3-RC3 config via diff
- Full chip erase and flash with 4.3-RC4
- Apply custom defaults after first initialization
- Apply diff
- During diff apply the usual beeps and bloops that one would hear do not appear to be happening
- Restart flight controller, appears fine, all settings I expect to be there are there
- Make a change to ports (for example was retuning GPS to change baud rate)
- Upon save and reboot operation, flight controller goes into described lockup state
- Reflashed full chip erase again, attempt to set up from scratch no diff (only applying custom defaults) results in same behavior when save/rebooting basically anywhere

- Reverted to 4.3-RC3, apply diff, and normal function resumes without issues

### Expected behavior

- Generally, I suppose, just not locking up when saving configuration of any sort in latest RC4 and above

### Flight controller configuration

```
  # 

# diff

# version
# Betaflight / STM32H743 (SH74) 4.3.0 Apr  6 2022 / 07:58:58 (2fd2e8eca) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEH7MINI, version: c24ea9c4, date: 2021-12-16T07:02:07Z

# start the command batch
batch start

board_name KAKUTEH7MINI
manufacturer_id HBRO

# name: ROMAF4LR

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature GPS

# serial
serial 1 8192 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 2 115200 115200 0 115200

# aux
aux 0 0 0 1975 2050 0 0
aux 1 1 4 950 1025 0 0
aux 2 46 3 1975 2050 0 0
aux 3 13 2 1975 2050 0 0
aux 4 19 0 900 900 0 40
aux 5 26 7 1975 2050 0 0
aux 6 28 5 1975 2050 0 0
aux 7 35 4 1975 2050 0 0
aux 8 40 6 1975 2050 0 0
aux 9 51 0 900 900 0 40
aux 10 52 2 1475 1525 0 0

# adjrange
adjrange 0 0 1 950 1025 29 1 0 0
adjrange 1 0 1 1475 1525 29 1 0 0
adjrange 2 0 1 1975 2050 29 1 0 0

# rxrange
rxrange 0 988 2011
rxrange 1 988 2011
rxrange 2 988 2011
rxrange 3 988 2011

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BAND_A   A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BAND_B   B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BAND_E   E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 AIRWAVE  F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 50 100 200 MAX

# vtx
vtx 0 6 0 0 1 975 1025
vtx 1 6 0 0 2 1175 1225
vtx 2 6 0 0 3 1375 1425
vtx 3 6 0 0 4 1575 1625
vtx 4 6 0 0 5 1775 1825

# master
set gyro_lpf1_static_hz = 0
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 0
set acc_calibration = 3,72,-215,1
set align_mag = CW180FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 1800
set mag_calibration = -920,-643,-1289
set min_check = 1010
set max_check = 1990
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 3000
set vbat_min_cell_voltage = 280
set vbat_warning_cell_voltage = 320
set vbat_scale = 109
set ibata_scale = 112
set beeper_dshot_beacon_tone = 4
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = WAAS
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 45
set gps_rescue_initial_alt = 55
set gps_rescue_descend_rate = 200
set gps_rescue_throttle_hover = 1335
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_alt_mode = FIXED_ALT
set deadband = 2
set yaw_deadband = 2
set osd_units = IMPERIAL
set osd_warn_over_cap = ON
set osd_cap_alarm = 2400
set osd_alt_alarm = 400
set osd_vbat_pos = 6648
set osd_rssi_pos = 14392
set osd_link_quality_pos = 6232
set osd_link_tx_power_pos = 2168
set osd_rssi_dbm_pos = 2200
set osd_tim_2_pos = 14833
set osd_flymode_pos = 14752
set osd_g_force_pos = 14720
set osd_throttle_pos = 2362
set osd_vtx_channel_pos = 2400
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 6551
set osd_mah_drawn_pos = 6521
set osd_gps_speed_pos = 6240
set osd_gps_lon_pos = 6624
set osd_gps_lat_pos = 6592
set osd_gps_sats_pos = 14368
set osd_home_dir_pos = 14702
set osd_home_dist_pos = 6540
set osd_flight_dist_pos = 6272
set osd_compass_bar_pos = 6218
set osd_altitude_pos = 6208
set osd_power_pos = 377
set osd_avg_cell_voltage_pos = 14808
set osd_battery_usage_pos = 47450
set osd_disarmed_pos = 14538
set osd_nheading_pos = 6188
set osd_rtc_date_time_pos = 2048
set osd_flip_arrow_pos = 6446
set osd_log_status_pos = 14756
set osd_efficiency_pos = 6583
set osd_stat_rtc_date_time = ON
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_max_curr = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_stat_total_time = ON
set osd_stat_total_dist = ON
set vtx_band = 3
set vtx_channel = 4
set vtx_power = 1
set vtx_freq = 5645
set vcd_video_system = PAL
set name = ROMAF4LR

profile 0

# profile 0
set feedforward_smooth_factor = 30
set feedforward_jitter_factor = 9

rateprofile 0

# rateprofile 0
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 1
set roll_expo = 65
set pitch_expo = 65
set yaw_expo = 55
set roll_srate = 120
set pitch_srate = 120
set yaw_srate = 90

# end the command batch
batch end

# 
```
```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FLASH_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: MOTOR 3
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: MOTOR 4
B11: PINIO 1
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: LED 1
C03: FREE
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: SERIAL_TX 2
D06: FREE
D07: FREE
D08: SERIAL_TX 3
D09: SERIAL_RX 3
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: CAMERA_CONTROL
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : CAMERA_CONTROL
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: DSHOT_BITBANG 2
DMA1 Stream 1: SPI_MISO 1
DMA1 Stream 2: SPI_MOSI 2
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 4
DMA1 Stream 5: SPI_MISO 4
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Flight controller

Holybro Kakute H7 Mini - From US retailer RDQ

### Other components

- Crossfire Nano on UART3 latest 6.17
- Flywoo GM8 w/compass GPS on UART6, compass I2C connected on provided SCL/SDA ports
- VTX (Flywoo HM600) attached via UART2, powered by VTX+
- Tekko FC via front connector
- Buzzer attached
- Plain ole runcam nano 2 camera attached

### Add any other context about the problem that you think might be relevant here

At first I was coming to this RC4 from flashing Ardupilot and testing it. Thought maybe that might have something to do with it, but I cant imagine what since the full chip was erased to reflash. Figure its worth mentioning at least

haslinghuis, 2022-04-07:
Yes confirm - sometimes it works all the time other times there is no port detected after the reboot took place.

[image]

In this case Windows gives the following warning:

[image]

See also: https://github.com/betaflight/betaflight/issues/11417

thujones, 2022-04-07:
Ahh, much appreciated. Ill keep an eye on that. Thanks!

ctzsnooze, 2022-04-15:
may be fixed in master - please test - by https://github.com/betaflight/betaflight/pull/11524

klutvott123, 2022-04-16:
@thujones which compass is that module using? Can you try setting mag_hardware manually in cli to the compass that's on the module?

haslinghuis, 2023-12-20:
Closing as stale.

closed 2023-12-20 by haslinghuis