betaflight/betaflight #11507: RC3 to RC4 Buzzer now activates 10-20seconds after arming, did not do it on RC3.

Bwsbmw, 2022-04-08:
### Describe the bug

VIFLY Buzzer Beeps unexpectedly after arming. Updated from 4.3 RC3 to 4.3 RC4. Saved diff all to Notepad++, Flashed 4.3 RC4 on FuryF4OSD, and TalonF7DJI. Pasted in my diff all, saved. I compared the RC3 Dump, Diff, Diff All, vs. RC4 Dump Diff, Diff All. Both Match, but upon arming. The Vifly buzzer beeps quickly i believe 5 times, then slowly 3 times. Just once after arming on a fresh battery. This did not happen on RC3 with the same diff. This has happend on 2 quads i have updated from RC3 to RC4.

### To Reproduce

Flash RC4, be using a external buzzer for a beeper. Beeper works as expected on bench. Taken out to fly, once armed about 10-20seconds after take off the buzzer beeps. This happens on 2 different quads i just flashed to RC4 from RC3. 

### Expected behavior

Expect to not hear any beeps after arming, like in RC3 and previously.

### Flight controller configuration

```
```Diff

# version
# Betaflight / STM32F405 (S405) 4.3.0 Apr  6 2022 / 07:49:51 (2fd2e8eca) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: FURYF4OSD, version: acd41165, date: 2021-04-27T20:17:19Z

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id DIAT

# name: MambaF405

# serial
serial 2 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# beeper
beeper -ON_USB

# map
map TAER1234

# aux
aux 0 0 0 1400 2100 0 0
aux 1 1 1 1400 1600 0 0
aux 2 13 2 1900 2100 0 0
aux 3 26 2 1400 1600 0 0
aux 4 35 1 900 1100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# rxfail
rxfail 5 s 1500

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 875
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_calibration = 20,33,138,1
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 1950
set serialrx_provider = CRSF
set min_throttle = 1050
set max_throttle = 1950
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set vbat_max_cell_voltage = 440
set ibata_scale = 266
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 2
set simplified_gyro_filter_multiplier = 175
set osd_units = IMPERIAL
set osd_warn_batt_warning = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_link_quality = ON
set osd_rssi_alarm = 50
set osd_link_quality_alarm = 70
set osd_rssi_dbm_alarm = -105
set osd_cap_alarm = 950
set osd_alt_alarm = 250
set osd_vbat_pos = 6156
set osd_link_quality_pos = 6497
set osd_link_tx_power_pos = 2433
set osd_rssi_dbm_pos = 6465
set osd_tim_1_pos = 2071
set osd_tim_2_pos = 6145
set osd_remaining_time_estimate_pos = 33
set osd_flymode_pos = 89
set osd_anti_gravity_pos = 321
set osd_throttle_pos = 6489
set osd_vtx_channel_pos = 33
set osd_current_pos = 2454
set osd_mah_drawn_pos = 6520
set osd_motor_diag_pos = 353
set osd_craft_name_pos = 10
set osd_altitude_pos = 321
set osd_pidrate_profile_pos = 6202
set osd_warnings_pos = 14634
set osd_avg_cell_voltage_pos = 6188
set osd_battery_usage_pos = 33133
set osd_disarmed_pos = 202
set osd_esc_tmp_pos = 87
set osd_flip_arrow_pos = 6414
set osd_core_temp_pos = 55
set osd_log_status_pos = 2081
set osd_pid_profile_name_pos = 183
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set osd_stat_max_fft = ON
set osd_stat_min_rssi_dbm = ON
set debug_mode = GYRO_SCALED
set vtx_band = 3
set vtx_channel = 6
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5905
set vcd_video_system = NTSC
set gyro_1_align_yaw = 1800
set name = MambaF405

profile 2

# profile 2
set dterm_lpf1_dyn_min_hz = 78
set dterm_lpf1_dyn_max_hz = 157
set dterm_lpf1_dyn_expo = 7
set dterm_lpf1_static_hz = 78
set dterm_lpf2_static_hz = 157
set vbat_sag_compensation = 100
set iterm_relax_cutoff = 12
set p_pitch = 59
set i_pitch = 95
set d_pitch = 40
set f_pitch = 151
set p_roll = 54
set i_roll = 86
set d_roll = 39
set f_roll = 138
set p_yaw = 54
set i_yaw = 86
set f_yaw = 138
set d_min_roll = 39
set d_min_pitch = 40
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 45
set feedforward_jitter_factor = 10
set dyn_idle_min_rpm = 35
set simplified_master_multiplier = 105
set simplified_i_gain = 90
set simplified_d_gain = 125
set simplified_pi_gain = 115
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 110
set simplified_pitch_d_gain = 90
set simplified_pitch_pi_gain = 105
set simplified_dterm_filter_multiplier = 105

rateprofile 0

# rateprofile 0
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 1
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 10
set roll_srate = 80
set pitch_srate = 81
set yaw_srate = 65
set tpa_rate = 70
set tpa_breakpoint = 1300

# end the command batch
batch end

# 
```
```
```resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: CAMERA_CONTROL
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 1
    CH2 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11:
    CH1 : CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Flight controller

MambaF405 mk3 bought from RDQ, TalonF7DJI bought from Heli Nation

### Other components

MambaF405 mk3 4.3rc4 - Vifly buzzer, Tbs Crossfire Nano Rx Se, Rush Tank ultimate mini, foxeer pred v5, rev35 esc.
TalonF7DJI 4.3 rc4 - Vifly Buzzer, Tbs Nano Rx SE, Caddx Vista Nebula Pro, mamba f55 f4 esc.

### How are the different components wired up

correctly as it worked properly on rc3 and previous versions.

### Add any other context about the problem that you think might be relevant here

Does not happen on 4.3 RC3, but happens on both with the same diff all pasted in. compare the rc3 vs rc4 with notepad ++, no changes, settings match on both quads.

haslinghuis, 2022-04-09:
Please try: [betaflight_4.3.0_FURYF4OSD_2fd2e8eca.zip](https://github.com/betaflight/betaflight/files/8455911/betaflight_4.3.0_FURYF4OSD_2fd2e8eca.zip)

Also please try without:
```
# rxfail
rxfail 5 s 1500
```

klutvott123, 2022-04-18:
Does it still do it with RC5?

Bwsbmw, 2022-04-18:
I just tried RC5 on 2 different quads. One being the FuryF4OSD target. Put rxfail 5 back to default on both quads. Buzzer is no longer beeping after takeoff on either quad, and is working properly.

closed 2022-04-18 by haslinghuis