betaflight/betaflight #11508: GPU is not detected in HGLRC F411

dukevah, 2022-04-09:
### Describe the bug

I have a working setup with ```BF 4.2.9``` and ```configurator 10.7.2``` 1. It detects the gps without any issues. Once I flash ```BF 4.3 RC4``` (using ```configurator 10.8.0RC4```), it doesn't detect the GPS. I follow exactly the same setting in both scenarios.

FC: ```HGLRC 16x16 FD411```
GPS: ```HGLRC M80```
I use ```UART 1``` for the setup (it is not a soft serial)

Thanks

### To Reproduce

Connect ```HGLRC M80``` to ```UART 1``` of ```HGLRC 16x16 FD411``` and set the protocol to ```UBOX```. 

### Expected behavior

I expect the firmware detect the gps

### Flight controller configuration

```
  # 

# diff

# version
# Betaflight / STM32F411 (S411) 4.3.0 Apr  6 2022 / 07:57:09 (2fd2e8eca) MSP API: 1.44
# config: manufacturer_id: HGLR, board_name: HGLRCF411, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

board_name HGLRCF411
manufacturer_id HGLR

# name: UFDC4

# resources
resource MOTOR 5 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 B03
resource SERIAL_RX 11 A08

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature TELEMETRY

# serial
serial 0 2 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 30 1 115200 57600 0 115200

# aux
aux 0 0 0 900 1300 0 0
aux 1 13 2 1700 2100 0 0
aux 2 28 2 900 1300 0 0
aux 3 35 2 1300 1700 0 0

# master
set gyro_lpf1_static_hz = 375
set gyro_lpf2_static_hz = 750
set dyn_notch_count = 1
set dyn_notch_q = 500
set dyn_notch_min_hz = 100
set gyro_lpf1_dyn_min_hz = 375
set gyro_lpf1_dyn_max_hz = 750
set acc_calibration = -23,-16,-22,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set rc_smoothing_auto_factor = 250
set rc_smoothing_auto_factor_throttle = 250
set rc_smoothing_setpoint_cutoff = 12
set rc_smoothing_feedforward_cutoff = 12
set rc_smoothing_throttle_cutoff = 20
set serialrx_provider = SBUS
set sbus_baud_fast = ON
set blackbox_device = NONE
set dshot_idle_value = 600
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set motor_output_reordering = 0,2,1,3,4,5,6,7
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 250
set vbat_warning_cell_voltage = 270
set current_meter = NONE
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set deadband = 1
set yaw_deadband = 2
set simplified_gyro_filter_multiplier = 150
set osd_vbat_pos = 2337
set osd_tim_1_pos = 2166
set osd_tim_2_pos = 2134
set osd_flymode_pos = 2401
set osd_throttle_pos = 2305
set osd_current_pos = 2504
set osd_mah_drawn_pos = 2510
set osd_craft_name_pos = 2368
set osd_gps_speed_pos = 2145
set osd_gps_lon_pos = 2051
set osd_gps_lat_pos = 2064
set osd_gps_sats_pos = 2113
set osd_home_dir_pos = 2242
set osd_home_dist_pos = 2177
set osd_flight_dist_pos = 2273
set osd_altitude_pos = 2209
set osd_warnings_pos = 14385
set osd_avg_cell_voltage_pos = 2516
set osd_battery_usage_pos = 33
set debug_mode = GYRO_SCALED
set gyro_1_align_yaw = 1800
set name = UFDC4

profile 0

# profile 0
set p_pitch = 84
set i_pitch = 90
set d_pitch = 56
set f_pitch = 112
set p_roll = 81
set i_roll = 86
set d_roll = 43
set f_roll = 108
set p_yaw = 81
set i_yaw = 86
set f_yaw = 108
set d_min_roll = 43
set d_min_pitch = 56
set feedforward_smooth_factor = 65
set feedforward_jitter_factor = 12
set simplified_master_multiplier = 180
set simplified_i_gain = 60
set simplified_d_gain = 80
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 50
set simplified_pitch_d_gain = 115

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 15
set pitch_expo = 15
set yaw_expo = 15
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75

# end the command batch
batch end

# 
```
```
  resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: SERIAL_RX 11
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: ADC_BATT
B01: FREE
B02: BEEPER
B03: SERIAL_TX 11
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: CAMERA_CONTROL
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : SERIAL_RX 11
TIM2:
    CH2 : SERIAL_TX 11
TIM3:
    CH1 : MOTOR 1
    CH2 : MOTOR 2
TIM4:
    CH1 : MOTOR 3
    CH2 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: MOTOR 1
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Flight controller

HGLRC 16x16 FD411

### Other components

HGLRC M80

### How are the different components wired up

It is connected to UART1

haslinghuis, 2022-04-10:
Please check:

[image]

And the next build for comparison.

dukevah, 2022-04-11:
@haslinghuis can you elaborate on what you mean? is that a fix submitted as a part of that built? should I reinstall the firmware?

dukevah, 2022-05-19:
This issue still exist on RC6. I tried the same configuration on 4.2 and it works while it doesn't work on RC6.

haslinghuis, 2022-05-19:
Please try with defaults. Only add GPS and it's port definition and look at `resource show all`

dukevah, 2022-05-20:
I flashed RC6 with the configurator RC6 into the board. Set the GPS port and its configuration settings. No other changes were applied.
```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: CAMERA_CONTROL
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1 : MOTOR 1
    CH2 : MOTOR 2
TIM4:
    CH1 : MOTOR 3
    CH2 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# ```

I repeated the same process with BF 4.2.11. Now GPS is working. No hardware changes whatsoever. So, it seems such a bug got introduced in 4.3

```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: CAMERA_CONTROL
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
TIM4:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
````

haslinghuis, 2022-05-21:
Please flash latest development build in configurator instead or release candidate.

[image]

dukevah, 2022-05-21:
It didn't fix the issues. Still no sign of GPS!

```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: CAMERA_CONTROL
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1 : MOTOR 1
    CH2 : MOTOR 2
TIM4:
    CH1 : MOTOR 3
    CH2 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 

```

dukevah, 2022-05-21:
It didn't fix the issue.

````
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: CAMERA_CONTROL
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1 : MOTOR 1
    CH2 : MOTOR 2
TIM4:
    CH1 : MOTOR 3
    CH2 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 

```

dukevah, 2022-05-21:
I just noticed another strange behavior. If I keep FC connected to BF for more than 1-2 minutes, the GPS icon turns on. If I unplug the FC and plug it back, it turns off again and has to wait again to see the icon. I observed this behavior both in the latest development and RC6.
NOTE: If I am not wrong, the icon only shows that the communication between FC and GPS is ok and it is nothing to do with the number of satellites. I also checked the number of satellites and it is zero and it is not locked.

vahdat-ab, 2022-05-28:
@haslinghuis can you please make this issue open so we can get more thoughts on this issue.

ea-fpv, 2022-06-02:
> @haslinghuis can you please make this issue open so we can get more thoughts on this issue.

I have been able to reproduce the issue on RC7 with the Betafpv405AIO target and the HGLRC M80 wired to either UART1, UART3 or UART4.

Then flashed 4.2.11 back and the GPS icon worked as expected.

Also note that using an NMEA GPS worked correctly in 4.3 RC6.

Must be something related to UBLOX or this specific module. I will appreciate if someone could take a look at this. My Flywoo Explorer 4" has taken serious damage as a test quad for 4.3 due to Crossfire baud rate negotiation (swithed to ELRS afterward), GPS rescue mode with ELRS (fell from 650m after the battery died since I was unable to recover the radio link) and now the replacement GPS (M80) is not being detected.

Thks

Spent

haslinghuis, 2022-06-02:
Please try build https://ci.betaflight.tech/job/Betaflight/2626/ and go up from there. Build 2627 introduces the problem, likely with PR https://github.com/betaflight/betaflight/pull/10921.
Otherwise please go up from there.

[image]
.

xoga74, 2022-06-12:
Same issue here with RC7 on omnibus F4 SD with UBLOX GPS

xoga74, 2022-06-12:
> Please try build https://ci.betaflight.tech/job/Betaflight/2626/ and go up from there. Build 2627 probably introduces the problem, likely in #10921. Otherwise please go up from there.
> 
> [image]

version 2626 not working either

haslinghuis, 2022-06-13:
Please see comment: https://github.com/betaflight/betaflight/issues/11453#issuecomment-1152866513

haslinghuis, 2022-06-13:
Closing as duplicate of #11453 - please continue there.

closed 2022-06-13 by haslinghuis

rw3aui, 2022-11-25:
m80 4.3.1 not work zeus 10aio, replace with bn-220, works fine, try to test m80 directly. Its nightmare.