betaflight/betaflight #11511: Motors Beep Multiple Times When Navigating OSD Menu

dezignhero, 2022-04-09:
### Describe the bug

The motors (set as buzzer) will beep when exiting the OSD menu when using Save & Exit (not Save & Restart!).  This did not occur for me on RC3.  Sometimes the motors will also beep when jumping in and out of menus on the OSD but not consistent as exiting.

### To Reproduce

Fly, land, enter OSD, fiddle with settings, select Save & Exit.  Get single motor beep (buzzer).

### Expected behavior

Buzzer/Motors should not beep when navigating OSD menus.

### Flight controller configuration

# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Apr  6 2022 / 07:58:20 (2fd2e8eca) MSP API: 1.44
# config: manufacturer_id: HGLR, board_name: HGLRCF722, version: 836acaf9, date: 2021-10-31T19:48:38Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name HGLRCF722
manufacturer_id HGLR
mcu_id 002800073756500620393558
signature 

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature MOTOR_STOP
feature OSD

# serial
serial 2 2048 115200 57600 0 115200

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1300 2100 0 0
aux 1 13 1 1800 2100 0 0
aux 2 26 2 1800 2100 0 0
aux 3 35 1 1300 1700 0 0
aux 4 36 3 1300 2100 0 0
aux 5 49 2 1300 1700 0 0

# adjrange
adjrange 0 0 0 900 2100 29 0 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 20 26 36
vtxtable powerlabels 25 100 400 MAX

# master
set gyro_lpf1_static_hz = 0
set dyn_notch_count = 2
set dyn_notch_q = 450
set dyn_notch_min_hz = 200
set dyn_notch_max_hz = 650
set gyro_lpf1_dyn_min_hz = 500
set gyro_lpf1_dyn_max_hz = 1000
set acc_calibration = 128,7,-132,1
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = CRSF
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_rate = 480
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set simplified_gyro_filter = OFF
set osd_vbat_pos = 14678
set osd_rssi_pos = 14337
set osd_tim_2_pos = 14658
set osd_flymode_pos = 409
set osd_throttle_pos = 14711
set osd_vtx_channel_pos = 2069
set osd_mah_drawn_pos = 408
set osd_motor_diag_pos = 353
set osd_craft_name_pos = 2434
set osd_display_name_pos = 14690
set osd_pidrate_profile_pos = 2105
set osd_warnings_pos = 2345
set osd_pit_ang_pos = 24
set osd_battery_usage_pos = 33097
set osd_flip_arrow_pos = 14574
set osd_core_temp_pos = 2454
set osd_log_status_pos = 14369
set osd_stick_overlay_left_pos = 257
set osd_stick_overlay_right_pos = 278
set osd_stat_max_spd = OFF
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 6
set vtx_power = 3
set vtx_freq = 5843
set vcd_video_system = NTSC
set rpm_filter_fade_range_hz = 100

profile 0

# profile 0
set dterm_lpf1_dyn_expo = 7
set anti_gravity_gain = 5500
set iterm_relax_cutoff = 25
set iterm_limit = 500
set pidsum_limit_yaw = 1000
set throttle_boost = 7
set p_pitch = 42
set i_pitch = 85
set d_pitch = 42
set p_roll = 37
set d_roll = 36
set f_roll = 122
set p_yaw = 46
set f_yaw = 80
set d_min_roll = 22
set d_min_pitch = 22
set d_max_gain = 35
set d_max_advance = 0
set launch_control_mode = PITCHONLY
set thrust_linear = 25
set feedforward_smooth_factor = 35
set feedforward_jitter_factor = 5
set feedforward_max_rate_limit = 100
set simplified_pids_mode = OFF

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 22
set pitch_rc_rate = 22
set yaw_rc_rate = 22
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 71
set pitch_srate = 70
set yaw_srate = 39
set tpa_rate = 70
set tpa_breakpoint = 1550

rateprofile 1

# rateprofile 1
set roll_rc_rate = 22
set pitch_rc_rate = 22
set yaw_rc_rate = 22
set roll_expo = 22
set pitch_expo = 22
set yaw_expo = 22
set roll_srate = 71
set pitch_srate = 60
set yaw_srate = 53
set tpa_rate = 70
set tpa_breakpoint = 1550

rateprofile 2

# rateprofile 2
set roll_rc_rate = 22
set pitch_rc_rate = 22
set yaw_rc_rate = 22
set roll_expo = 22
set pitch_expo = 22
set yaw_expo = 22
set roll_srate = 59
set pitch_srate = 59
set yaw_srate = 49
set tpa_rate = 70
set tpa_breakpoint = 1550

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 

### Flight controller

HGLRC F722 20x20

### Other components

TBS Tracer v6.17

### How are the different components wired up

VTx and Rx both wired directly to Flight Controller.

### Add any other context about the problem that you think might be relevant here

Seems like a momentary Rx loss or failsafe.

klutvott123, 2022-04-13:
save and exit will do a write to flash which takes a while and with the new failsafe code in RC4 this can trigger a rx loss. should be better with https://github.com/betaflight/betaflight/pull/11509

klutvott123, 2022-04-13:
If you could try the latest development build and report back, that would be great. You don't have to fly it. Just test the OSD menu and see if it still beeps.

klutvott123, 2022-04-14:
@dezignhero I just tested it myself and still get the RX_LOST beep when saving to flash. In RC5 the time it takes for rx loss to trigger is 100ms while in BF 4.2 it's 200ms. The time it takes to write the config to flash is roughly 200ms so this explains why it didn't do this before.

dezignhero, 2022-04-14:
> 

Thanks for the update!  Grrrr was really hoping RC5 would fix the issue.  I know it's not official release but generally I assume release candidates would imply a certain level of stability as opposed to development builds.  The last two RC4/RC5 have been rushed quite quickly compared to say the prior release candidates :(

klutvott123, 2022-04-14:
I'll look into it. I don't consider it to be a safety issue as writing to flash isn't allowed when armed, but it's annoying for sure and will leave people wondering what's going on. I believe the reason for this not being discovered by those doing the failsafe testing is that the craft has to be armed at least once since boot for this beep to activate when rx loss occurs and I suspect writing to flash wasn't part of the test procedure 🙂

dezignhero, 2022-04-14:
Thanks @klutvott123 !  This got me thinking but I think these two might be related - https://github.com/betaflight/betaflight/issues/11510

Essentially if I try to turtle mode after saving & exiting OSD menu in RC4, I can't get out of turtle mode and the drone is unresponsive.  My guess is it might be failsafe related too.

klutvott123, 2022-04-14:
Could you please try this? https://github.com/betaflight/betaflight/pull/11524

PR #11524 (2022-04-14): Prevent false failsafe / rx loss on eeprom write for all RX protocols
Currently a write to flash can trigger the failsafe code for non pwm/ppm protocols. 
This PR sets the failsafe valid data received timestamp and link state in order to prevent triggering failsafe during and right after a eeprom write. This change only has an effect in the special case of a write to flash

Binaries for testing:
[betaflight_4.3.0_STM32F7X2_norevision.zip](https://github.com/betaflight/betaflight/files/8489681/betaflight_4.3.0_STM32F7X2_norevision.zip)
[betaflight_4.3.0_STM32F405_norevision.zip](https://github. [... cut]
closed 2022-04-14 by haslinghuis