betaflight/betaflight #11514: dshot_bitbang = off disables RPM Filter on FLYWOOF411

JAL513, 2022-04-10:
### Describe the bug

set dshot_bitbang = OFF disables bidirectional dshot on FLYWOOF411.  BF 4.3 RC4 flashed through 10.8 RC4.

### To Reproduce

enable bidirectional dshot and then set dshot_bitbang = OFF, go back and check bidirectional dshot

### Expected behavior

CRAZYBEEF4DX does not disable bidirectional dshot when turning off dshot_bitbang.  I was expecting similar behavior, but don't know enough about this stuff.

### Flight controller configuration

```
# 

# diff

# version
# Betaflight / STM32F411 (S411) 4.3.0 Apr  6 2022 / 07:57:09 (2fd2e8eca) MSP API: 1.44
# config: manufacturer_id: FLWO, board_name: FLYWOOF411, version: 7b156dec, date: 2021-07-26T13:19:23Z

# start the command batch
batch start

board_name FLYWOOF411
manufacturer_id FLWO

# name: AOS Toothpick

# feature
feature -RX_PARALLEL_PWM

# serial
serial 0 64 115200 57600 0 115200
serial 1 8192 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# led
led 0 4,7::C:10
led 1 5,7::C:10
led 2 6,7::C:10
led 3 7,7::C:10
[... 313 lines cut ...]
GYRO=MPU6000, ACC=MPU6000
OSD: MSP
System Uptime: 60 seconds, Current Time: 2022-04-10T18:16:39.431+00:00
CPU:51%, cycle time: 127, GYRO rate: 7874, RX rate: 15, System rate: 9
Voltage: 117 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS NOPREARM CLI MSP RPMFILTER

status with dshot_bitbang = OFF (which disabled bidirectional dshot), GYRO rate is 8000

# status
MCU F411 Clock=120MHz (PLLP-HSE), Vref=3.31V, Core temp=34degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3994, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000
OSD: MSP
System Uptime: 80 seconds, Current Time: 2022-04-10T18:19:20.498+00:00
CPU:29%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 117 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS NOPREARM CLI MSP

JAL513, 2022-04-10:
Not why or how it would impact things, but LQ of ELRS was very unstable with a new antenna.  At 500mw I was seeing LQ down in the 80s on the other side of my house.  Just incase it is relevant.

haslinghuis, 2022-04-10:
Short analysis: difference here is gyro using dma or not. Status shows `Gyros detected: gyro 1 locked dma` in second case.

klutvott123, 2022-04-11:
try `set dshot_burst = off`

closed 2022-05-20 by github-actions[bot]

smcStevesmc, 2023-05-29:
I had this same issue in my Cetus X, turning dshot off fixed it.