betaflight/betaflight #11516: Mamba h743 forgets config and goes into DFU

Remleyfpv, 2022-04-11:
### Describe the bug

On bF 4.3 RC4 my h743 mamba FC randomly forgets its whole config and sometimes when i try to connect to it it is in DFU mode. And no the bootloader button is not pressed:) 

### To Reproduce

I gues flash bf rc4 to mamba h743 fc. My build has HDzero in it and Elrs (IDk if that has something to do with it)

### Expected behavior

keep config and dont go in to DFU  mode unwanted.

### Flight controller configuration

Entering CLI Mode, type 'exit' to return, or 'hel

# diff all

# version
# Betaflight / STM32H743 (SH74) 4.3.0 Apr  6 2022 / 07:58:58 (2fd2e8eca) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: MAMBAH743, version: 48ff4861, date: 2022-04-03T17:41:53Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAH743
manufacturer_id DIAT
mcu_id 001700353230510a33343930
signature 

# name: cinelifter

# feature
feature -RX_PARALLEL_PWM
feature GPS
feature TELEMETRY
feature ESC_SENSOR

# serial
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 1 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# mixer
mixer OCTOX8

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1300 1700 0 0
aux 2 46 2 1700 2100 0 0
aux 3 13 1 1300 1700 0 0
aux 4 19 3 1700 2100 0 0
aux 5 35 1 1700 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY    0    0    0    0    0    0    0    0
vtxtable band 2 BOSCAM_B B FACTORY    0    0    0    0    0    0    0    0
vtxtable band 3 BOSCAM_E E FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY    0 5760    0 5800    0    0    0    0
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I FACTORY    0    0    0    0    0    0    0    0
vtxtable powerlevels 5
vtxtable powervalues 14 23 27 30 0
vtxtable powerlabels 25 200 500 MAX 0

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 52,-4,63,1
set rc_smoothing_auto_factor = 190
set rc_smoothing_auto_factor_throttle = 50
set rc_smoothing_setpoint_cutoff = 15
set rc_smoothing_feedforward_cutoff = 15
set rc_smoothing_throttle_cutoff = 20
set serialrx_provider = CRSF
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set motor_output_reordering = 7,2,5,0,6,3,4,1
set small_angle = 45
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set report_cell_voltage = ON
set osd_rssi_pos = 64
set osd_link_quality_pos = 14368
set osd_link_tx_power_pos = 14336
set osd_rssi_dbm_pos = 129
set osd_tim_2_pos = 14776
set osd_current_pos = 14795
set osd_gps_speed_pos = 14746
set osd_gps_sats_pos = 14426
set osd_home_dir_pos = 14460
set osd_home_dist_pos = 14393
set osd_altitude_pos = 14359
set osd_power_pos = 14784
set osd_warnings_pos = 14761
set osd_avg_cell_voltage_pos = 14808
set osd_esc_tmp_pos = 14752
set osd_core_temp_pos = 14720
set osd_stat_max_dist = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_stat_total_time = ON
set osd_displayport_device = MSP
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 1
set displayport_msp_serial = 2
set name = cinelifter

profile 0

# profile 0
set feedforward_averaging = 3_POINT
set feedforward_smooth_factor = 75
set feedforward_jitter_factor = 9
set dyn_idle_min_rpm = 10

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 2
set pitch_rc_rate = 2
set yaw_rc_rate = 2
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 

### Flight controller

Mamba H743 with mpu 6000

### Other components

Hdzero 1w, Elrs matek diversity, Dual P60a  4in1 ESC (it's in a cinelifter), TBS gps.

### How are the different components wired up

normal way  (https://rotorbuilds.com/build/29224)

klutvott123, 2022-04-12:
Did it happen on RC3? Did you do a full chip erase?

Remleyfpv, 2022-04-12:
> 

I did indeed not do a full chip erase when on rc3 flash to rc4 but since i have so that wasnt it.
It also did not happen on rc3 btw and tommorw will try flashing rc3 to see if that at least works

Seedytim, 2022-04-13:
I had same issue with matek h743 slim board  I had to reflash firmware to get it to work again

Remleyfpv, 2022-04-13:
i will try reflashing :)

Bruce-surf, 2022-04-29:
same issue in Holybro Kakute H7 with RC6

haslinghuis, 2022-05-03:
Seems related to 
```
set osd_displayport_device = MSP
set displayport_msp_serial = 2
```
See also: #11496

Sekilsgs2, 2022-05-07:
same issue

haslinghuis, 2022-05-07:
Please check with https://github.com/betaflight/betaflight-configurator/actions/runs/2001105326 to have some statistics on MSP.

haslinghuis, 2022-05-14:
Please try `set osd_displayport_device = AUTO` instead

zytra, 2022-05-21:
> same issue.. And FC have random freezes when using configurator. No problems in flight..

same here - random crashes of the configurator.

Sekilsgs2, 2022-05-21:
Im fixing all problems in 4.2.11 (add dps baro, usb msc, serial ...) and using 4.2.12 - no problems at all with 4.2.11

matyas-brenner, 2022-05-23:
My Mamba mk3 H743 doesn't forget config, but goes to DFU by itself repeatedly. - BF 4.2.11

haslinghuis, 2022-05-24:
Without the hardware I'm unable to debug the issue. Please check with this configurator build: https://github.com/betaflight/betaflight-configurator/pull/2932
You might need latest development firmware build.

Please report the console while using F12 when the problem occurs.

Sekilsgs2, 2022-05-25:
Problem usually occur when saving config. Maybe related to flash driver....

haslinghuis, 2022-05-26:
@Sekilsgs2 please check with https://github.com/betaflight/betaflight-configurator/pull/2932
Also might need latest development build to include latest fixes in firmware.
Please provide a console.log (using F12 chrome devtools) when the board goes into DFU mode.

Sekilsgs2, 2022-06-03:
> @Sekilsgs2 please check with [betaflight/betaflight-configurator#2932](https://github.com/betaflight/betaflight-configurator/pull/2932) Also might need latest development build to include latest fixes in firmware. Please provide a console.log (using F12 chrome devtools) when the board goes into DFU mode.

I'm sorry but maybe later - i'm not using 4.3 now - for me 4.2.11 with my fixes and with latest configurator all working with 4.2.11 - maybe this not related to configurator?

matyas-brenner, 2022-06-04:
> Please try `set osd_displayport_device = AUTO` instead

that disables the OSD, so it is unusable as a fix

matyas-brenner, 2022-06-08:
> My Mamba mk4 H743 doesn't forget config, but goes to DFU by itself repeatedly. - BF 4.2.11

Still true with BF 4.3.0 RC7

haslinghuis, 2022-06-08:
Please provide a screenshot and if using a nightly we can check the console.

Also try changing a setting in OSD and save and reboot (minimal 5 times in a row).

matyas-brenner, 2022-06-08:
> Please provide a screenshot and if using a nightly we can check the console.
> 
> Also try changing a setting in OSD and save and reboot (minimal 5 times in a row).

I'm sorry, I don't get it. How can I provide a screenshot of BF configurator going to DFU and not forgetting configuration?
I'm not using Chrome and I don't thing that the stand-alone application has a console.
I'm not using nightly. I'm using BF Configurator 10.8-RC7.

I don't get the note about OSD either. Of course I can change things there, but what does it have to do with BF going to DFU on its own?

haslinghuis, 2022-06-08:
Investigating save and reboot here as it uses a reboot mode (firmware, DFU, MSC).

> Problem usually occur when saving config. Maybe related to flash driver....

There must be an event triggering the reboot?
If in DFU mode can you go to firmware flasher tab and press the Exit DFU button?

How do you mean it goes into DFU on its own? 
Just by plugging USB into configurator what happens?

And you're right the nightly does not include the debug console. (not the chrome console) - need a development build

matyas-brenner, 2022-06-08:
> There must be an event triggering the reboot?
> If in DFU mode can you go to firmware flasher tab and press the Exit DFU button?

Yes, that works as expected.

> There must be an event triggering the reboot?

If there is one, I'm not aware of any.

> How do you mean it goes into DFU on its own?

I'm connected to configuration, going through tabs, not changing anything (with an intent of change something). Suddenly it goes to DFU without me clicking anything. It happens "randomly" not on a specific tab and not related to anything I can thing of.

I'm quite sure that it doesn't happen while being in CLI.

haslinghuis, 2022-06-08:
Yes please. Link here: https://github.com/betaflight/betaflight-configurator/pull/2945#issuecomment-1148711357

matyas-brenner, 2022-06-10:
Here are 2 screenshots of 2 occurrences of the issue with development build.

[image]

[image]

haslinghuis, 2022-06-10:
Thanks this shows it looses serial connection. But there is no boot loader request or other problems showing so must be firmware related.

ligenxxxx, 2022-06-15:
> Problem usually occur when saving config. Maybe related to flash driver....

HDZero VTX does request to write EEPROM after power up.
After the VTX is powered on, it will automatically configure the VTX table and request to write to the EEPROM.
Any good suggestions?

matyas-brenner, 2022-06-15:
> > Problem usually occur when saving config. Maybe related to flash driver....
> 
> HDZero VTX does request to write EEPROM after power up. After the VTX is powered on, it will automatically configure the VTX table and request to write to the EEPROM. Any good suggestions?

I don't thing that it is related. I'm using Freestyle VTX hooked to the FC 9V BEC, which is (to my knowledge) not powered from USB. That should mean that the VTX is not powered on and thus should not request anything.

I can confirm that there is no difference if a battery is connected or if FC is powered only from USB.

matyas-brenner, 2022-06-20:
It is not explicitly written here, just to be sure.
When I set `osd_displayport_device = NONE`, this issue stops (but OSD is obviously not working).
I think that this means that it has something to do with the MSP display device function and I don't think that it is an HDZero issue, because the VTX is not power on when FC is power from USB in my case.

haslinghuis, 2022-06-20:
@matyas-brenner

Using `osd_displayport_device = MSP` and `save` just resets the setting if nothing is attached (on KAKUTEH7MINI).

On the SPRACINGH7EXTREME however the settings survives the reboot after save but I can't get it too fail

Dragony, 2022-07-07:
I have a very similar issue. I'm using a T-Motor F7 Mini and the HDZero VTX V2. I have two setups with those two components. On one of the quads the FC will go into DFU mode randomly withing the first 5-20 seconds after being connected to Betaflight via USB. If I disconnect the Race V2 (I have it soldered to a plug) the issue goes away right away. Since the problem only occurs when I have the VTX plugged in, I assume it's not a Configuration Issue. Any Ideas?

matyas-brenner, 2022-07-09:
In latest hdzero vtx manual there is this troubleshooting suggestion for this issue:

Add a 200ohm resistor between the wire from FC UART.TX to VTX UART.RX

Untested by me so far and I don't intend to.

wernerterreb, 2023-01-14:
Also happens  to me as well.
I have Betaflight / STM32F7X2 (S7X2) 4.3.2 and using Walksnail Vtx 
When I do  "set osd_displayport_device = AUTO" the problem goes away, but it its not ideal.

haslinghuis, 2023-10-23:
Should be fixed in the 4.4.3(-zulu) release version

closed 2023-10-23 by haslinghuis