betaflight/betaflight #11520: SmartAudio Intermittent Connection

dakoellis, 2022-04-13:
### Describe the bug

This seems to be an issue introduced in 4.3, as I have no issues with SmartAudio in 4.2.11 (using the same backup file).  When connecting to Betaflight Configurator and going to the video transmitter tab, A vast majority of the time I get VTX Type: SmartAudio Unknown and Device Ready: No. in the past 3 days while troubleshooting, I've seen VTX Type: SmartAudio 2.1 and Device Ready: yes 3 times, but as soon as power is cycled it reverted. Loading the exact same backup file in 4.2 (which completes with errors) works and smartaudio gives device ready: yes.  

Same issue with both RC3 and RC4.  Also hasn't been controllable through OSD or Backpack, while both work on 4.2.

### To Reproduce

- Flash 4.3 RC4 Firmware
- Go to Video Transmitter Tab
- SmartAudio is not working

### Expected behavior

- Video Transmitter tab shows following: 
  - VTX Type: SmartAudio 2.0 (I believe its a 2.0 device and not 2.1?)
  -Device Ready: yes
- VTX is controllable via SmartAudio

### Flight controller configuration

```
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Apr  6 2022 / 07:58:20 (2fd2e8eca) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: MAMBAF722_I2C, version: ec268d11, date: 2021-07-26T13:10:45Z

# start the command batch
batch start

board_name MAMBAF722_I2C
manufacturer_id DIAT

# serial
serial 2 2048 115200 57600 0 115200

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BAND_A   A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BAND_B   B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BAND_E   E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 AIRWAVE  F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set acc_calibration = -136,1,-4211,1
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end

```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
    CH4 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 1

```

### Flight controller

 Diatone Mamba DJI F722 MK2

### Other components

vTX: Rush FPV Ultimate Mini

### How are the different components wired up

vTX SA to TX3
vTX VID to VID (RX3)

klutvott123, 2022-04-16:
@dakoellis DId it work on earlier 4.3 RC versions?

dakoellis, 2022-04-16:
I just checked on RC1 and RC2 (and RC5 just for kicks) and no not working on any of those.  It's a new FC for me and the first I've tried 4.3 on

Wy1c, 2022-04-25:
I have the same FC and mine flys fine for 30 sec then shoots straight up and yaws by itself,  tried RC3 and 5 ?

klutvott123, 2022-04-25:
@Wy1c Could you open an issue for that and provide the requested iformation when doing so?

urban2015, 2022-08-02:
I have the f722 fc and a Rush FPV Ultimate 2. I cannot get smart audio to work on the befaflight 4.3 and it works intermitanly on the 4.2*

GnarLito, 2022-09-07:
I had the same problem.
Try using 
```
vtxtable powervalues 14 23 27 30
```

urban2015, 2022-09-12:
Just tried that @GnarLito same issue

Prajwal-Koirala, 2023-12-27:
Hi there,

I've been looking into this issue you reported with SmartAudio not working on Betaflight 4.3 RC4 firmware, and I understand it's quite frustrating.

It seems like you've tried a few troubleshooting steps already, including flashing different firmware versions, but the problem persists. And it looks like you're not alone, as other users with the Mamba F722 MK2 flight controller and Betaflight 4.3 RC4 combination have reported similar issues.

Unfortunately, there's no known fix for this issue yet. However, I'd recommend staying tuned to the Betaflight GitHub repository and discussion forums, as the developers are actively working on resolving it. You can also try searching for workarounds or alternative solutions in the meantime.

In the meantime, I'd be happy to help with any other questions you have or if there's anything else I can do.

haslinghuis, 2024-05-12:
Closing as outdated / stalled

closed 2024-05-12 by haslinghuis