betaflight/betaflight #11527: 4.3RC4 ELRS cannot regain control after failsafe (GPS rescue) because stick input is not registering therefore stick treshold cannot be reached

pzencar, 2022-04-15:
### Describe the bug

## Problem description

When failsafe occurs and GPS rescue failsafe is set, during GPS rescue when connection is reestablished (RSSI and LQ values are back active), it is impossible to regain control no matter what I do. I found out (by adding stick overlay to OSD) that this is because when connection is back, no stick movement registers and therefore stick treshold cannot be achieved to gain back conenction.

## Temporary functional fix

In order to be able to regain control after GPS rescue failsafe, `failsafe_stick_threshold` must be set to **0**. This means that the moment the connection is back, control is regained. Even tho during the failsafe stick movements do not register, after setting the treshold to 0, the sticks register correctly after regaining control

### To Reproduce

## Reproduce in air

Turn off the controller while using ELRS and GPS rescue failsafe, turn the controller back again and there is no way to regain control as sticks do not register (preferably enable stick overlay in OSD or log to blackbox). Drone will auto emergency land where the home point is set.

## Reproduce without flying

- Preferably set GPS rescue home point to 50 m and set maximum throttle lower
- Take off props
- Enable `set homepoint only once`
- Connect battery, wait for GPS lock and arm the quad to set home point
- walk 50+ meters away
- Rearm the quad, turn off the controller to enduce GPS rescue failsafe (motors spinning at max set RPM)
- Turn on controller and try to regain control, sticks do not register therefore treshold cannot be reached

### Expected behavior

When GPS rescue failsafe is active and connection is regained, sticks movement should register and when set treshold is set, control should be regained of the quad.

### Flight controller configuration

```
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Apr  6 2022 / 07:58:20 (2fd2e8eca) MSP API: 1.44
# config: manufacturer_id: IFRC, board_name: IFLIGHT_BLITZ_F722, version: bc605bcd, date: 2021-10-29T02:43:03Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFLIGHT_BLITZ_F722
manufacturer_id IFRC
mcu_id 0039001b4e30500c20343135
signature 

# name: Nazgul5 V2

# feature
feature GPS
feature LED_STRIP

# serial
serial 0 8192 115200 57600 0 115200
serial 2 2 115200 57600 0 115200

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -SYSTEM_INIT
beeper -ON_USB

# led
led 0 0,0::L:0
[... 348 lines cut ...]
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 2
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: MOTOR 3
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: MOTOR 4

# 
```

### Flight controller

IFlight F722 (Nazgul 5 V2 OEM)

### Other components

- Happymodel ELRS ES-24TX PRO module (FW version 2.3)
- Betaflight ELRS Nano RX module (FW version 2.3)
- Radiomaster TX12 (Edge TX)
- TBS GPS

haslinghuis, 2022-04-15:
Please try RC5

pzencar, 2022-04-15:
@haslinghuis Hm, RC5 does the same regarding sticks not registering, but whats worse, instead of `!FS!` mode, the quad goes to `ANGLE` mode when controller is disconnected and cant do anything when controller is back on. This means that the GPS rescue would not work if I tried to test in flight and I am not sure what it would do.

Beeper beeps when controller is disconnected and stops beeping when connection is reestablished

haslinghuis, 2022-04-15:
@ctzsnooze

pzencar, 2022-04-16:
@haslinghuis @ctzsnooze Rolled back to RC3 and everything works correctly. Upon controller turn off , it goes to GPS rescue failsafe and when controller is back on, the stick movement registers in OSD and I recover the control once I reach set stick threshold.

haslinghuis, 2022-04-16:
@pzencar please try https://github.com/betaflight/betaflight/pull/11530

pzencar, 2022-04-16:
@haslinghuis Can you please provide the FW files to load prior to flashing or elaborate how to flash it ? Have not tried to use any PR firmware before.

haslinghuis, 2022-04-16:
11530: https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=6277&view=artifacts&pathAsName=false&type=publishedArtifacts

11531: https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=6278&view=artifacts&pathAsName=false&type=publishedArtifacts

pzencar, 2022-04-16:
Thank you, 11530 seems to be working correctly on the bench without props! Thanks for the quick response guys, gonna try in flight today also. Only difference is that `mode` OSD field now displays `RESC` instead of `!FS!` when in GPS rescue failsafe, but failsafe warning is being displayed and functionally everything is correct so not sure if intended, just letting you know.

Should I also try the 11531 ?

haslinghuis, 2022-04-16:
Yes please it seems an alternative to 11530 (so a bench test of 11531 is appreciated).

Also 11532 (build on top of 11531) seems to deal with the !FS!: https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=6281&view=artifacts&pathAsName=false&type=publishedArtifacts

pzencar, 2022-04-16:
11532 works the same way as 11530 from what I tested. Same `RESC` mode is displayed instead of `!FS!`. 

However 11532 keeps beeping even when everything is connected and ARMed and not sure why, only changes I made was plug a fresh battery and turn off RC smoothing.

ctzsnooze, 2022-04-16:
Flight mode should be RESC when the quad is in GPS Rescue mode. That’s the correct behaviour of the flight mode field. 

You should also see RXLOSS during the Stage 1 period. 

Can I clarify that you were able to successfully restore control by restoring the link during a true signal failure GPS Return?

ctzsnooze, 2022-04-16:
Mode display will say !FS! only after drop or during and after Stage 2 Landing mode.

pzencar, 2022-04-16:
@ctzsnooze Yes, just came back from flying (FW PR #11532)  and GPS rescue failsafe was used 7+ times both by turning off the controller and also naturally by signal loss due to range. Everytime regaining control worked like it's supposed to, control was regained after I moved the sticks the specified amount.

closed 2022-04-16 by haslinghuis

ctzsnooze, 2022-04-17:
@pzencar thank you so much for your careful testing.  I’m very grateful - and relieved.
I’d also appreciate feedback on 11532 if you could.

pzencar, 2022-04-18:
@ctzsnooze Ofcourse, rather thank you guys for quick response and great work you doing on this awesome project :)