betaflight/betaflight #11535: RXLOSS during flight triggers five+three beep code

joshuabardwell, 2022-04-16:
### Describe the bug

I've gotten two separate reports this morning of people flying either RC4 or RC5 (not able to verify which) where during flight they got a RXLOSS warning and the FC made the five+three beep code. Example: https://youtu.be/mwUJhmp30Hk

### To Reproduce

Be flying. Get RXLOSS during flight. Have a buzzer wired up.

### Expected behavior

Arming Prevention beep codes indicate why arming is not occurring at the time arming is attempted. They should not trigger while the quad is armed.

### Flight controller configuration

Don't have.

### Flight controller

Don't know.

joshuabardwell, 2022-04-16:
Since I know RXLOSS warning has been re-worked in the latest RC's I wanted to pass this back as I think it may be related. Sorry I don't have more information about the specifics.

haslinghuis, 2022-04-16:
Please use development build to verify.
Also https://github.com/betaflight/betaflight/pull/11532 might relax the issue here.

jfrowies, 2022-04-17:
I was one of the people that asked Joshua about the problem.

I can confirm that for me it was consistently happening in RC4 and I can also confirm it's fixed in RC5.

Flight controller
SucceX-E F4 V2.2 Flight Controller

Other components
SucceX-E 45A 2-6S BLHeli_S Dshot600 4-in-1 ESC
Caddx vista
ELRS receiver

I have now successfully tested that I don't hear the beeps anymore in any situations. I've tried ERLS in 150mhz up to 500mhz refresh rate and with different telemetry ratios. It worked consistently well in all configurations.

haslinghuis, 2022-04-17:
Fixed in RC5

haslinghuis, 2022-04-17:
Fixed in RC5

closed 2022-04-17 by haslinghuis