betaflight/betaflight #11539: [G4] OSD causes firmware lockup

DusKing1, 2022-04-19:
### Describe the bug

No matter which CPU clock is, when enable `OSD` feature via configurator or CLI, the FC will lockup after save and reboot

### To Reproduce

Flash latest G4 firmware (nightly build), enable OSD, then FC will lockup.

### Expected behavior

FC won't lockup and OSD can be used normally.

### Flight controller configuration

```
diff

# version
# Betaflight / STM32G47X (SG47) 4.3.0 Apr 15 2022 / 10:18:11 (9360ab1) MSP API: 1.44
# config: manufacturer_id: NERC, board_name: HYBRIDG4, version: feb0d368, date: 2021-12-25T12:38:42Z

# start the command batch
batch start

board_name HYBRIDG4
manufacturer_id NERC

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL

# serial
serial 0 1 115200 57600 0 115200
serial 2 0 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1950 2050 0 0
aux 1 1 2 1950 2050 1 0
aux 2 13 3 1950 2050 0 0
aux 3 26 1 1950 2050 0 0
aux 4 28 2 950 1050 1 0

# master
set gyro_hardware_lpf = OPTION_2
set gyro_lpf1_static_hz = 0
set gyro_lpf2_type = PT2
set gyro_lpf2_static_hz = 135
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_calibration = 0,-111,44,1
set serialrx_provider = SBUS
set blackbox_sample_rate = 1/1
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set align_board_yaw = -90
set current_meter = ADC
set battery_meter = ADC
set beeper_dshot_beacon_tone = 2
set small_angle = 35
set pid_process_denom = 1
set simplified_gyro_filter = OFF
set debug_mode = GYRO_SCALED
set gyro_1_align_yaw = 0

profile 0

rateprofile 0

# end the command batch
batch end

#
```

```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: I2C_SCL 1
B00: MOTOR 1
B01: MOTOR 2
B02: MOTOR 3
B03: MOTOR 4
B04: FREE
B05: SPI_MOSI 3
B06: CAMERA_CONTROL
B07: I2C_SDA 1
B08: FREE
B09: FLASH_CS
B10: FREE
B11: FREE
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: BEEPER
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH1 : CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM15: FREE
TIM16: FREE
TIM17: FREE
TIM20: FREE

Currently active DMA:
--------------------
DMA1 Channel 1: DSHOT_BITBANG 2
DMA1 Channel 2: SPI_MOSI 1
DMA1 Channel 3: SPI_MISO 1
DMA1 Channel 4: SPI_MOSI 3
DMA1 Channel 5: SPI_MISO 3
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA1 Channel 8: FREE
DMA2 Channel 1: FREE
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: ADC 1
DMA2 Channel 6: FREE
DMA2 Channel 7: FREE
DMA2 Channel 8: FREE

#
```

### Flight controller

AirFleetG4 dev board v1.1

klutvott123, 2022-04-19:
Did it work in the previous nightly build?

DusKing1, 2022-04-19:
It worked before the SPI DMA

klutvott123, 2022-04-19:
this one https://github.com/betaflight/betaflight/pull/10909 ?

DusKing1, 2022-04-19:
Maybe much earlier than that.. I can't remember it exactly.

haslinghuis, 2022-05-08:
Closing as inactive

closed 2022-05-08 by haslinghuis