betaflight/betaflight #11544: Function To Determine if Quad is Inverted

KenImhof, 2022-04-21:
### Is your feature request related to a problem? Please describe

I would like to implement 3d horizon mode to self-level when inverted, but need to determine if quad is upside-down.

### Describe the solution you'd like

Does BetaFlight have a function similar to iNAV to determine if upside-down.  The iNAV function is:

/*-----------------------------------------------------------
 * Detects if thrust vector is facing downwards
 *-----------------------------------------------------------*/
bool isThrustFacingDownwards(void)
{
    return (calculateCosTiltAngle() >= 0); //cos(90) = 0 kbi
}

### Describe alternatives you've considered

Continue using iNAV.

klutvott123, 2022-04-22:
ìsUpright` may be what you want

KenImhof, 2022-04-22:
That is great!!!!!

Thanks for your help.

KenImhof, 2022-04-22:
Thanks for your help and quick response!!!!!

closed 2022-04-22 by KenImhof