betaflight/betaflight #11547: ACC Trim sticks input discrepancy in BF 4.3 RC5 firmware vs configurator

Martivip, 2022-04-21:
### Describe the bug

I'm noticed that one Acc Trim command increment done by sticks command are 2 value increments from the configurator perspective. From configurator perspective there is possibility to perform the ACC trim increments by one single step.

I think that discrepancy stick commands vs configurator should be removed, via one ACC trim increment performed via stick command should be done by 1 value from configurator perspective.

This discrepancy has been found during the works on https://github.com/betaflight/betaflight/issues/11438 and thanks to **@klutvott123** that he found that below discrepancy in the firmware code

https://github.com/betaflight/betaflight/blob/954b7373bbe8a5a64be68dbfa5d030d03f0c56bb/src/main/fc/rc_controls.c#L317-L334

hopefully fix for that looks easy and only changes in above code lines are required and only from firmware perspective not configurator.
Sorry I'm not a Dev so I can't perform that easy changes by my self I still need to learn how the GitHub works 😄

### To Reproduce

Perform the Acc Trim command via stick command and you will found that one increment ACC Trim via stick command is 2 increments from configurator perspective.

### Expected behavior

One ACC Trim increment made via stick command should be made in step of one, not two 

### Flight controller configuration

```
N/A
```
```
N/A
```

### Flight controller

KAKUTEF7

limonspb, 2022-05-07:
Could you please test:
https://github.com/betaflight/betaflight/pull/11588

Hex files:
https://artprodeau1.artifacts.visualstudio.com/A3a9037bc-5464-4b2e-9e8d-ed1a5aa29705/ce1d2eeb-ce65-4b95-a02a-7fee6c006c71/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL0JldGFmbGlnaHQvcHJvamVjdElkL2NlMWQyZWViLWNlNjUtNGI5NS1hMDJhLTdmZWU2YzAwNmM3MS9idWlsZElkLzY0MDkvYXJ0aWZhY3ROYW1lL2JldGFmbGlnaHQ1/content?format=zip

Martivip, 2022-05-07:
@limonspb ok after upload firmware from https://github.com/betaflight/betaflight/pull/11588 
everything works as it should 😄
ACC Trim increments made via stick command are made in step of one increments

limonspb, 2022-05-10:
> @limonspb ok after upload firmware from #11588 everything works as it should 😄 ACC Trim increments made via stick command are made in step of one increments

Could you please continue discussion here:
https://github.com/betaflight/betaflight/pull/11588#issuecomment-1121476008

closed 2022-05-13 by haslinghuis