betaflight/betaflight #11548: Very high CPU load when Flash Mamba F722 APP to 4.3

ArrasPilot, 2022-04-21:
### Describe the bug

Hello, use FC Mamba F722 APP (MAMBA F722 IC2). When I want to upgrade my Fc to version 4.3, the cpu load rises to 35% with I2C errors. If any preset is applied, the CPU load rises to 60% ..... (It heats up quite a bit). It does it to me with any version of 4.3...
Also, if I try motors one by one, everything goes well, but if I arm motors with the controller (TS16 with LRSExpress), there is a problem: example, you go straight and the drone has a very fast impulse to the right (like a phase shift of some motor). I know that the problem comes from the version because I installed 4.2.11 and I don't have any errors and it works perfectly with CPU load at 4%, engines ok etc...

I have tried at a friend's house to install 4.3 on an F4 Hobbywing and an F722 Mamba and the same error happened.

I have checked PC, drivers, versions, etc... but nothing.

Is it normal to have a CPU load at 60%, is there no risk that it gets too hot?

### To Reproduce

Upgrade my Fc to version 4.3

### Expected behavior

The cpu load rises to 35% with I2C errors. If any preset is applied, the CPU load rises to 60% ..... (It heats up quite a bit). It does it to me with any version of 4.3...

Also, if I try motors one by one, everything goes well, but if I arm motors with the controller (TS16 with LRSExpress), there is a problem: example, you go straight and the drone has a very fast impulse to the right (like a phase shift of some motor)

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

DIAT- MAMBA F722 I2C

haslinghuis, 2022-04-21:
Please edit your comment to include the  `diff` and `resource show all` as requested when opening the issue.

haslinghuis, 2022-04-22:
Please read: https://github.com/betaflight/betaflight/wiki/4.3-Tuning-Notes#betaflight-43-introduction

[image]

closed 2022-04-22 by haslinghuis