betaflight/betaflight #11551: High CPU usage BF 4.3 tested by Crazybeef4fs and Matek F405ctr

tn2009, 2022-04-22:
### Describe the bug

BF 4.3 (50%) has too much CPU usage compared to BF 4.2.11 (12%). The difference can be seen in the screenshots.
Tested by Crazybeef4fs and Matek F405ctr

### To Reproduce

Flashing BF 4.3

### Expected behavior

The CPU load should be as in BF 4.2.11

### Flight controller configuration

# diff all

# version
# Betaflight / STM32F411 (S411) 4.3.0 Apr 22 2022 / 08:03:16 (67ffcf8f1) MSP API: 1.44
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FS, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4FS
manufacturer_id HAMO
mcu_id 003700473439511536383737
signature 

# name: Mobula 6

# serial
serial 1 2048 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1250 2100 0 0
aux 1 1 2 900 1225 0 0
aux 2 35 3 1250 2100 0 0
aux 3 36 1 1250 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 50 100

# master
set dyn_notch_count = 1
set dyn_notch_q = 250
set dyn_notch_min_hz = 125
set dyn_notch_max_hz = 350
set acc_calibration = -31,30,-234,1
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 10
set airmode_start_throttle_percent = 50
set dshot_idle_value = 600
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set bat_capacity = 300
set vbat_max_cell_voltage = 450
set vbat_min_cell_voltage = 320
set vbat_warning_cell_voltage = 325
set ibata_scale = 1175
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 5
set yaw_deadband = 5
set runaway_takeoff_prevention = OFF
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_rssi_pos = 2457
set osd_tim_2_pos = 2519
set osd_flymode_pos = 2553
set osd_throttle_pos = 2488
set osd_current_pos = 2464
set osd_mah_drawn_pos = 2496
set osd_craft_name_pos = 394
set osd_warnings_pos = 2537
set osd_avg_cell_voltage_pos = 2529
set osd_disarmed_pos = 266
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 1
set vtx_freq = 5658
set gyro_1_align_yaw = 900
set flysky_spi_tx_id = 1364672457
set flysky_spi_rf_channels = 80,151,107,26,71,156,114,40,57,129,89,12,45,162,122,19
set name = Mobula 6

profile 0

# profile 0
set anti_gravity_gain = 4500
set crash_recovery = ON
set iterm_relax_cutoff = 20
set p_pitch = 71
set i_pitch = 63
set d_pitch = 67
set f_pitch = 249
set p_roll = 68
set i_roll = 60
set d_roll = 62
set f_roll = 239
set p_yaw = 68
set i_yaw = 60
set f_yaw = 239
set level_limit = 65
set d_min_roll = 0
set d_min_pitch = 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 110
set pitch_rc_rate = 110
set yaw_rc_rate = 170
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 5
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 75
set tpa_breakpoint = 1250

rateprofile 1

# rateprofile 1
set roll_rc_rate = 110
set pitch_rc_rate = 110
set yaw_rc_rate = 170
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 5
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 75
set tpa_breakpoint = 1250

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: RX_SPI_EXTI
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 1
TIM3: FREE
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 2
DMA1 Stream 1: MOTOR 1
DMA1 Stream 2: SPI_MISO 3
DMA1 Stream 3: MOTOR 3
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 4
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

### Flight controller

Crazybeef4fs and Matek F405ctr

klutvott123, 2022-04-22:
The way cpu usage is calculated was changed in 4.3. What you're seeing is normal

daleckystepan, 2022-04-22:
Please read https://github.com/betaflight/betaflight/wiki/4.3-Tuning-Notes especially 8)

haslinghuis, 2022-04-22:
[image]

haslinghuis, 2022-04-22:
Closing as duplicate of https://github.com/betaflight/betaflight/issues/11548

closed 2022-04-22 by haslinghuis