betaflight/betaflight #11554: Discussion of MSP abstraction for DJI OSD

howels, 2022-04-24:
### Is your feature request related to a problem? Please describe

I am recording a conversation from discord. This is not a personal request but an attempt to record a discussion to aid discussion in the longer term and avoid chat topics being lost and needlessly repeated.
Many of the developers are familiar with the problems relating to DJI OSD, as laid on in the PR https://github.com/betaflight/betaflight/pull/9987 and numerous other issues. 
A question was raised on discord "given some changes in the project's leadership, would the team be open to looking at changes to enable DJI OSD?".  This does not relate to this single issue, but leads to a wider question "what is the process to evaluate whether features are suitable for inclusion?".  

### Describe the solution you'd like

Clarification on the process for feature inclusion and prioritisation of that work.

- Process to nominate new features
- Factors that may likely bar a new feature from inclusion
- Guidelines for prioritisation and timelines to ensure adequate testing and stability

### Describe alternatives you've considered

Specifically for the DJI OSD, I am maintaining a simple patch for the OSD but this over-writes the `pilotConfigMutable()->name` field which results in loss of the craft_name and resulting problems like the warning messages being used when saving config dumps etc.
Going forward we could adopt a cleaner, but more intrusive, solution - by abstracting the MSP protocol implementation and having a standard and DJI variant for MSP. However this clearly breaks some of the previous guidance in Betaflight such as not compromising standards to workaround vendor-specific bugs.
Guidance on the development team's position on this and similar items would be helpful. If a groundswell of opinion backs a feature such as this then should it be included?

### Other information

For clarification, I personally agree with the prior decision to not include DJI OSD in mainline BF builds and would rather wait for DJI goggles to receive root and full MSP displayport support instead of implementing a hack. On the other hand if MSP abstractions might be required for other purposes this could be an interesting use of such a system. 
Those interested in the current hack for DJI OSD can see my scribblings at https://github.com/howels/betaflight/releases

bakwc, 2022-04-24:
As a BF and DJI user who experienced **several crashes** because of missed warnings, I strongly disagree with the prior decision.

Current "hack":
 - Doesn't break any existing functionality
 - Has a low maintanance cost
 - Has no impact on arhitecture
 
I haven't seen any real argument why this should not be included in a BF. All arguments sounds like "this should be done on a DJI side" and "this is a hack".

Two years have passed, this is not done on a DJI side, we have a comunity of users suffering from missing warnings. There is no winner here. The monopolic corporation on one side, and a few perfectionist guys on the other.

Storing this hack in a separate repo as a patch has following downsides:
 - It is much harder to find and install this patch
 - If we have two hacks (or two rejected features) - the users without programming backround would not be able to use them both at the same time

Ideally, solution should work by default for DJI users without need to even configure anything. The number of users that needs warning in DJI googles is much greater than the number of users who need craft name.

stefan73, 2022-05-04:
Betaflight has in other areas greatly supported users with vendor-related "hacks" & extensions. 
Two examples out of more:
#1 The support of the Flysky FS-ia6 receiver, which can be modified for a kind of IBUS protocol 'dialect'. Betaflight happily supports this "hack" in src/main/rx/ibus.c   
#2 Video Transmitters, where the manufacturer (e.g. Matek) has screwed up the Tramp protocol and inverted the PIT setting.

There is the howels way. There is INAV simply doing it. Both seem to not cause a problem.
It is great that BF devs are after quality, standards & user support. Maybe time is due now.

jmullo, 2022-05-05:
I've barely flown two batteries with my DJI goggles and here I am wondering why this is still an open issue...

As a software developer I get it, we want clean and uncompromised architecture. But the fact is that the users don't care how elegant your code is, they only care about the features they need.

So it really is a simple choice: do we want to serve the users or do we want to serve the developers. And that should be a no-brainer.

bakwc, 2022-05-05:
Betaflight team members, please comment on this issue. Obviously the strategy "force DJI to fix it" doesn't work. May be it's time to reconsider decision?

howels, 2022-05-05:
One reason that this particular issue may not be solved is the existence and rapid pace of development of items such as https://github.com/bri3d/msp-osd.
However the question of general policy stands.

a1j, 2022-05-26:
I'm forced to use hacked (and possibly untested) source branch until this is implemented. I think pilot's need is more important than ideology in this particular case.

haslinghuis, 2022-05-26:
Unlocking DJI FPV and understanding the Fatshark Dominator / WalkSnail Avatar HD FPV System

https://www.youtube.com/watch?v=fvXxu5mqSoo

Crashin2416, 2022-06-26:
No, please do not let this go stale and fall to the wayside.  
Eagerly awaiting a resolution implementing BF OSD in whole or in part by official main branch.

chilternflyer, 2022-07-30:
I've just rebuilt my Titan DC5 and have had to put INAV back on it because of the lack of warnings in DJI OSD. (Went all the way to the field yesterday only to have it not arm but I've no frickin' idea why!) 
I'd much prefer to be using Betaflight 4.3.1 but there isn't a specific build for my F7 twing FC with DJI OSD warnings on it.

I get the devs arguments about not dropping standards, but come on guys, there are enough DJI users out there that think this situation really sucks now.
 
PLEASE allow a merge or something so that howels isn't continually getting badgered by people like me who would love to use the latest version of betaflight but can't!

howels, 2022-07-30:
I can do a build, I'll try to get it up soon.

sugaarK, 2022-07-30:
> I've just rebuilt my Titan DC5 and have had to put INAV back on it because of the lack of warnings in DJI OSD. (Went all the way to the field yesterday only to have it not arm but I've no frickin' [...]

Easy man there are 50 members of the core bf you buy 25 of them a set of goggles and 2 air units then we can come up with an elegant solution to this problem…. We can test what we don’t have hardware for and I don’t see dji helping any of us out… instead of stamping your feet help get equipment for the team to work with..

a1j, 2022-07-30:
The hacks are out there tested by many https://github.com/howels/betaflight/releases , DJI is never going to help since they discontinuing V2 line. both inav and emu have this implemented for ages and patches can be backported probably.

chilternflyer, 2022-07-30:
> I can do a build, I'll try to get it up soon.
> 
> I am also intending to do a PR to incorporate it into a CLI option but life has gotten in the way recently.

Wow - Thank you!
I've used one of your builds on my Chimera 4 and it works great. So getting the same on the Titan would be awesome.
for reference:- The Titan FC identifies as itself as  EXF722DUAL

chilternflyer, 2022-07-30:
> > I've just rebuilt my Titan DC5 and have had to put INAV back on it because of the lack of warnings in DJI OSD. (Went all the way to the field yesterday only to have it not arm but I've no frickin' idea why!) I'd much prefer to be using Betaflight 4.3.1 but there isn't a specific build for my F7 twing FC with DJI OSD warnings on it.
> > I get the devs arguments about not dropping standards, but come on guys, there are enough DJI users out there that think this situation really sucks now.
> > PLEASE allow a merge or something so that howels isn't continually getting badgered by people like me who would love to use the latest version of betaflight but can't!
> 
> Easy man there are 50 members of the core bf you buy 25 of them a set of goggles and 2 air units then we can come up with an elegant solution to this problem…. We can’t test what we don’t have hardware for and I don’t see dji helping any of us out… instead of stamping your feet help get equipment for the team to work with..

Sorry for the rant.. I've been having a bad time recently with too many different firmwares and too many things to remember/ no working. It seemed an easy fix from the outside ;-)

sugaarK, 2022-07-30:
Look we have 7 people who can merge code who are active and none of them have dji.. 4 are needed to merge stuff. See the issue? It’s expensive kit that none of the core devs own. We are taking to snail so we hill get gear and we will work with them.. that is kinda how it works.. it’s frankly not that Kool
To buy stuff from bad companies and expect people in open source teams to provide support for bad equipped in their spare time for free

chilternflyer, 2022-07-30:
Don't get me wrong. I completely get that the devs are all working for the love of the hobby and I'm not in any position to demand anything. That wasn't my issue. I wish I hadn't asked the question now.

howels, 2022-07-30:
@chilternflyer I've also been testing MSP displayport on DJI via https://github.com/bri3d/msp-osd#easy-installation recently. They have a full web-based rooting, installation and configuration utility now. Better long term solution than praying to DJI to update a product they haven't updated in years.
I'll look at getting new 4.3.1 builds up soon but the full OSD is a huge improvement.

sugaarK, 2022-07-31:
> The hacks are out there tested by many https://github.com/howels/betaflight/releases , DJI is never going to help since they discontinuing V2 line. both inav and emu have this implemented for ages and patches can be backported probably.

Go take a good look at emus code base it’s a total mess and what Inav does is not Betaflights problem.  Don’t compare us to them because what the do or don’t do is not relevant or important to Betaflight

Ale758122, 2024-09-28:
I attach videos of the viewers https://www.youtube.com/watch?v=DMxT9nvyqKI

SteveCEvans, 2024-09-28:
Please update to the latest version which addresses this.

closed 2026-09-26 by blckmn