betaflight/betaflight #11560: problems with SPI FR

ALFJAVIVI, 2022-04-26:
### Describe the bug

With RC4, RC5 and RC6 RC problems with Spi Fr.
Rxloxx in Rc4 and RC5 at distances of 10 to 15m and with RC6 the distance is about 25m for an hour with which the RC works best is in RC3.
The FC is a crazybeex FR versions v1 and v2

### To Reproduce

that the rc signal is coo in 4.3 Rc3

### Expected behavior

that the range of the RC signal is coo Rc3

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

crazybeex FR

daleckystepan, 2022-04-26:
Can you please send us log with RC6 firmware, debug mode set to `rx_timing`?

ALFJAVIVI, 2022-04-27:
Good morning this fc crazybeex FR does not have a black box

bike797, 2022-04-28:
my quad(SPI FR) is the same. Short range.

daleckystepan, 2022-04-28:
Can you please paste your `diff`?

bike797, 2022-04-28:
`diff `

```
# 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.3.0 Apr 25 2022 / 01:03:20 (9d71184) MSP API: 1.44
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FR, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4FR
manufacturer_id HAMO
mcu_id 004500683339510839363632
signature 

# name: Mobula6

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature MOTOR_STOP
feature TELEMETRY

# serial
serial 1 2048 115200 57600 0 115200

# map
map TAER1234

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1700 2100 0 0
aux 4 28 1 1300 2100 0 0
aux 5 35 2 1300 2100 0 0
aux 6 35 3 1300 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 2 1 0
vtxtable powerlabels 25 200 100 25

# master
set gyro_lpf1_static_hz = 350
set gyro_lpf2_static_hz = 700
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 350
set gyro_lpf1_dyn_max_hz = 700
set acc_calibration = 39,-76,-197,1
set mag_hardware = NONE
set baro_hardware = NONE
set fpv_mix_degrees = 35
set blackbox_device = NONE
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set failsafe_delay = 4
set failsafe_recovery_delay = 20
set ibata_scale = 1175
set yaw_motors_reversed = ON
set small_angle = 180
set simplified_gyro_filter_multiplier = 140
set report_cell_voltage = ON
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 30
set osd_rssi_dbm_alarm = -95
set osd_cap_alarm = 100
set osd_vbat_pos = 2456
set osd_rssi_pos = 2145
set osd_link_quality_pos = 161
set osd_rssi_dbm_pos = 128
set osd_tim_1_pos = 33
set osd_tim_2_pos = 2081
set osd_flymode_pos = 2401
set osd_throttle_pos = 2361
set osd_vtx_channel_pos = 2099
set osd_current_pos = 2423
set osd_mah_drawn_pos = 2392
set osd_craft_name_pos = 2433
set osd_debug_pos = 65
set osd_pidrate_profile_pos = 2330
set osd_warnings_pos = 14441
set osd_avg_cell_voltage_pos = 408
set osd_battery_usage_pos = 65
set osd_disarmed_pos = 2442
set osd_esc_tmp_pos = 2093
set osd_esc_rpm_pos = 257
set osd_rtc_date_time_pos = 73
set osd_flip_arrow_pos = 2349
set osd_core_temp_pos = 2088
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set debug_mode = FFT
set cpu_overclock = 120MHZ
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5880
set vcd_video_system = NTSC
set frsky_spi_tx_id = 35,43,2
set frsky_spi_offset = -17
set frsky_spi_bind_hop_data = 0,43,86,129,172,215,23,66,109,152,195,3,46,89,132,175,218,26,69,112,155,198,6,49,92,135,178,221,29,72,115,158,201,9,52,95,138,181,224,32,75,118,161,204,12,55,98,0,0,0
set gyro_1_align_yaw = 900
set name = Mobula6

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 105
set dterm_lpf1_dyn_max_hz = 210
set dterm_lpf1_static_hz = 105
set dterm_lpf2_static_hz = 210
set anti_gravity_gain = 5000
set iterm_relax = RPY
set pidsum_limit = 1000
set p_pitch = 94
set i_pitch = 5
set d_pitch = 48
set f_pitch = 250
set p_roll = 90
set i_roll = 160
set d_roll = 80
set f_roll = 240
set p_yaw = 43
set i_yaw = 94
set f_yaw = 240
set angle_level_strength = 0
set horizon_level_strength = 0
set horizon_transition = 0
set d_min_roll = 60
set d_min_pitch = 0
set dyn_idle_min_rpm = 70
set simplified_master_multiplier = 200
set simplified_dterm_filter_multiplier = 140

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 15
set pitch_rc_rate = 15
set yaw_rc_rate = 15
set yaw_srate = 75
set roll_level_expo = 5
set pitch_level_expo = 5

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

mitcht, 2022-04-30:
This is "me too", but I can say I've just compiled a hex using master with target CrazyBeeF4FR as of the date of this post.  I can confirm dropout at 10m when using Frsky D 8 (Moblite7). I've tried overcloxk, normal clock, dshot_bitbang = true, less filters, attempt to lower cpu usage, tried forcing FrskyX (no luck) and just about every combination of things you could think of.  No luck.  Was fine on 4.2.  I'll be available to build specific branches if needed. No need to compile a hex for me.  I can also help code, if you can help me dial into the correct place to make adjustments!  That sounds silly but I'm not super familiar with the structure of the project yet.

daleckystepan, 2022-05-01:
Thank you for data. I will retest with your settings.

daleckystepan, 2022-05-01:
[image]

Look at `debug[0]` peaks. Each peak means connection issue. It is nearly unusable on D8.

Professor77, 2022-05-02:
I have the same issue with my Hamo Crazybeef4fr
Radio is frsky x-lite on D8. Have not tried it with Frsky X or LBT.

daleckystepan, 2022-05-02:
Please retest with X2 mode. It is still not ideal but it seems to be better.

joeyw007, 2022-05-02:
This is an odd duck of an issue. I had failsafe, not range issues with the following boards after fresh installing RC3, RC4 or RC5. The CrazybeeX FR v2.2, CrazybeeX FR v1.0, Crazybee F4 Lite, CrazyF411, CrazyF411 Pro, Nanohawk, Nanohawk X, Tinyhawk II FS, Tinyhawk II Race and Tinyhawk III boards. PR 11536 was hammered out, tested and merged which fixed the failsafe issue but apparently there is still some sort of issue although I am not experiencing it nor loss of range. In fact D8 has never worked better after merging PR 11536 with master for RC6 release. Also everything was fine and functional in RC3 as well as 4.2.11. Quite a contradiction. I am using a T12 Pro Hall running OpenTX 2.3.14. All tests, range testing and flight testing, were done with factory reset radio with new model that was frequency tuned. The quads were freshly configured, no diffs or dumps carried from one RC to the next. Configurations were left as default as possible obviously some settings need to be changed to get into the air but only personalization and VTX settings. As of the last several days I've been flying RC6 on all these boards quite frequently with no issues at all. It appears there is a sort of anomaly causing issue.

ALFJAVIVI, 2022-05-03:
So I understand that from RC6 to D8 the RC problems have been solved?
At the moment I am going in RC3 and D16 and for the moment everything
is fine to one that I do notice that the RSSI of the betaflight OSD is
not very stable since it goes up and down very fast to a point where I
have not had any RXLOXX moving away more than 100 meters has never
dropped the RSSI below 35.

joeyw007, 2022-05-03:
I ran debug mode on RX_TIMING and got this but I'm not having any issues at all
[image]
.

mitcht, 2022-05-03:
Just how far should range for a CrazyBeeF4FR quad?  I was seeing 15-20m tops on two separate boards.  I just installed rc6 on one of them last night.  If 20m is "what to be expected"... well...

joeyw007, 2022-05-04:
I’m not sure exactly what the deal is. I was getting regular failsafes after RC3 in RC4 and RC5 on the CrazybeeX FR v2.2, Crazybee F4 Lite and the NanohawkX MatekF411 board. I mentioned it in Slack and one of the Devs was able to recreate the issue repeatably and said he found an issue. He went to work on some code, see PR11536. What he did fixed the failsafe issues for me. I’m assuming others must have been having this issue as well for a PR to be created. I’m just a beta tester. There were no issues with range at the time at least for me, either. D8 works great in RC6. Now, week or so later a good hand full of people are saying they are having range issues running D8 in RC6. I wonder if they had issues with RC4 or RC5, failsafe or range? Never the less there seems to be an issue that’s very intermittent. Limon on Slack has the BetaFPV 1s AIO for the Meteor 65 and was the first to raise the range issue concerning that board. I purchased the same board, flashed up to RC6 fresh configured it leaving everything as default as possible just the config changes to get in the air. I used his test methodology and could not recreate the range issue. I installed the board in a whoop and have been ripping it around for days. Long story short curtain people are being adversley effected by PR111536 and others are fine and from what I’ve observed there is no rhyme or reason as to who or what board, even of the same make and model, is having an issue with range or isn’t. That’s all I know man.

mitcht, 2022-05-04:
I have even built master with the CrazyBeeF4FR target.  Same thing happens.

joeyw007, 2022-05-04:
Oh and to answer your question about range. Running RC6 with my radio in range test mode I’m getting 1 meter more consistently over RC3 or 4.2.11. The distance while the radio is in range test mode is around 15m and that’s through 3 walls. Outside, depending on conditions, range is much better considering it’s a short range protocol. I haven’t walked it off and measured it. I use D8 almost exclusively and I can go places and farther than any previous release running when RC6.

mitcht, 2022-05-04:
Completely forgot to mention I'm seeing 15m _outdoors_.

joeyw007, 2022-05-04:
Given some of the complaints I’ve seen in slack and on GitHub it appears there is an issue. I have no idea what the sample size is or if any of the devs are able to recreate these issues. I know that much won’t happen if they can’t. Is that 15m in range test mode or normal flight?

mitcht, 2022-05-04:
Normal flight.  I could possibly tackle debugging and dev work (I am a software engineer heh) but I'm not really familiar with the Betaflight project.  I don't even know what files I'd want to target my changes on! Lots to learn.

Professor77, 2022-05-04:
I still havent tried my board on frsky X. I didnt update my radio to Accst
2.0 so I don't have X2 to test.

After moving most of my models to ELRS 900 there's just 1 other drone that
is still using Accst 1.0 rx so I might flash that one.

Anyway could it be an ACCST 1.0 issue?

The only other thing I can think of is that I replaced my SPI rx antenna
with a homemade one about a year ago after it got chopped up. I still had
great range with that tho. 200m no problem.

joeyw007, 2022-05-04:
I don’t think it’s your antenna as long as it is the exact same length as the one that was chopped and the wire is the same impedence or very close. Basically use the kind of wire as was on it. But if it was fine before it should be fine now. FrSky_X is fine as far as I can see and no one is complaining of issues with it. Redpine even works great for people having D8 range issues. It only seems to be something in the D8 protocol and the way it interfaces with BF 4.3s new scheduler but pr 11536 supposedly fixed that. On top of all that like I said people who have the same board are getting different results so until one of the devs can reliably recreate the issue, create a PR, find, fix, test and verify it will persist. It seems to be a very small sample size of people having this issue. One of the reasons I wanted to know what radio you were running, what firmware if applicable and what version because I’m trying to nail it down to an issue with the way the radio is communicating with BF. I’m running OpenTx 2.3.14 and not having issues. I did have issues with EdgeTx but they were not range issues. I think EdgeTx development is so fast that they are leaving older protocols to the wayside. If you don’t mind telling me what firmware you are running and what version, I’ll test with it and see if I can’t replicate the issue.

daleckystepan, 2022-05-04:
Thank all of you that you are willing to help. Do you have any SPI RX board with blackbox or Openlager? It would be beneficial to have some logs from failsafed flight.

To clarify what's is going on and what changes have been made according to my understanding.
1) BF 4.2.x - baseline, default
2) BF 4.3 RC3 hopefully the same as previous
3) ELRS SPI changes that affected also FrSky.
4) Failsafe changes: failsafe detection, guard times, ...
5) BF 4.3 RC6 SPI should be fixed, failsafe guard intervals should match behavior of 4.2.x more closely

So there are at least two aspects:
1) FrSky SPI code that is already fixed. It affects FrSky RF chip directly.
2) Failsafe tuning, especially guard intervals
3) One bonus: FrSky is not willing to help in any way. Everything we know about FrSky is hard work of BF devs and reverse engineering.

Regarding D8 mode: It has big frame loss in 4.2.x and also in 4.3. I would recommend to not using it. Try to switch to X2.

*TLDR*: Rx loss can be caused because of FrSky driver issue or because of too sensitive failsafe settings. Try X2 mode.

HW
- CrazybeeF4 with BF 4.3 latest
- TX16S with the lastest EdgeTx 2.8
- MPM module with 1.3.3.7 firmware

mitcht, 2022-05-04:
> Thank all of you that you are willing to help. Do you have any SPI RX board with blackbox or Openlager? It would be beneficial to have some logs from failsafed flight.
...
I keep threatening to buy an openlager.  Maybe its time.

SteveCEvans, 2022-05-04:
With a bit of creativity one can add flash to support black box…

[image]

mitcht, 2022-05-04:
I'm pretty sure I can't solder any chips on my board https://www.racedayquads.com/products/happymodel-diamondf4-aio-whoop-flight-controller-f4-fc-5a-1-2s-4in1-esc-25-200mw-vtx-choose-your-rx

SteveCEvans, 2022-05-04:
You need access to SPI, which is fairly easy to do on the MAX7456, and a chip select line. Then declare the FLASH to be on the same bus as the OSD. Not quite as simple as it sounds, but doable.

Professor77, 2022-05-05:
After texting on Frsky X ACCST 1.0 there were no short range failsafes,
which is great. However I was seeing constant RX LOSS warnings even with an
RSSI value of 60~70.

RX LOSS warnings did not interfere with smooth flight, but when I tried
constant stick movements, weaving and quick turns there were a lot of bumps
and jerks, like my quad was hitting ghost branches, leaves or grass.

I do not know a lot about rc smoothing, but it seems like I need to adjust
the settings, and look at some sensor charts.

Does anyone have tips for smoothing out the rc link?

daleckystepan, 2022-05-05:
@Professor77  You can see RXLOSS in OSD after 100ms without RC frame. So it is roughly 11 frames in FrSky with 111Hz refresh rate.

Reason for bumps after RXLOSS
1) RC command is not smoothed enough because it is not designed to such a long RXLOSS
2) The same applies for FeedForward. FF has probably even bigger impact on bumps than RC smoothing itself.

How to solve it?
* Don't increase RC smoothing and FF smoothing much. If you do so you will get very big latency soon.
* Proper solution: Fix FrSky ... that's is the hardest part of the game.

mitcht, 2022-05-05:
> 3. One bonus: FrSky is not willing to help in any way. Everything we know about FrSky is hard work of BF devs and reverse engineering.

_What is this all about?_

joeyw007, 2022-05-05:
@mitcht FrSky refuses to work with anyone. They are very closed source company. Who really knows why? Because of this BFs ability to implement FrSky protocols is only through developers reverse engineering these protocols with a lot of time and effort. Hope this answers your question

mitcht, 2022-05-05:
I would help, but I just blew my drone budget on new frames and other things. I'll have to wait a bit to order openlager. Once I get one I'll debug.

joeyw007, 2022-05-05:
I'll gather data do you guys (devs) think the best place to drop it off would be in slack or here. I'll start where I originally had an issue with failsafe in RC4 before PR11536 fixed it and maybe it'll shine some light on something. I'll get started as soon as the device arrives and I wire it up and learn blackbox. Shouldn't take me long, prolly longer to ship than anything else.

Professor77, 2022-05-06:
I am a filthy casual with no coding ability but I can try to log my rx
packet loss if someone explains how to do it.

I can find the sensor tab under expert mode, but I'm not sure how to set up
the log to show the rx information.

haslinghuis, 2022-05-06:
`set debug_mode = RX_SIGNAL_LOSS`

[image]

joeyw007, 2022-05-06:
I ran debug mode on RX_SIGNAL_LOST and this is what I got. I'm not having issues with D8 at all in RC6. I was curious as to whether or not this looks correct. I did move the throttle stick around a bit and got some tracing on debug 3 as expected.
[image]
?

SteveCEvans, 2022-05-06:
@joeyw007 Is this still with OpenTX 2.3.14 on a Jumper T12? Is that using the Jumper JP4-in-1 module? What version of firmware is that running?

it would be interesting to flash a multi-protocol module with the same version as you’re using to see if that or the OpenTx version makes a difference.

joeyw007, 2022-05-06:
Yes I'm running a Jumper T12 Pro Hall with internal JP 4 in 1 module v1.3.3.7. It's running OpenTx 2.3.14. I'm curious of the same thing. I have suspicions about EdgeTx 2.6 up but have not had time test. Will do soon.

SteveCEvans, 2022-05-06:
So the module is up to date and the same version used by others reporting issues. Updating to EdgeTx to see if that breaks things would be informative.

joeyw007, 2022-05-06:
@SteveCEvans 1st chance I get I'll give it a whirl. Any particular version of EdgeTx you want me to try?

joeyw007, 2022-05-06:
@SteveCEvans Well I guess that squashed any suspicions I had. I upgraded T12 Pro from OpenTx 2.3.14 to EdgeTx 2.7. The Range test yielded Identical results, the debug traces were identical in debug mode for RX_SIGNAL_LOST, RX_TIMING and actual flight tests were identical on 3 quads each with a different F411 D8 SPI boards, the same boards I have been using for testing. NanohawkX Matek F411 board, CrazybeeX FR v2.2, and Crazy F411 Pro v3.1. Three 5 min flights each. That was the last discrepancy that I had to test and it passed with flying colors.

mitcht, 2022-05-06:
I'm the worst. Still experiencing it. Cannot log.  ¯\_(ツ)_/¯ sorry

Professor77, 2022-05-06:
I am using x-lite on opentx 2.3.14 Frsky X in range test mode. Crazybee
3.0 on 4.3 RC6. I attached screen shots from various debug mode tests.

mitcht, 2022-05-09:
Sadly I'm bringing all my quads back to 4.2.x  ... this new version is just too buggy / different / untenable for me.  I can't get my whoops to fly right for anything. I almost had a mob6 flyaway last night. My Roma had crunchy sounding motors.  Idk. Good luck.

closed 2022-06-16 by github-actions[bot]

joeyw007, 2022-10-11:
I just contribute to the project by testing nightlies, releases, reporting results and if the devs can reproduce what I’m seeing they act. I have a very limited knowledge of code although I am a serious study of BF, can make it sing and work hard with the devs to help make it better. What radio are you using, what OS, what version?

ALFJAVIVI, 2022-10-11:
Putting the 4.3 rc3 I have not had problems