betaflight/betaflight #11562: problems with crsf 6.17 and rc5 and 6 rx loss within 100yards when never had that issue with rc4

paulfrmn74, 2022-04-27:
### Describe the bug

all my gear is up to date and flying in a spot i have had no issues until rc 5 and 6 crsf  rx loss. i have even tried 1w and still would get the rssi 1 start dropping then rx loss within 100 yrds. rc4 didnt have that problem for crsf. i also use frsky r9 and r9 is fine. no failsafes. something is going on with the crsf and rc 5 and 6

### To Reproduce

just fly with crsf and watch the link

### Expected behavior

rx loss

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

all fc should have that problem

klutvott123, 2022-04-27:
Please provide the information that was requested when opening the issue

Joerg-H-Wagner, 2022-05-01:
Hello. This does not belong to BF. Had the same issue on one of my quads. Depending on the distance, every second. Regardless if BF RC4,5 or 6. This procedure helped in my case: Open TBS Agent X or Lite on Radio. Go to Radio Settings and set Operational Mode from "Normal" to "Force Telemetry". If your receiver looses binding (LED is still red) even the radio is powered on, go again to the Settings and set it back to "Normal". After this switching the telemetry worked properly in my case.

dipanfilo, 2022-05-14:
Hello everyone,
I'm flying my drone whit the last BF4.3-RC6 and the last crsf fw. My drone still Crashing every time when i try to make Sharp turns or Flying for some minutes. The radio don't tell my any signal lost or telemetry lost after the crash the green binding leds still on, but the flight control can't get any commands from the drone. The only way to restart the drone is to cat the battery and connect again. Also the DJI Goggles still recoding in the goggles after crash. Never happen before something like this🥺 
How i can help to fix this problem? 
I try already to make factory reset from the receive, transmitter and radio. 
But The problem is still there

Thanks for every help ☺️
And I'm great full to contribute to find to solve the problem ☺️

closed 2022-06-22 by github-actions[bot]