betaflight/betaflight #11570: GPS Rescue Current altitude + initial altitude summed

PaulFPV, 2022-04-29:
### Is your feature request related to a problem? Please describe

I haven't personally tested but, if I understood correctly, when GPS rescue altitude mode is set to "current altitude" if rescue is triggered the altitude will stay the same and the drone will go back to home.... to me it seems more logical to sum to the actual altitude the "initial altitude value"... and then go back home... there is a good probability that while raising the signal comes back... and there is even more probabilities to avoid trees or other obstacles in the way home.

Example (initial altitude set to +20m, altitude mode:"current altitude")

I fly to 100m altitude, when GPS rescue kicks in the altitude will raise by 20m and then it starts going to home direction.

This will be my to go choice when flying around mountains:

-max altitude could use a lot of battery to raise again to the top of the mountain
-fixed altitude when surfing mountain doesn't make sense... 
-current altitude + an amount is the perfect choice

### Describe the solution you'd like

I'd like to see the initial altitude summed to current altitude by default (if you want the current altitude like before just write "0" as initial altitude.

### Describe alternatives you've considered

.

PaulFPV, 2022-05-03:
In "gps_rescue.c" I found that this value: "rescueAltitudeBufferM = 15" is added to the current altitude, so, if I understand correctly there is already a 15m added to the current altitude... but it's a value not available in the configurator.

So it could be made visible in the configurator (but I think it would be confusing to have this buffer value around).. or we could do what I proposed before by changing this:

            case CURRENT_ALT:
                newAltitude = rescueState.sensor.currentAltitudeCm + gpsRescueConfig()->rescueAltitudeBufferM * 100;
                break;
 
to this:

            case CURRENT_ALT:
                newAltitude = rescueState.sensor.currentAltitudeCm + gpsRescueConfig()->initialAltitudeM * 100;
                break;

or.. if we want to keep the added buffer.. to this:

            case CURRENT_ALT:
                newAltitude = rescueState.sensor.currentAltitudeCm + gpsRescueConfig()->initialAltitudeM * 100 + gpsRescueConfig()->rescueAltitudeBufferM * 100;
                break;

...I'm I right?

ctzsnooze, 2022-05-16:
@PaulFPV
The `Current Altitude` option already does what you want.  Or should :-)
In that mode, the target altitude on enabling GPS rescue will be your current altitude *plus* the value you put into the CLI as your `gps_rescue_alt_buffer` in metres.  This value is re-named to `rescueAltitudeBufferM` in the code.
Just be sure to not failsafe behind an object taller than that buffer, and recognise that the system is not perfect so give yourself some leeway.
https://github.com/betaflight/betaflight/pull/11579 doesn't change this behaviour, but I will be setting an absolute minimum of 5m above home altitude for any return, mostly for people who take off above their intended flightpath and accidentally use `Current_Altitude` mode without enough climb altitude.
If that makes sense, you can close this issue.

PaulFPV, 2022-05-16:
@ctzsnooze

Thanks for your big job on: Update GPS Rescue.

I understood that using gps_rescue_alt_buffer will do the job, but this way this value won't be a buffer anymore (it should be there as a.... buffer... to compensate precision issues).

What I propose is that even "current altitude mode" will be adding the initial altitude value, exactly as "fixed altitude" and "maximum altitude mode" already do (max alt mode uses buffer too... I don't get why fixed altitude mode don't uses it).

And this way, every GPS rescue mode will make use of "initial altitude" the same way... why just a mode should be setup differently(CLI)? When I switch between current and max altitude I'll have to keep changing the buffer value, which doesn't make much sense to me.

To make it clearer->

currently:
-fixed altitude (initial value, no buffer added)
-current altitude (current altitude + buffer)
-maximum altitude (max altitude + initial value + buffer)

after:
-fixed altitude (initial value + buffer)
-current altitude (current altitude + initial value + buffer)
-maximum altitude (max altitude + initial value + buffer)

"I will be setting an absolute minimum of 5m above home altitude for any return".... isn't this already the role of the gps_rescue_alt_buffer?

closed 2022-06-23 by github-actions[bot]