betaflight/betaflight #11572: MATEKF405CTR identifies as legacy target MATEKF405

fallingcats, 2022-04-30:
### Describe the bug

MATEKF405CTR identifies as legacy target MATEKF405 even on newest firmware

### To Reproduce

Flash MATEKF405CTR board with the newest firmware (4.3-RC6) for that board.
Open "Update Firmware" in BF Configurator with a MATEKF405CTR board connected, it will still select the old MATEKF405 target that isn't supported anymore.

### Expected behavior

"Update Firmware" should select the MATEKF405CTR firmware

### Flight controller configuration

```
# version
# Betaflight / STM32F405 (S405) 4.3.0 Apr 25 2022 / 01:02:32 (9d71184) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF405CTR, version: b020e572, date: 2022-01-16T20:04:31Z

# start the command batch
batch start

board_name MATEKF405
manufacturer_id MTKS

# feature
feature -OSD
feature LED_STRIP

# beeper
beeper -SYSTEM_INIT

# aux
aux 0 0 2 1700 2100 0 0
aux 1 2 0 900 1700 0 0
aux 2 6 0 1300 1700 0 0
aux 3 27 3 1700 2100 0 0
aux 5 4 1 1700 2100 0 0

# master
set acc_calibration = 24,52,-11,1
set align_mag = CW180
set mag_align_yaw = 1800
set mag_hardware = AUTO
set min_check = 1000
set max_check = 2000
set min_throttle = 1145
set max_throttle = 1830
set motor_pwm_protocol = MULTISHOT
set bat_capacity = 1200
set vbat_warning_cell_voltage = 360
set deadband = 2
set yaw_deadband = 50
set gyro_1_align_yaw = 2700

profile 1

rateprofile 0

# end the command batch
batch end

```
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: LED 2
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: LED_STRIP
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: PREINIT 3
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: SDCARD_CS
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: ADC_CURR
C05: ADC_BATT
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: FREE
C11: FREE
C12: FREE
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1 : MOTOR 1
TIM4:
    CH1 : LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH2 : MOTOR 2
    CH3 : MOTOR 3
    CH4 : MOTOR 4
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: SPI_MISO 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: SPI_MOSI 3
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

### Flight controller

MATEKSYS F405-CTR

haslinghuis, 2022-04-30:
Did you use BFC 10.8RC6? Also detect just reads the current `board_info` on the board.

fallingcats, 2022-05-02:
I've used the nightly `v20220428.104-1` which calls itselft 10.0.8.0-debug fa66d2

fallingcats, 2022-05-02:
`v20220428.104-1` seems not not have this issue, thanks

closed 2022-05-02 by fallingcats