betaflight/betaflight #11575: some firmware setting values of Betaflight 4.3.0-rc? may no longer be compatible with unified targets

pitts-mo, 2022-05-01:
### Describe the bug

Some of the allowed values firmware 4.3.0-rc? settings are no longer compatible with some of the settings found within unified target configs.

Should compatibility for established values be re-added to the firmware OR should we fork unified targets and update them for use with 4.3.X?

### To Reproduce

My test case:
-Revolt board
-loaded STM32F405 Firmware

(https://github.com/betaflight/unified-targets/blob/master/configs/default/AIRB-OMNIBUSF4.config)
**OMNIBUSF4 unified target config load in CLI results in this error:**
# set max7456_clock = FULL
###ERROR IN set: INVALID VALUE###
Allowed values: HALF, NOMINAL, DOUBLE

(https://github.com/betaflight/unified-targets/blob/master/configs/default/GEPR-GEPRCF411_AIO.config)
**GEPRCF411_AIO  unified target config load in CLI results in these errors:**
# board_name GEPRCF411_AIO
###ERROR IN board_name: BOARD_NAME CANNOT BE CHANGED. CURRENT VALUE: 'OMNIBUSF4'###

# manufacturer_id GEPR
###ERROR IN manufacturer_id: MANUFACTURER_ID CANNOT BE CHANGED. CURRENT VALUE: 'AIRB'###

### Expected behavior

load defaults for unified target without error

### Flight controller configuration

```
none
```
```
defaults
```

### Flight controller

Revolt w/STM32F405 firmware

### Other components

Attempt to load unified targets files for OMINBUSF4 and GEPRCF411_AIO (I know different CPU).

### How are the different components wired up

NA

### Add any other context about the problem that you think might be relevant here

NA

haslinghuis, 2022-05-01:
Seems we need new config files for the `max7456_clock` setting or revert the name change if it was done in 4.3
```
configs\default/AIRB-OMNIBUSF4.config:123:set max7456_clock = FULL
configs\default/HENA-TALONF4V2.config:92:set max7456_clock = DEFAULT
configs\default/HGLR-HGLRCF405.config:133:set max7456_clock = DEFAULT
configs\default/HGLR-HGLRCF405AS.config:123:set max7456_clock = FULL
configs\default/HGLR-HGLRCF745.config:130:set max7456_clock = DEFAULT
configs\default/VIVA-VIVAF4AIO.config:102:set max7456_clock = DEFAULT
```

The second part looks like using the wrong target as I didn't verify which target to use but there are two REVOLT legacy targets. Not sure if they work as I don't have the board.

pitts-mo, 2022-05-01:
Oops, I notice the error in my alternate GEPRCF411_AIO test is wrong/different than what I see on my actual GEPRCF411_AIO board.

corrected issue description with this info:

(https://github.com/betaflight/unified-targets/blob/master/configs/default/GEPR-GEPRCF411_AIO.config)
**GEPRCF411_AIO  unified target config load in CLI results in these errors:**
# set dashboard_i2c_bus = 1
###ERROR IN set: INVALID NAME###

haslinghuis, 2022-05-02:
Dashboard is only available on targets > 512K. So more target changes needed here.

pitts-mo, 2022-05-03:
Thank you @haslinghuis

I renamed this issue and think we should leave this open in case another setting/value causes this.

potential unified targets issues/questions:
-script to check unified target settings and values are valid in firmware source?
-maintain different branches of unified targets to maintain compatibility with betaflight versions?

haslinghuis, 2022-05-03:
Thank you @pitts-mo for reporting.
- A script would certainly come in handy here.
- Maintaining multiple branches may be a solution for future development, but currently the impact is small.

closed 2022-05-06 by blckmn