betaflight/betaflight #11577: Heavy D-Term oscillations with BF4.3 RC6 and Diatone Mamba F55 128K BL32 55A ESC

Joerg-H-Wagner, 2022-05-01:
### Describe the bug

PID and Filtering are all set to default which is 
perfect for my quad up to BF4.3 RC5.
In RC6 the D-Term shows spikes up to 450deg/s in Pitch
and 63deg/s in Roll. Or in BBE: oscillations from -34% to 37%.
The motor outputs have an overlay of the D-Term.
First I thought a damage of one motor during the flight.
But all motors are driven in the same way. 
- Regardless if fixed 48kHz PWM or  "by RPM"
- Regardless if 16 degree or 23 degree motor angle

A second quad with same FC eqipped with Speedybee ESC 
does not have this issue.

### To Reproduce

Use Betfaflight 4.3 RC6 with stock tune PID+Filtering.
Use Diatone Mamba F55 128K BL32 55A ESC with latest 32.9.
[Uploading btfl_001.bbl.csv…]()

### Expected behavior

Low D-Term.

### Flight controller configuration

```
  diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Apr 15 2022 / 10:17:24 (9360ab1) MSP API: 1.44
# config: manufacturer_id: SPBE, board_name: SPEEDYBEEF7V2, version: 55694498, date: 2021-05-15T13:56:51Z

# start the command batch
batch start

board_name SPEEDYBEEF7V2
manufacturer_id SPBE

# name: FIVE33

# feature
feature GPS
feature TELEMETRY

# serial
serial 1 1 115200 57600 0 115200
serial 4 128 115200 57600 0 2000000
serial 5 2 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 900 1300 0 0
aux 2 2 2 1300 1700 0 0
aux 3 13 3 1700 2100 0 0
aux 4 26 3 1300 1700 0 0

# adjrange
adjrange 0 0 3 900 2100 12 2 0 0

# master
set gyro_lpf2_static_hz = 0
set dyn_notch_count = 2
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 800
set gyro_lpf1_dyn_min_hz = 120
set gyro_lpf1_dyn_max_hz = 350
set acc_calibration = 24,9,130,1
set mag_i2c_address = 13
set rc_smoothing_auto_factor = 20
set rc_smoothing_setpoint_cutoff = 25
set rc_smoothing_feedforward_cutoff = 25
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_switch_mode = STAGE2
set failsafe_procedure = AUTO-LAND
set ibata_scale = 80
set yaw_motors_reversed = ON
set gps_provider = UBLOX
set gps_ublox_use_galileo = ON
set simplified_gyro_filter = OFF
set osd_warn_link_quality = ON
set osd_vbat_pos = 2542
set osd_link_quality_pos = 2536
set osd_throttle_pos = 462
set osd_crosshairs_pos = 2318
set osd_gps_speed_pos = 2479
set osd_altitude_pos = 2535
set osd_warnings_pos = 14539
set osd_avg_cell_voltage_pos = 2546
set osd_pit_ang_pos = 2510
set debug_mode = GYRO_SCALED
set name = FIVE33

profile 1

rateprofile 0

# rateprofile 0
set thr_expo = 50
set roll_rc_rate = 5
set pitch_rc_rate = 5
set yaw_rc_rate = 5
set roll_expo = 30
set pitch_expo = 30
set roll_srate = 47
set pitch_srate = 47
set throttle_limit_type = SCALE
set throttle_limit_percent = 80

# end the command batch
batch end

# 
```
```
  resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 1
A09: MOTOR 2
A10: MOTOR 3
A11: USB
A12: USB
A13: SWD
A14: FLASH_CS
A15: PINIO 1
B00: LED_STRIP
B01: FREE
B02: BEEPER
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: SERIAL_TX 1
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: MOTOR 4
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : DSHOT_BITBANG 1
    CH3 : DSHOT_BITBANG 3
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1 : CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH2N: LED_STRIP
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: DSHOT_BITBANG 1
DMA2 Stream 2: LED_STRIP
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: DSHOT_BITBANG 3
DMA2 Stream 7: FREE

# 
```

### Flight controller

Speedybee F7 V2

### Other components

Diatone Mamba F55 128K BL32 55A 4-in-1 ESC Version 32.9
T-Motor F80 1950KV

Joerg-H-Wagner, 2022-05-02:
[btfl_004.bbl.csv](https://github.com/betaflight/betaflight/files/8605418/btfl_004.bbl.csv)
This log is recorded with Five33 Switchback HD, Ovonic 1600mAh 120C 
and T-Motor 5150 3-blade, 660g total weight.

Joerg-H-Wagner, 2022-05-03:
Downgraded to RC4, nothing else changed. 
Copter was kicked 10 Meters high and I disarmed immediately. At the landing an arm broke. 
[btfl_005.bbl.csv](https://github.com/betaflight/betaflight/files/8613751/btfl_005.bbl.csv)
At the bench I checked motor directions and armed without the props. Everything was correct.
I will downgrade to 4.2.11. To be a test pilot is too expensive for me.

klutvott123, 2022-05-03:
I can't view your log files

Joerg-H-Wagner, 2022-05-03:
To be able to upload files I added the suffix .txt. Remove .txt after Download to open it PID Toolbox or BBE.

RC6:
[btfl_001.bbl.txt](https://github.com/betaflight/betaflight/files/8614183/btfl_001.bbl.txt)

RC4:
[btfl_001.bbl.txt](https://github.com/betaflight/betaflight/files/8614187/btfl_001.bbl.txt)

limonspb, 2022-05-07:
> To be able to upload files I added the suffix .txt. Remove .txt after Download to open it in PID Toolbox or BBE.
> 
> RC6: [btfl_001.bbl.txt](https://github.com/betaflight/betaflight/files/8614183/btfl_001.bbl.txt)
> 
> RC4: [btfl_001.bbl.txt](https://github.com/betaflight/betaflight/files/8614187/btfl_001.bbl.txt)

there is some confusion/misinformation here that requires clarification and more careful testing:
1. You say that you used default PID-filters, but your DIFF and Blackbox shows that this is not true, you did not use default filters.
2. RC4 blackbox settings are even more crazy, i see you set some very low static notches and that explains even more crazy behavior:
[image]
3. You say "perfect for my quad up to BF4.3 RC5.", but then you share a crazy log from RC4.

So if your issue claims "RC4 flies good on defaults, but RC5 flies bad on the SAME quad" could you please:
1. Flash RC4 with full chip erase, make minimal setup like RPM filters ON and record blackbox that it flies good.
2. Flash RC6 with full chip erase (same drone, not just identical!!!) and record a new blackbox.

So far from your RC6 blackbox it looks like something is shaking on your drone on pitch exclusively, you can see its just pitch. But this thread is not for tuning-troubleshooting your drone, its for discussing why RC4 might fly good when RC6 does not.

Joerg-H-Wagner, 2022-05-07:
First of all thank you for your time. I alwasy perform a full erase and use the same drone.
The drone's behaviour came up with RC6 and it seems to be coincindental.
It's still there with FB 4.2.11, so it has nothing to do with RC6. I changed motors and now
I downgraded BLHeli to 32.8 and try to find the problem.
It's not my intention to get tuning tips. 
I'm really sorry for your time you spent.

closed 2022-05-07 by Joerg-H-Wagner

limonspb, 2022-05-07:
> First of all thank you for your time. I alwasy perform a full erase and use the same drone. The drone's behaviour came up with RC6 and it seems to be coincindental. It's still there with FB 4.2.11, [...]

No worries at all. Jump in beta flight discord, there is a support channel with plenty of people capable to figure out what's wrong with the drone by photos and blackbox. Cheers!