betaflight/betaflight #11578: Mamba Stack FC F405 MK2 not working with Mamba Flashbang SW601

Xenogears33, 2022-05-02:
### Describe the bug

MAMBA Flash-Bang LED Board SW601 are not lighting up nor saving inn BetaFlight 

### To Reproduce

Email or Instructions

### Expected behavior

Help on how to connect the LED

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

Mamba F405 mk2

### How are the different components wired up

White Cable - LED Pad BF
Black Cable - Ground Pad
Red Cable - V5

Xenogears33, 2022-05-02:
The information has been added as per your request.

Xenogears33, 2022-05-04:
Just to follow up as this is my first time doing this?

haslinghuis, 2022-05-04:
No worries.

https://www.youtube.com/watch?v=ltWvEvukEao

Xenogears33, 2022-05-04:
I have watched that video several times, my FC does not have a button that controls them life his video does as mine is Mamba f405 MK2.. it seems that is a firmware issue.

haslinghuis, 2022-05-04:
If you think it's a firmware issue is earlier firmware working (like 4.2.x) for you?

Xenogears33, 2022-05-04:
I can try but then if I flash the 4.3 I will not be able to fix the rotation of my motors.  They are in the correct spot but one of them was rotating incorrectly.

haslinghuis, 2022-05-04:
Direction can be set in ESC firmware too.

Xenogears33, 2022-05-05:
I do not have the F22 mini I have the Mamba Stack F405 MK2

closed 2022-06-11 by github-actions[bot]