betaflight/betaflight #11590: VTX POWER LEVEL SWITCHES IN FLIGHT

Dfixer4, 2022-05-08:
### Describe the bug

Ive tried with 2 of the same model fc -iflight succex-e f722- and 2 of the same vtx - rush tank solo 1w- and the issue persist. and that issue is that the power level will change while in mid flight with smartaudio enabled on the vtx. it usually will do it on longer flights, at least thats what ive noticed. though it can be random. ill have it set on 1watt and get about 1/2 a mile out and it will change to the lowest setting, 25mw. When i disable smartaudio and set it all manually there is no issue. also the change in the osd doesnt happen instantly, it switches to a lower level and says that its maxed, then changes a bit later. in the vid i sent at one watt i dont ever get that much breakup, also the radio acts like it has a failsafe audiably. im on elrs 2.0. none of these issues happen if smartaudio is disabled

https://user-images.githubusercontent.com/104888987/167277534-953cc324-21a2-44e1-a17c-6a09a0c346d4.mp4

.

### To Reproduce

turn smartaudio on and fly. thats all ive had to do to recreate the issue.

### Expected behavior

i expected for it to stay at the setting i set it to.

### Flight controller configuration

```
 # 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Apr 25 2022 / 01:06:24 (9d71184) MSP API: 1.44
# config: manufacturer_id: IFRC, board_name: IFLIGHT_SUCCEX_E_F7, version: c2b8c0d3, date: 2021-06-18T03:42:42Z

# start the command batch
batch start

board_name IFLIGHT_SUCCEX_E_F7
manufacturer_id IFRC

# name: Nazgul5 V2 FRME

# feature
feature GPS
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR

# serial
serial 3 2 115200 115200 0 115200
serial 4 1024 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
[... 502 lines cut ...]
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Flight controller

iflight succex-e f722, RDQ

### Other components

rx frsky r9mm flashed to explrs 2.0
rush tank solo 1w
flywoo v2.0 gps
betaflight 4.3.0

### How are the different components wired up

with the harnesses from the peripherals manufacturer and the fc harnesses that iflight supplies to the fc

### Add any other context about the problem that you think might be relevant here

it only happens when smartaudio is enabled on the port the vtx is.

JyeSmith, 2022-05-11:
A possible reason for this is the ELRS VTx Administrator sending a VTx command to BF mid flight.  ELRS does this when ever a tlm connection is made e.g. tlm is dropped and regained inflight.  But is only meant to happen when a quad is first power and a tlm connection made.  If this happens BF will change power (and pitmode) but not change the frequency.

Now, this shouldnt happen because ELRS has a guard to not send VTx commands when armed.  So I would check you have the arm switch set up correctly **Arming state must be HIGH on AUX1 for safety and feature reasons**.  This will prevent the VTx command being sent while armed.

You can also go into the VTx Administrator in Lua and change the band setting to "-".  This will disable it completely.  But please check your arm switch.

Dfixer4, 2022-05-16:
Thanks for the reply.  The arm swtch is set correctly. It only happens on
BF 4.3. Is what I've noticed. I don't have any issues at all on 4.2.11

It doesn't have telm issues or smartaudio issues. But as soon as I flash to
4.3 it starts.

I appreciate what u do!

Thanks again!

Dfixer4, 2022-05-17:
I have tried the arming in both positions and it does the same thing.

JyeSmith, 2022-05-17:
> You can also go into the VTx Administrator in Lua and change the band setting to "-". This will disable it completely. 

How about with the VTx Admin disabled?  This will rule ELRS out completely.

Dfixer4, 2022-05-23:
It still does it. Also tried the betaflight lua and it dies not work on 4.3
rc6, but does as soon as I go to 4.2.11. I'm going to try 4.2.11 In a
moment to check if it still does it. At the moment on 4.3 it is cycling
through all pwd lvls when armed.

Dfixer4, 2022-05-23:
I think I have found the issue. It seems that the vtxs pwr lvl button
was not soldered to the board securely. I removed it and it seems to not
change unless its told to, lol! Apologies for the issue. Though if it does
start again I will surely let u know.
  Though the bf lua is not working on 4.3 rc6. Ive tried 1:2 ratio at the
200hz setting and all other combinations. It does work on bf 4.2.11 for
surely. Some reason it doesn't want to work on rc6.

Once again,
Thnx!

haslinghuis, 2022-05-27:
If needed please open a new support ticket for the lua script issue.
Closing as hardware issue has been resolved.

closed 2022-05-27 by haslinghuis