betaflight/betaflight #11592: GPS not detected RC6

kensherman, 2022-05-09:
### Describe the bug

Firmware is configured for GPS, NEMA on Uart 4 and gps not detected. Have double checked wiring diagram.

### To Reproduce

Configure GPS on UART 4, enable GPS using NEMA. On boot GPS lights blink then stop. OSD menu long / lat does not display, altitude = '-'

### Expected behavior

gps detected and long/lat display on osd. When powered on gps light solid blue, blinking when aquired.

### Flight controller configuration

```
 # diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Apr 25 2022 / 01:06:24 (9d71184) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: MAMBAF722, version: ec268d11, date: 2021-07-26T13:10:45Z

# start the command batch
batch start

board_name MAMBAF722
manufacturer_id DIAT

# name: Rekon 4

# feature
feature GPS
feature RSSI_ADC

# serial
serial 2 8192 115200 57600 0 115200
serial 3 2 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 800
vtxtable powerlabels 25 100 200 400 800

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = -103,-10,-3888,1
set mag_align_pitch = 1800
set mag_align_yaw = 1800
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_output_reordering = 2,3,0,1,4,5,6,7
set gps_rescue_allow_arming_without_fix = ON
set osd_vbat_pos = 2477
set osd_rssi_dbm_pos = 2232
set osd_flymode_pos = 2169
set osd_vtx_channel_pos = 2516
set osd_mah_drawn_pos = 2444
set osd_craft_name_pos = 2498
set osd_gps_speed_pos = 2209
set osd_gps_lon_pos = 2129
set osd_gps_lat_pos = 2113
set osd_gps_sats_pos = 2201
set osd_home_dir_pos = 2285
set osd_home_dist_pos = 2177
set osd_flight_dist_pos = 2273
set osd_altitude_pos = 2241
set vtx_band = 2
set vtx_channel = 8
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5866
set gyro_1_align_yaw = 1800
set name = Rekon 4

profile 0

rateprofile 0

```
```
 Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 1
A09: MOTOR 2
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: SERIAL_TX 1
B07: SERIAL_RX 1
B08: CAMERA_CONTROL
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_RSSI
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 3
C09: MOTOR 4
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH3 : CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 1
    CH4 : DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 3

```

### Flight controller

Mamba DJI F722 Mini MK2 FC from Raceday quads

RipperGOT, 2022-05-11:
I have a similar problem in an Iflight SucceX-E F7

When I connect the USB or the battery, sometimes the GPS icon appears quickly and sometimes it takes up to a minute or 2 to detect it.  The same on OSD .
In other FC (Matek) with 4.3 I don't have that problem

kensherman, 2022-05-13:
I took out the chip from my quad where gps is working with rc6 and plugged it into the quad that's not working on RC 6 and the led's indicate signal aquired and I've double checked the wiring diagram to confirm that I don't have any cross wiring issues.

renatux, 2022-05-13:
I have a similar problema com M80 and Mamba F722S. On the Matek, is OK.

kensherman, 2022-05-16:
Rolled back to R2 since a few of the newer releases indicated gps rescue didn't work. GPS not showing up in OSD, altitude shows a "-" and no text what so ever for long / lat. The led on the gps chip is blinking which indicates positive signal from satelites.