betaflight/betaflight #11600: link quality is not working

pan-sattan, 2022-05-12:
### Describe the bug

Hi,

I installed latest RC6 of bf4.3 in my fc and I noticed, that link quality in OSD is still 0 (even the warning is on the screen)
In latest official bf release its working pretty good, no issues with the same components

### To Reproduce

its appear all the time

### Expected behavior

We should see proper values of link quality e.g. 7

### Flight controller configuration

# 

# diff

# version
# Betaflight / STM32F745 (S745) 4.3.0 Apr 25 2022 / 01:05:26 (9d71184) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEF7, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id HBRO

# name: Ape-X

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY

# serial
serial 0 8192 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# beeper
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -DISARM_REPEAT
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1400 1675 0 0
aux 2 2 2 1800 2100 0 0
aux 3 13 1 1800 2100 0 0
aux 4 19 3 1800 2100 0 0
aux 5 27 7 1800 2100 0 0
aux 6 35 4 1800 2100 0 0
aux 7 36 3 1400 1700 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 25 100 400 600
vtxtable powerlabels 25 100 400 800

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 0
set dyn_notch_count = 2
set dyn_notch_q = 250
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 400
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_calibration = -70,23,-46,1
set baro_hardware = AUTO
set rssi_scale = 200
set fpv_mix_degrees = 30
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set failsafe_switch_mode = STAGE2
set bat_capacity = 1500
set ibata_scale = 2886
set yaw_motors_reversed = ON
set small_angle = 180
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_alt_mode = CURRENT_ALT
set deadband = 6
set yaw_deadband = 2
set simplified_gyro_filter = OFF
set osd_warn_batt_not_full = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_rssi = ON
set osd_warn_over_cap = ON
set osd_rssi_alarm = 45
set osd_cap_alarm = 1200
set osd_alt_alarm = 120
set osd_tim1 = 2561
set osd_tim2 = 2563
set osd_rssi_pos = 2081
set osd_link_quality_pos = 2085
set osd_flymode_pos = 2113
set osd_vtx_channel_pos = 2537
set osd_current_pos = 449
set osd_mah_drawn_pos = 2092
set osd_altitude_pos = 2103
set osd_warnings_pos = 14729
set osd_avg_cell_voltage_pos = 2465
set osd_disarmed_pos = 2186
set osd_esc_tmp_pos = 504
set osd_flip_arrow_pos = 2318
set osd_core_temp_pos = 2552
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_max_dist = ON
set osd_stat_min_batt = OFF
set osd_stat_endbatt = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5658
set vcd_video_system = PAL
set gyro_1_align_yaw = 2700
set name = Ape-X

profile 1

# profile 1
set dterm_lpf1_dyn_min_hz = 0
set dterm_lpf1_static_hz = 0
set dterm_lpf2_type = BIQUAD
set dterm_lpf2_static_hz = 105
set yaw_lowpass_hz = 0
set simplified_dterm_filter_multiplier = 70

rateprofile 1

# rateprofile 1
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 1
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set roll_srate = 100
set pitch_srate = 100
set yaw_srate = 100

# end the command batch
batch end

# 

### Flight controller

HBRO/KAKUTEF7(STM32F745

### Other components

R9m, F-port

pan-sattan, 2022-05-12:
[image]

aowi7280, 2022-06-06:
Me too. Omnibus F405SD.

haslinghuis, 2022-06-07:
Could try the new ELRS protocol. Just needs another external TX module for your radio

https://www.expresslrs.org/1.0/quick-start/tx-r9m/

pan-sattan, 2022-07-09:
still not working even with official 4.3 version of BF

sugaarK, 2022-07-09:
> 

On r9 ?

pan-sattan, 2022-07-09:
yes FRSKY R9 system

kimxilxyong, 2022-07-31:
LQ still 0 in 4.3.1 with SRXL2

IlyaLight, 2023-08-20:
Same issue in STM32F722RET6 for 4.4.2

haslinghuis, 2023-08-20:
Try 4.4.3-zulu or 4.5-zulu

aowi7280, 2023-08-23:
What is a zulu? Is it stable enough to fly on a daily basis?

haslinghuis, 2023-08-23:
[zulu time ](https://time.is/nl/Z)

All developers use zulu and master builds are assumed in a working state. As any project has unresolved issues it should work for the majority. Just know which issues are there and use the standard pre-flight checks. YMMV as it depends on hardware combinations, build quality, configuration and tuning.

d3vv3, 2024-07-09:
I am facing this issue too

closed 2026-09-26 by blckmn