betaflight/betaflight #11601: Drone does a flip after arming (HGLRC F722 BF 4.3.0)

nast43, 2022-05-12:
### Describe the bug

Po aktualizacji HGLRC F722 do najnowszej wrsji BF 4.3.0 RC-6 dron zaraz po uzbrojeniu robi salta. Przed aktualizacja było dobrze. Posiadam dwia FC z tym problemem.

### To Reproduce

Every drone arming ends in a flip.

### Expected behavior

I was expecting a nice flight

### Flight controller configuration

# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Apr 25 2022 / 01:06:24 (9d71184) MSP API: 1.44
# config: manufacturer_id: HGLR, board_name: HGLRCF722, version: 836acaf9, date: 2021-10-31T19:48:38Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name HGLRCF722
manufacturer_id HGLR
mcu_id 000900153756500b20393558
signature 

# name: Zwiadowca

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature OSD
feature ESC_SENSOR

# serial
serial 1 2 115200 115200 0 115200
serial 2 8192 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map
map EATR1234

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 2 1300 1700 0 0
aux 2 2 2 1700 2100 0 0
aux 3 13 1 1700 2100 0 0
aux 4 19 6 1700 2100 0 0
aux 5 20 6 1700 2100 0 0
aux 6 35 5 1300 1700 0 0
aux 7 40 7 900 1300 0 0
aux 8 41 7 1700 2100 0 0
aux 9 49 5 1700 2100 0 0

# master
set gyro_lpf1_static_hz = 262
set gyro_lpf2_static_hz = 525
set gyro_lpf1_dyn_min_hz = 262
set gyro_lpf1_dyn_max_hz = 525
set acc_calibration = 19,32,-70,1
set mag_calibration = 1029,-4275,4399
set baro_bustype = SPI
set baro_spi_device = 1
set baro_i2c_device = 0
set serialrx_provider = SBUS
set dshot_idle_value = 500
set bat_capacity = 1800
set vbat_min_cell_voltage = 340
set vbat_warning_cell_voltage = 360
set simplified_gyro_filter_multiplier = 105
set osd_warn_batt_not_full = OFF
set osd_warn_crash_flip = OFF
set osd_warn_esc_fail = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_cap_alarm = 1650
set osd_vbat_pos = 2519
set osd_tim_1_pos = 459
set osd_crosshairs_pos = 6381
set osd_ah_pos = 2158
set osd_mah_drawn_pos = 4561
set osd_craft_name_pos = 34
set osd_gps_speed_pos = 6604
set osd_gps_lon_pos = 2063
set osd_gps_lat_pos = 2051
set osd_gps_sats_pos = 6593
set osd_home_dir_pos = 6599
set osd_home_dist_pos = 2497
set osd_flight_dist_pos = 449
set osd_compass_bar_pos = 361
set osd_altitude_pos = 18849
set osd_warnings_pos = 6537
set osd_avg_cell_voltage_pos = 6615
set osd_battery_usage_pos = 398
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set osd_stat_flight_dist = ON
set displayport_max7456_inv = ON
set displayport_max7456_wht = 0
set name = Zwiadowca

profile 0

# profile 0
set auto_profile_cell_count = 4

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

# rateprofile 1
set thr_expo = 61
set rates_type = BETAFLIGHT
set roll_rc_rate = 150
set pitch_rc_rate = 150
set yaw_rc_rate = 100
set roll_expo = 53
set pitch_expo = 53
set yaw_expo = 53
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 1

# save configuration
save
# 

### Flight controller

HGLRC F722

haslinghuis, 2022-05-12:
Did it flip on RC5 or earlier builds?

haslinghuis, 2022-05-12:
Also please provide information about the Control Link (receiver and transmitter) firmware versions

nast43, 2022-05-12:
Receiver: Frsky MX+(ACCST D16 v2.1.2)
Transmitter: Taranis x-lite S (ISRM  v2.1.6)
I checked RC5/RC4/RC3 and it is the same on all of them :/

nast43, 2022-05-13:
When I tested on RC5,RC4,RC3, I uploaded the above settings via cli. Maybe this was not a good idea and it would be better to reconfigure on RC5?

Wild-FPV, 2022-06-04:
@nast43 Have you checked -> motor mapping, motor direction, FC orientation, channel mapping, accelerometer calibration (if you're in angle mode)?

nast43, 2022-06-06:
@Wild-FPV Yes, I checked everything. I recently installed a new version of RC-7 and the quad behaves correctly :)

haslinghuis, 2022-06-07:
Fixed in RC7

closed 2022-06-07 by haslinghuis