betaflight/betaflight #11603: NBD Infinity AIO V2 DMA allocation issues

Tallestthomas, 2022-05-13:
### Describe the bug

There seems to be an issue with the DMA arbiter setting up the proper configuration on the NBD Infinity AIO V2, which uses the STM32F745 chip. This was found by seeing that current/voltage sensing was not working and after troubleshooting in the discord we discovered that the ADC DMA was not set.

### To Reproduce

This occurs from a fresh flash of the firmware with default configurations.

### Expected behavior

```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 4
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: DSHOT_BITBANG 3
DMA2 Stream 7: FREE
```

### Flight controller configuration

```
  diff

# version
# Betaflight / STM32F745 (S745) 4.3.0 Apr 25 2022 / 01:05:26 (9d71184) MSP API: 1.44
# config: manufacturer_id: NEBD, board_name: NBD_INFINITYAIOV2, version: d65edf3b, date: 2022-04-26T08:30:15Z

# start the command batch
batch start

board_name NBD_INFINITYAIOV2
manufacturer_id NEBD

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 165,-134,0,1
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300

profile 0

rateprofile 0

# end the command batch
batch end

# 
```
``resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: FLASH_CS
B01: GYRO_EXTI
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: FREE
B06: FREE
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FREE
B13: FREE
B14: FREE
B15: FREE
C00: LED 1
C01: ADC_BATT
C02: ADC_CURR
C03: FREE
C04: FREE
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: SPI_MOSI 3
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: BEEPER
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: GYRO_CS 1
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH2 : BEEPER
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Flight controller

NewBeeDrone Infinity AIO V2

haslinghuis, 2022-05-13:
Seems the target needs: 
```
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
```

haslinghuis, 2022-05-13:
Also check: https://github.com/betaflight/betaflight/pull/11604

haslinghuis, 2022-05-14:
Ping @SteveCEvans 

[image]

Tallestthomas, 2022-05-14:
#11604 and settings `dma ADC 1 1` does seem to fix the initial issues, but a new one has been discovered.
Having `osd_displayport_device` set to MSP seems to be having a conflict with the OSD dma.

This is having the HDZero freestyle vtx connected to UART2 for video/osd and UART1 for SA

```
# 

# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 4
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: DSHOT_BITBANG 3
DMA2 Stream 4: ADC
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# get osd_displayport_device
osd_displayport_device = MSP
Allowed values: NONE, AUTO, MAX7456, MSP, FRSKYOSD
Default value: AUTO

# get displayport_msp_serial
displayport_msp_serial = 1
Allowed range: -1 - 40
Default value: -1

# 
```
vs having `osd_displayport_device = AUTO`

```
# 

# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 4
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: DSHOT_BITBANG 3
DMA2 Stream 4: ADC
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

Tallestthomas, 2022-05-14:
#11604 and `dma ADC 1 1` fixes the issue, thanks!

ot0tot, 2022-05-14:
Why is `dma ADC 1 1` not being set properly by Betaflight? Should the target config be updated to include that, or is it a more general issue with DMA assignments?

haslinghuis, 2022-05-14:
Yes needs target update.

haslinghuis, 2022-05-15:
Will be closed upon merging the fixes

closed 2022-05-20 by blckmn