betaflight/betaflight #11610: Autopilot and list of auto commands?

sp1r1t3, 2022-05-18:
### Is your feature request related to a problem? Please describe

We have work with cinilifter taurus and thinking about some feature, for example "it will be good if we have autopilot which can flight like DJI - fly for the target"

### Describe the solution you'd like

So due to main idea of this topic, do we have list of automatic commands in beta flight for coding some kind of commands ? 
Or maybe somebody have any others ideas for autopilot mode

### Describe alternatives you've considered

We using taurus with  BLITZ F7 Pro Flight Controller that's why we have any kind of alternatives.

KarateBrot, 2022-05-19:
This sounds like something "iNav waypoint missions" could probably do for you.

sp1r1t3, 2022-05-19:
Yes and no. We want create some application for  cinematic drone with typical list of commands for FPV witch help them to use it without 100+ hours training on fpv). Thats why I research this topic.

closed 2022-06-26 by github-actions[bot]