betaflight/betaflight #11612: Just updated to 4.3 RC6- Receiver tab- Model on the preview going crazy

rlesnet2, 2022-05-19:
### Describe the bug

When going to the receiver tab, All of the controls seem to work correctly according to the slider scales, but, on the preview, the representation of the drone is flipping out (please see attached video). I flashed back to 4.2.11 and it did not do that. I replicated this three times on 4.3 RC6 with the same results each time

https://user-images.githubusercontent.com/104951377/169403574-ce3c29aa-d4a0-4dc9-a1f6-b258f3d99234.mp4

.

### To Reproduce

Flashed back and forth between 4.2.11 and 4.3 RC6 three times with the same results each time.

### Expected behavior

When flashing to 4.3 RC6, I expected the receiver tab to react normally. I expected to be able to set my sticks and be able to control the preview as you would normally be able to with my radio.

### Flight controller configuration

```
 # version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Apr 25 2022 / 01:06:24 (9d71184) MSP API: 1.44
# config: manufacturer_id: ZEEZ, board_name: ZEEZF7, version: 309c0f10, date: 2021-06-14T21:40:58Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name ZEEZF7
manufacturer_id ZEEZ
mcu_id 004e00203538511934393631
signature 

# serial
serial 3 0 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# map
map RETA1234

# master
set acc_calibration = -128,-52,118,1

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

```
```
  # resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FLASH_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: CAMERA_CONTROL
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: LED_STRIP
B01: FREE
B02: BEEPER
B03: FREE
B04: MOTOR 4
B05: SPI_MOSI 3
B06: MOTOR 3
B07: MOTOR 2
B08: MOTOR 1
B09: FREE
B10: SERIAL_TX 3
B11: PINIO 1
B12: GYRO_CS 1
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_BATT
C03: ADC_CURR
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: GYRO_EXTI
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: FREE
C14: LED 1
C15: LED 2
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : CAMERA_CONTROL
    CH2N: LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: ADC
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE
```

### Flight controller

Zeez F722 FC bought from RDQ

### Other components

ZEEZ 32Bit 60A 4in1 ESC bought from RDQ as a stack
VIFLY Finder 2 Buzzer also from RDQ
TBS Unify Pro32 Nano 5G8 V1.1 500mW Video Transmitter from GetFPV
TBS CROSSFIRE Nano RX Do not remember where from
Foxeer Arrow Pro Camera Given to me by a friend

### How are the different components wired up

Please see the attached wiring diagram.
1 difference is that the rx is wired to uart 5 instead of 4. Lost a pad on uart 4 and had to switch.

[image]
.the 

### Add any other context about the problem that you think might be relevant here

Can't think of anything.

KarateBrot, 2022-05-19:
Maybe you are using an old version of the configurator that had a bug. Try updating to the latest Configurator and see if that fixes it.

rlesnet2, 2022-05-19:
That was quick... Thanks,

This is the version I am using. Sorry about not putting it in the notes. 
As far as I know, this is the latest version.

haslinghuis, 2022-05-19:
Please check with https://github.com/betaflight/betaflight-configurator-nightlies/releases

rlesnet2, 2022-05-19:
That works, Thanks so much. Do I need to withdraw my bug report, or, do 
you guys do that?

Thanks again for all you guys do,

Rob L.

closed 2022-05-20 by haslinghuis