betaflight/betaflight #11616: no more connexion after flash

pl2267, 2022-05-23:
### Describe the bug

Hello,
I had the Cinerace connected, bind with my jumper ( t8sg). The drone was flying well exept the beep beep beep...
So I tried to uptdate by flashing the microprogram, selected flywoof745 as indicated is the manual, and then I can no more connect the drone to betaflight, It does no more respond to controler a if it was no more binded...
Help please

### To Reproduce

just after flashing microprogram. The green bar filled, all seem OK until is gone wrong...

### Expected behavior

I thought The beep beep stoped...

### Flight controller configuration

I can not access to cli because of no connection...

### Flight controller

GOKU FC series for Flywoo cinerade HD

### Other components

VTX is the Dji air unit

haslinghuis, 2022-05-23:
Please provide:

- a link to the manual
- configurator version used
- firmware version flashed.

pl2267, 2022-05-24:
Hi Haslinghuis,
Here is a link to the cinerace HD manual: https://flywoo.tawk.help/article/cinerace20-series
I use Betaflight 10.7.2
I have tried to use the FLYWOOF745
I am not sure to anwser correctly because I begin with betaflight...

haslinghuis, 2022-05-24:
From here: https://flywoo.net/collections/cinerace20-collection/products/cinerace20-v1-2-neon-led-dji-hd-w-polar-nano-2inch and from the manual you might try flywoof745nano

[image]

CLI: https://cdn.shopifycdn.net/s/files/1/0010/7410/2324/files/FLYWOO_GNF745NANO_16-16mm_28fc54fb-95bd-4b91-9b22-4cb36c1d4b00.txt?v=1631786490

closed 2022-07-01 by github-actions[bot]