betaflight/betaflight #11619: BF 4.2.11 - Mamba MK4 H743 - Mass storage device not supported

matyas-brenner, 2022-05-24:
### Describe the bug

Mass storage device for downloading blackbox files is not possible Using BF 4.2.11 and BF Configurator 10.7.2.

[image]

### To Reproduce

Go to Blackbox tab and click activate mass storage mode.
"This operation is not supported by your hardware" will appear.

### Expected behavior

Mass storage device should be activated and a new drive should pop-up.

### Flight controller configuration

```
  # 

# diff

# version
# Betaflight / STM32H743 (SH74) 4.2.11 Nov  9 2021 / 20:29:57 (948ba6339) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: MAMBAH743, version: e4b51680, date: 2021-10-04T00:52:01Z

# start the command batch
batch start

board_name MAMBAH743
manufacturer_id DIAT

# name: MB

# resources
resource MOTOR 1 A03
resource MOTOR 2 A02
resource MOTOR 3 A01
resource MOTOR 4 A00

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature GPS
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CRASH_FLIP

# serial
serial 2 1 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200
serial 7 2048 115200 57600 0 115200

# aux
aux 0 0 2 1350 2100 0 0
aux 1 1 0 1750 2100 0 0
aux 2 2 0 1300 1700 0 0
aux 3 13 3 1700 2100 0 0
aux 4 28 2 1700 2100 0 0
aux 5 35 3 1350 1700 0 0
aux 6 36 1 1700 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY    0    0    0    0    0    0    0    0
vtxtable band 2 BOSCAM_B B FACTORY    0    0    0    0    0    0    0    0
vtxtable band 3 BOSCAM_E E FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY    0 5760    0 5800    0    0    0    0
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I FACTORY    0    0    0    0    0    0    0    0
vtxtable powerlevels 5
vtxtable powervalues 14 23 27 30 0
vtxtable powerlabels 25 200 500 MAX 0

# master
set gyro_lowpass2_hz = 425
set dyn_lpf_gyro_min_hz = 340
set dyn_lpf_gyro_max_hz = 850
set acc_calibration = -53,-23,93,1
set mag_hardware = NONE
set min_check = 1010
set max_check = 1990
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set vbat_max_cell_voltage = 435
set ibata_scale = 80
set beeper_inversion = ON
set beeper_od = OFF
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set report_cell_voltage = ON
set osd_warn_batt_not_full = OFF
set osd_warn_core_temp = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_link_quality = ON
set osd_rssi_alarm = 40
set osd_vbat_pos = 471
set osd_rssi_pos = 34
set osd_link_quality_pos = 14337
set osd_tim_2_pos = 14816
set osd_flymode_pos = 14362
set osd_throttle_pos = 14784
set osd_vtx_channel_pos = 399
set osd_crosshairs_pos = 237
set osd_ah_pos = 110
set osd_current_pos = 14752
set osd_mah_drawn_pos = 14809
set osd_craft_name_pos = 8426
set osd_gps_speed_pos = 14777
set osd_gps_lon_pos = 97
set osd_gps_lat_pos = 129
set osd_gps_sats_pos = 14368
set osd_home_dir_pos = 14348
set osd_home_dist_pos = 14350
set osd_altitude_pos = 14743
set osd_warnings_pos = 14666
set osd_avg_cell_voltage_pos = 14840
set osd_disarmed_pos = 14635
set osd_flip_arrow_pos = 14478
set osd_stat_max_dist = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_displayport_device = MSP
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 2
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5880
set displayport_msp_serial = 2
set name = MB

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 119
set dyn_lpf_dterm_max_hz = 289
set dterm_lowpass2_hz = 255
set p_pitch = 69
set d_pitch = 44
set f_pitch = 190
set p_roll = 63
set d_roll = 40
set f_roll = 180
set p_yaw = 68
set f_yaw = 180
set d_min_roll = 0
set d_min_pitch = 0

rateprofile 1

# rateprofile 1
set roll_rc_rate = 127
set pitch_rc_rate = 127
set yaw_rc_rate = 127
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 40
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 72
set tpa_breakpoint = 1750

# end the command batch
batch end

# 
```
```
  resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 4
A01: MOTOR 3
A02: MOTOR 2
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: FREE
B01: FREE
B02: SPI_MOSI 3
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: PREINIT 4
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: PINIO 1
C06: FREE
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: SERIAL_RX 4
D01: SERIAL_TX 4
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: SERIAL_TX 3
D09: SERIAL_RX 3
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: SERIAL_TX 8
E02: FREE
E03: BEEPER
E04: LED 2
E05: LED 1
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: GYRO_CS 2
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: MOTOR 4
    CH2: MOTOR 3
    CH3: MOTOR 2
    CH4: MOTOR 1
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 4
DMA1 Stream 1: MOTOR 3
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: MOTOR 1
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Flight controller

Mamba Basic MK4 H743 X8

### Other components

MAMBA F55_128K ESC
TBS Tracer
HDZero freestyle VTX
bn-180 GPS

Sekilsgs2, 2022-05-30:
[betaflight_4.2.11_STM32H743_.zip](https://github.com/betaflight/betaflight/files/8799642/betaflight_4.2.11_STM32H743_.zip)
Please test this fw - i'm fix all problems what im find..

matyas-brenner, 2022-06-04:
Mass Storage works with BF 4.3.0 RC 7

closed 2022-06-04 by haslinghuis