betaflight/betaflight #11620: RushFPV RUSH BLADE F722 FC Blackbox Problems

Molukker, 2022-05-26:
### Describe the bug

I flashed 4.3 RC1 / RC3 and RC6 and i have allways a corrupt File in Blackbox.

No logs could be parsed succesfully.

### To Reproduce

-

### Expected behavior

I want to look at Blackbox File for tuning the Copter

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Dec 24 2021 / 18:42:12 (norevision) MSP API: 1.44
# config: manufacturer_id: RUSH, board_name: BLADE_F7_HD, version: c2b8c0d3, date: 2021-06-18T03:42:42Z

# start the command batch
batch start

board_name BLADE_F7_HD
manufacturer_id RUSH

# feature
feature -LED_STRIP
feature -ESC_SENSOR
feature GPS
feature TELEMETRY

# serial
serial 0 2 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 1 1800 2100 0 0
aux 1 46 3 1825 2100 0 0
aux 2 13 2 1300 2100 0 0
aux 3 35 2 1825 2100 0 0

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 35,-44,39,1
set rc_smoothing_auto_factor = 60
set serialrx_provider = CRSF
set dshot_idle_value = 0
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set align_board_roll = 180
set yaw_motors_reversed = ON
set small_angle = 60
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 40
set gps_rescue_initial_alt = 70
set gps_rescue_ascend_rate = 400
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set deadband = 4
set yaw_deadband = 4
set pid_process_denom = 2
set osd_vbat_pos = 2486
set osd_gps_speed_pos = 2072
set osd_gps_sats_pos = 2065
set osd_home_dir_pos = 2477
set osd_home_dist_pos = 2507
set osd_flight_dist_pos = 2049
set osd_avg_cell_voltage_pos = 2454
set debug_mode = GYRO_SCALED

profile 0

# profile 0
set vbat_sag_compensation = 100
set motor_output_limit = 90
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 45
set feedforward_jitter_factor = 10
set dyn_idle_min_rpm = 22

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set throttle_limit_type = SCALE

# end the command batch
batch end

### Flight controller

RushFPV RUSH BLADE F722 Flight Controller for DJI Digital

### Other components

Caddx Vista
Tekko 32 65A Metal F4 ESC

### How are the different components wired up

All works without problems except Blackbox

Denethor85, 2022-05-26:
you probably have a memory problem, because I have the exact same Flight Controller and I have no problems with the blackbox

Molukker, 2022-05-27:
Ok thank you very much

 
 

Gesendet: Freitag, 27. Mai 2022 um 00:17 Uhr

etracer65, 2022-05-27:
@Molukker Before you mistakingly replace your flight controller due to bad information, the problem is simply that you haven't set up a switch to activate the BLACKBOX recording on the Modes tab. You don't have a failed flash memory chip and you don't need to replace the flight controller. The message "no logs could be parsed" just means that there are no logs recorded (because you never activated recording), not that the files are "corrupt".

Molukker, 2022-05-28:
Hi and thank you for this information

 

but then its the first FC where i have to set a Blackbox Switch..

Best Regards

 

Matthias

 
 

Gesendet: Freitag, 27. Mai 2022 um 17:02 Uhr

closed 2022-07-06 by github-actions[bot]