betaflight/betaflight #11622: BTFL_4.3.0_BETAFPVF4SX1280 - display tab not working in the gui but can be activated in the CLI

Nikotttin, 2022-05-27:
### Describe the bug

Using ELRS v2.x on a SPI RX in the BETAFPVF4SX1280, with an Eachine nano VTX v1 and DJI goggles with analog input.
BF latest version, using the betafpv CLI found on their website.

The display tab on the parameter section of BF 4.3 RC6 cannot be turned on. You can activate it but saving reverts it to inactive. 

Solution: In the CLI type feature -display and the feature will be enabled.
In the GUI, the display tab will still appear as inactive, but one now sees the OSD  

### To Reproduce

Turn on the "display" in the parameter tab, save and reboot... Still no OSD. Enable the feature via CLI and it works.

### Expected behavior

Display should be activated in the GUI

### Flight controller configuration

```
  # version
# Betaflight / BETAFPVF4SX1280 (BHER) 4.3.0 Apr 25 2022 / 00:42:18 (9d71184) MSP API: 1.44

# start the command batch
batch start

board_name BETAFPVF4SX1280
manufacturer_id BEFH

# name: Meteor65-ELRS

# feature
feature -LED_STRIP
feature -ANTI_GRAVITY

# serial
serial 30 8192 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# aux
aux 0 0 0 1825 2100 0 0
aux 1 1 2 900 1125 0 0
aux 2 2 2 1350 1625 0 0
aux 3 13 3 1375 1575 0 0
aux 4 35 3 1875 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 25 100 200 400
vtxtable powerlabels 25 100 200 400

# master
set gyro_notch1_hz = 400
set gyro_notch2_hz = 200
set acc_calibration = 85,-207,340,1
set max_check = 2000
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 300
set current_meter = NONE
set ibata_scale = 179
set yaw_motors_reversed = ON
set small_angle = 180
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_link_quality = ON
set osd_vbat_pos = 2433
set osd_rssi_pos = 193
set osd_link_quality_pos = 2273
set osd_rssi_dbm_pos = 2305
set osd_tim_1_pos = 386
set osd_tim_2_pos = 2455
set osd_remaining_time_estimate_pos = 33
set osd_flymode_pos = 375
set osd_throttle_pos = 353
set osd_vtx_channel_pos = 32
set osd_current_pos = 343
set osd_craft_name_pos = 393
set osd_home_dist_pos = 98
set osd_flight_dist_pos = 130
set osd_warnings_pos = 2378
set osd_avg_cell_voltage_pos = 289
set osd_disarmed_pos = 2315
set osd_esc_tmp_pos = 163
set osd_esc_rpm_pos = 150
set osd_core_temp_pos = 278
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_max_fft = ON
set osd_stat_min_rssi_dbm = ON
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5806
set expresslrs_uid = 74,21,186,207,161,173
set expresslrs_rate_index = 2
set name = Meteor65-ELRS

profile 2

# profile 2
set p_pitch = 51
set i_pitch = 99
set d_pitch = 42
set f_pitch = 105
set p_roll = 46
set i_roll = 94
set d_roll = 39
set f_roll = 99
set p_yaw = 50
set i_yaw = 99
set f_yaw = 99
set d_min_roll = 25
set d_min_pitch = 28

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70

# end the command batch
batch end

```
```
 # resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 2
A01: ADC_BATT
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: SERIAL_TX 11
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: BEEPER
A15: RX_SPI_CS
B00: FREE
B01: FREE
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: GYRO_EXTI
B07: MOTOR 4
B08: MOTOR 1
B09: RX_SPI_EXPRESSLRS_RESET
B10: MOTOR 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: RX_SPI_EXTI
C14: LED 1
C15: LED
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : SERIAL_TX 11
TIM2:
    CH1 : MOTOR 2
    CH3 : MOTOR 3
TIM3: FREE
TIM4:
    CH2 : MOTOR 4
    CH3 : MOTOR 1
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

BETAFPVF4SX1280 

### Other components

eachine nano vtx IRC v1,
foxeer pico camera
0802 22'000kv

### How are the different components wired up

all direct soldered

haslinghuis, 2022-05-27:
Which command are you using?
> Enable the feature via CLI and it works.

DISPLAY is for external OLED screen. You need OSD feature which is enabled by default.

[image]

Nikotttin, 2022-05-27:
If you have oled displays you need this active. See here:
https://youtu.be/qqJKlsEyf7U

The CLI command was: feature -display

haslinghuis, 2022-05-27:
> The CLI command was: feature -display

This commands disables the DISPLAY feature. (Try `feature DISPLAY` and you will see it won't stick)

This feature can only be enabled on targets > 512K

Nikotttin, 2022-05-27:
Oh sorry then I made a mistake in the command I used! I had no OSD... After
using the CLI (and the command without -), it worked. However in the GUI
the display tab is inactive...

haslinghuis, 2022-05-27:
Type `feature` in CLI to see what's enabled. It's not possible to enable `DISPLAY` as I said the board has only 512K and the feature is now only available for targets > 512K.

```
# feature
Enabled: SOFTSERIAL TELEMETRY LED_STRIP OSD AIRMODE RX_SPI ANTI_GRAVITY
```

Nikotttin, 2022-05-27:
I did this before and that's what pushed me to use the CLI command...
Here's the pasted command:
# Feature
Enabled: SOFTSERIAL TELEMETRY OSD AIRMODE RX_SPI

This said, and sorry for not being clear, but what you said about using
external displays seems to also be required to turn on the OSD.

In my case, as you said, the OSD button was on... but no OSD. I tried
pal/ntsc, I tried disabling and re-enabling the osd, I tried moving
elements on the OSD, I tried disabling and re-enabling softserial, etc.

The only thing that worked was that display command.

Prajwal-Koirala, 2023-12-27:
The display tab in the Betaflight configurator GUI is not working for the user because they are using a flight controller with 512K of flash memory. The display feature is now only available for flight controllers with more than 512K of flash memory.

The workaround is to use the CLI to enable the OSD feature. This will enable the OSD, but the display tab in the GUI will still be inactive.

I hope this summary is helpful! Let me know if you have any other questions.

haslinghuis, 2024-09-10:
Stale.

closed 2024-09-10 by haslinghuis