betaflight/betaflight #11625: someone tries to share malicious content in wiki

SirHerpDerp, 2022-05-30:
### Is your feature request related to a problem? Please describe

Dear BetaFlight team,

a friend of mine is currently trying to configure his new drone and since he is not speaking english properly I'm helping him out. I finally found something interesing inside the wiki. It seems like some users are trying to spread malicious content here.

Take a look at the links to the tools zadig and ImpulseRC_Driver_Fixer, they are not leading to the correct software. I have edited this once yesterday but the user https://github.com/lampii is reverting these changes so that users will download this malicious content again.

The currently linked archive contains the following files

[image]

But zadig is just a simple *.exe file (~2.7 MB) which do not need to be installed. Inside this archive you can see a 700MB installer and there are some interesting files inside.

Just a snippet of the 'config.cfg'
```
[ColorsLight]
graphFont=FF000000
BuyOrder=FF000000
BuyPendingOrder=FFB00000
SellOrder=FF0000FF
LastPrice=FF800000
```

>> seems like it has something to do with cryptocurrencies and not with zadig.

And even the readme.md file contains some mysterious content.
```
# Advanced Trading System
Private Trading Software with flexible settings for profitable trading.
```

This has definitely nothing to do with zadig! Seems like someone is trying to trick the users here. Please try to fix this. I don't know how wikis are working here but I think it's not good that everybody can make changes here.

Thanks in advance and have a nice day.

### Describe the solution you'd like

read only for wiki or changes have to be reviewed

### Describe alternatives you've considered

-

KarateBrot, 2022-05-30:
Which wiki pages are we talking about? Do you have the link available?

SirHerpDerp, 2022-05-30:
Sorry, yes of course ... the most important information ....
here's the link, it's the installation instruction

https://github.com/betaflight/betaflight/wiki/Installing-Betaflight

KarateBrot, 2022-05-30:
Thank you very much for your report!

hydra, 2022-05-30:
> ... I guess now we need to lock the Wiki. Sorry guys.

can you not just ban the user?

KarateBrot, 2022-05-30:
> can you not just ban the user?

There was another spam account which edited the page beforehand (already banned by Github). But I suspect it won't be the last spam account trying to sneak in links to malware.

haslinghuis, 2022-05-30:
Thanks @SirHerpDerp. Wiki is protected now.

closed 2022-05-30 by haslinghuis

KarateBrot, 2022-06-03:
Update: spam account lampii got banned