betaflight/betaflight #11629: Dedicted Signal-to-Noise OSD widget

kevinsumner, 2022-06-03:
### Is your feature request related to a problem? Please describe

Currently, signal-to-noise ratio is reported by many modern control links (Crossfire, ELRS, etc.), but Betaflight can only show SNR by replacing the RSSI dBm widget with the `csrf_use_rx_snr` configuration item. This makes RSSI dBm and SNR mutually exclusive in the OSD.

### Describe the solution you'd like

Much like LQ and RSSI dBm value, add a dedicated OSD widget to Betaflight for SNR.

Work in Configurator would also need to be done to add it eventually, but even setting via CLI only would be an improvement. (I'm happy to file a related bug directly to Configurator if needed.)

### Describe alternatives you've considered

Not using SNR at all like we always have, or using telemetry plus a sound trigger on EdgeTX.

### Other information

This is a direct result of @joshuabardwell's SNR video that was released today. https://www.youtube.com/watch?v=seUi-_HV_O4

I didn't see another issue specifically for SNR OSD widget already open.

SaharMMesri, 2022-06-04:
Just commenting here to say I also support this feature request. As a new betaflight user, having complete information about RSSI, and SNR, and LQ simultaneously help me understand the limits of the hardware and fly safely.

DavBfr, 2022-06-04:
A PR is pending here: #11062

haslinghuis, 2023-01-29:
Fixed in #11062

closed 2023-01-29 by haslinghuis