betaflight/betaflight #11631: Request: hardware (eg SWD) flash and debug guide

David-OConnor, 2022-06-04:
### Is your feature request related to a problem? Please describe

Currently, there appears to be no guide or instructions on how to get started developing for BF.

### Describe the solution you'd like

It would be nice to have a guide in the [development folder](https://github.com/betaflight/betaflight/tree/master/docs/development) that describes how developers new to BF can set up a dev environment, and flash/debug using an SWD connection like ST-Link. For example, which operating systems are supported for development, which flash tools (STM32Cube IDE? OpenOCD?, PlatformIO?), how to configure them etc for BetaFlight.

It would be in the form of concise, step-by-step instructions, with amplifying information as required.

### Describe alternatives you've considered

We could rely on other communications channels like Slack and Discord. I think it's better to have an official doc in the repo, since this would save people's time. Additionally, it seems like there isn't much development-related chat on those mediums currently.

### Other information

I'd be happy to write the document if someone can talk me through it!

haslinghuis, 2022-06-07:
> It would be nice to have a guide in the [development folder](https://github.com/betaflight/betaflight/tree/master/docs/development) that describes how developers new to BF can set up a dev environment

These are the latest documents updated (excludes hardware debugging):
https://github.com/betaflight/betaflight/blob/master/docs/development/Building%20in%20Ubuntu.md
https://github.com/betaflight/betaflight/blob/master/docs/development/Building%20in%20Fedora.md

More useful links:
https://github.com/betaflight/betaflight/wiki/SWD-debug-port-connection-conflicts.
https://github.com/betaflight/betaflight/wiki/Betaflight-Contribution

Personally would like to have a single entity for information as wiki has been restricted now and using something like Sphinx to upload content to the betaflight.com website.

Any contribution on documentation is appreciated.

David-OConnor, 2022-06-08:
Thank you. Of note, I've found those first 2 links to work well, and I've used them to successfully build binaries for BF. The main thing I think they're missing is differentiation or examples between building legacy and unified targets. As you point out, if there's a problem when building and flashing from these binaries (As I've found out!), you can't really troubleshoot - Ie these are guides to build final/tested firmware; not to develop or troubleshoot with.

The 4th link is a great guide on Git and PRs, but doesn't cover development and debugging. Of note, the `squash` section there is the clearest I've found anywhere.

David-OConnor, 2022-07-05:
[Example guide, from ExpressLRS](https://www.expresslrs.org/2.0/software/toolchain-install/).

closed 2026-09-26 by blckmn