betaflight/betaflight #11634: "Matek f405 SE" works unstable.

jsungho, 2022-06-05:
### Describe the bug

BETAFLIGHT 4.3 There is a problem with all versions. But in version 4.2.11 it works fine.

Manufactured by: Matek F405 SE

<Abnormal behavior list>
- Hangs when connecting to Betaflight Configurator 4.3 (Stop 1 out of 2)
- When FC stopped, it stops after about 10 seconds.
- When FC stopped, the telemetry output is also stopped. (receiver : r9mx)

### To Reproduce

-

### Expected behavior

can't fly
If I fly before the FC stops, I can fly normally.

### Flight controller configuration

```
# diff all

# version
# Betaflight / STM32F405 (S405) 4.3.0 Jun  1 2022 / 06:20:04 (4fa2dc1) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF405SE, version: 42dd44a1, date: 2020-07-24T06:02:46Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF405SE
manufacturer_id MTKS
mcu_id 0033003b5350501520353043
signature 

# name: aramir_5inch

# feature
feature -SOFTSERIAL
feature -AIRMODE
feature INFLIGHT_ACC_CAL
feature GPS

# serial
serial 3 1 115200 57600 0 115200
serial 4 2 115200 57600 0 115200

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TREA1234

# aux
aux 0 0 0 1500 2100 0 0
aux 1 1 4 900 1325 0 0
aux 2 2 4 1700 2100 0 0
aux 3 6 3 900 1300 0 0
aux 4 46 5 1700 2100 0 0
aux 5 13 1 900 1300 0 0
aux 6 19 2 900 1300 0 0
aux 7 28 3 900 1700 0 0
aux 8 28 8 900 1900 0 0
aux 9 35 5 900 1300 0 0
aux 10 36 1 1700 2100 0 0
aux 11 37 7 1500 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY    0    0    0    0    0    0    0    0
vtxtable band 2 BOSCAM_B B FACTORY    0    0    0    0    0    0    0    0
vtxtable band 3 BOSCAM_E E FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY    0 5760    0 5800    0    0    0    0
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I FACTORY    0    0    0    0    0    0    0    0
vtxtable powerlevels 4
vtxtable powervalues 14 23 27 0
vtxtable powerlabels 25 200 500 0

# rxfail
rxfail 3 s 1275
rxfail 8 s 1900

# master
set dyn_notch_count = 2
set dyn_notch_q = 400
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 800
set gyro_lpf1_dyn_min_hz = 200
set gyro_lpf1_dyn_expo = 8
set acc_calibration = 42,29,-317,1
set align_mag = CUSTOM
set mag_align_roll = -20
set mag_align_pitch = 1800
set mag_align_yaw = 2700
set mag_calibration = 137,3,209
set baro_hardware = DPS310
set max_check = 1950
set rssi_scale = 200
set fpv_mix_degrees = 15
set serialrx_provider = FPORT
set airmode_start_throttle_percent = 15
set serialrx_halfduplex = ON
set blackbox_device = NONE
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 5
set failsafe_procedure = GPS-RESCUE
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = MSAS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_angle = 35
set gps_rescue_initial_alt = 20
set gps_rescue_descent_dist = 100
set gps_rescue_ground_speed = 1250
set gps_rescue_throttle_min = 1150
set gps_rescue_throttle_max = 1700
set gps_rescue_ascend_rate = 416
set gps_rescue_descend_rate = 208
set gps_rescue_min_sats = 6
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set deadband = 4
set yaw_deadband = 4
set pid_process_denom = 2
set simplified_gyro_filter = OFF
set tlm_halfduplex = OFF
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_batt_critical = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_rssi = ON
set osd_rssi_alarm = 30
set osd_cap_alarm = 0
set osd_tim1 = 3
set osd_vbat_pos = 14700
set osd_rssi_pos = 14705
set osd_rssi_dbm_pos = 451
set osd_tim_1_pos = 14808
set osd_flymode_pos = 14414
set osd_throttle_pos = 14594
set osd_current_pos = 14754
set osd_mah_drawn_pos = 14787
set osd_craft_name_pos = 14816
set osd_gps_speed_pos = 14585
set osd_gps_lon_pos = 14350
set osd_gps_lat_pos = 14340
set osd_gps_sats_pos = 14336
set osd_home_dir_pos = 14447
set osd_home_dist_pos = 14617
set osd_flight_dist_pos = 14777
set osd_compass_bar_pos = 138
set osd_altitude_pos = 30937
set osd_warnings_pos = 14829
set osd_avg_cell_voltage_pos = 14723
set osd_pit_ang_pos = 14529
set osd_rol_ang_pos = 14561
set osd_battery_usage_pos = 14794
set osd_disarmed_pos = 234
set osd_esc_tmp_pos = 54
set osd_rtc_date_time_pos = 14368
set osd_core_temp_pos = 14361
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_flight_dist = ON
set osd_displayport_device = MSP
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 1
set vtx_freq = 5769
set displayport_msp_serial = 3
set gyro_1_align_yaw = 2700
set timezone_offset_minutes = 540
set name = aramir_5inch

profile 0

# profile 0
set dterm_lpf1_dyn_expo = 0
set dterm_lpf1_static_hz = 0
set anti_gravity_gain = 4000
set p_pitch = 71
set i_pitch = 95
set d_pitch = 42
set f_pitch = 155
set p_roll = 59
set i_roll = 79
set d_roll = 39
set f_roll = 129
set p_yaw = 59
set i_yaw = 79
set f_yaw = 129
set d_min_roll = 39
set d_min_pitch = 42
set feedforward_smooth_factor = 40
set feedforward_jitter_factor = 12
set dyn_idle_min_rpm = 18
set simplified_master_multiplier = 120
set simplified_i_gain = 75
set simplified_d_gain = 110
set simplified_pi_gain = 110
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 90
set simplified_pitch_d_gain = 95
set simplified_pitch_pi_gain = 115

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 20
set thr_expo = 35
set rates_type = BETAFLIGHT
set roll_rc_rate = 214
set pitch_rc_rate = 213
set yaw_rc_rate = 208
set roll_expo = 77
set pitch_expo = 72
set yaw_expo = 55
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0
set throttle_limit_type = SCALE
set throttle_limit_percent = 90

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration

```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

Matek f405 se

### Other components

- VTX : HDZERO 500W
- RECIVER : FRSKY R9MX
- ESC : RacerStar 35A BLHeliS

### How are the different components wired up

-

### Add any other context about the problem that you think might be relevant here

-

jsungho, 2022-06-12:
> 

I found the cause.
Disconnecting the GPS (M8Q-5883) works fine.
Stuck on Fc(matek f405 se) connection

It works fine in betaflight 4.2.11 and inav 4.1 (even when connected to GPS)
It only doesn't work in beta 4.3.

VMaximI, 2022-06-14:
I have a similar problem on this controller. Version BF 4.3.0. I solved the problem by turning on the speed of the port on which the GPS hangs in AUTO mode. After that the config works fine.

haslinghuis, 2022-06-14:
Thanks @VMaximI for providing a solution. Closing as configuration issue.

closed 2022-06-14 by haslinghuis

jsungho, 2022-06-17:
The manufacturer answered.

It should be a bug in BF4.3 to use 2x I2C bus.
Disable i2c2 bus, connect compass to I2C1 bus,  Barometer is on I2C1 bus also.

Type CLI command
resource I2C_SCL 2 NONE
resource I2C_SDA 2 NONE
set mag_i2c_device = 1

After a while I go home, change the compass connection and set up and use the i2c port.

jsungho, 2022-06-17:
> 제조사가 대답했다.
> 
> 2x I2C 버스를 사용하는 것은 BF4.3의 버그여야 합니다. i2c2 버스를 비활성화하고 나침반을 I2C1 버스에 연결하면 기압계도 I2C1 버스에 있습니다.
> 
> 유형 CLI 명령 리소스 I2C_SCL 2 NONE 리소스 I2C_SDA 2 NONE 설정 mag_i2c_device = 1
> 
> 잠시 후 집에 와서 나침반 연결을 변경하고 i2c 포트를 설정하여 사용합니다.

jsungho, 2022-06-17:
I changed it to i2c1 and it worked fine.

haslinghuis, 2022-06-18:
@MATEKSYS do we need a unified target config update here?

MATEKSYS, 2022-06-18:
> @MATEKSYS do we need a unified target config update here?

MatekF405SE unified target has been there for 2 years without issues.
I don't think we need to update target.