betaflight/betaflight #11635: TMotorF7V2 Needs Dshot_Burst Enabled For Motor 1 To Work

ItsBlunty, 2022-06-05:
### Describe the bug

(This is a secondhand report after helping another user and solving a missing motor issue. They don't want to submit the bug so I am submitting instead. For this reason I only have the dump, not the diff, so please excuse the bug report length.)

Motor 1 does not work at all with the default settings after flash and applying custom defaults. It does not spin up in the motor test and it doesn't not show up in BLHeliSuite32, but the other three motors do show up.

If you set dshot_burst = on, then all 4 motors will work and connect with BLHeliSuite32 properly.

This error occurs on 4.2.11 and 4.3 RC7, but enabling dshot_burst was only confirmed as a fix on 4.2.11.

### To Reproduce

Flash clean version of 4.2.11 or 4,3 RC7 and apply custom defaults. Try to use all 4 motors. Motor 1 will not spin or connect to BLHeliSuite32.

### Expected behavior

Expected all 4 motors to work without changing CLI settings.

### Flight controller configuration

```
# 

# DUMP

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.11 Nov  9 2021 / 20:29:32 (948ba6339) MSP API: 1.43
# config: manufacturer_id: TMTR, board_name: TMOTORF7V2, version: 637194cf, date: 2021-04-06T04:23:26Z

# start the command batch
batch start

board_name TMOTORF7V2
manufacturer_id TMTR

# name: -

# resources
resource BEEPER 1 C13
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 B04
resource MOTOR 4 B05
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 C08
resource MOTOR 8 C09
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
[... 1066 lines cut ...]
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

# 
```
```
```

### Flight controller

T-Motor F7 Pro - V2

### Other components

Holybro 65A 32bit ESC

### Add any other context about the problem that you think might be relevant here

I am submitting this as it may be a bug or just a hardware compatibility issue, but I'm sure some users will/have run into this issue in the past.

haslinghuis, 2022-06-05:
`set dshot_burst = on` is the default for the target and was created on 2-2-2020

daleckystepan, 2022-06-05:
It seems that each motor is on different PORT so it needs 4 channels to control 4 motors. Maybe the CPU is not able to keep up. But I am just guessing.

@ItsBlunty Can you provide: `resource show all` from the latest BF 4.3?

haslinghuis, 2022-06-05:
We need a clean diff as the dump is corrupted
'''
set motor_poles = 0
###ERROR: dump: CORRUPTED CONFIG: motor_poles = 0###
'''

ItsBlunty, 2022-06-05:
As I stated as the start of the initial post, I don't have this board, this is a secondhand report. What I provided is all I can provide for this target.

SteveCEvans, 2022-06-05:
`dma show` and `timer show` dumps would be useful.

ItsBlunty, 2022-06-05:
> `dma show` and `timer show` dumps would be useful.

I included them in the initial post.

Kal-69, 2022-06-05:
Hello I am having the same issue with the T-Motor F7 Pro - V2 where motors 1 and two aren't spinning.  I tried set dshot_burst = OFF and that did not help.

Here is the diff 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Apr 25 2022 / 01:06:24 (9d71184) MSP API: 1.44
# config: manufacturer_id: TMTR, board_name: TMOTORF7V2, version: 637194cf, date: 2021-04-06T04:23:26Z

# start the command batch
batch start

board_name TMOTORF7V2
manufacturer_id TMTR

# serial
serial 0 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200

# master
set gyro_lpf1_static_hz = 0
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set motor_pwm_protocol = DSHOT600
set motor_output_reordering = 1,0,2,3,4,5,6,7
set yaw_motors_reversed = ON
set simplified_gyro_filter = OFF
set osd_vbat_pos = 2071
set vcd_video_system = NTSC

profile 0

# profile 0
set dterm_lpf1_static_hz = 150
set dterm_lpf2_static_hz = 300
set iterm_limit = 500
set pidsum_limit_yaw = 1000
set yaw_lowpass_hz = 0
set throttle_boost = 1
set p_pitch = 43
set i_pitch = 92
set d_pitch = 38
set f_pitch = 130
set p_roll = 42
set i_roll = 87
set d_roll = 37
set f_roll = 125
set i_yaw = 92
set f_yaw = 115
set feedforward_smooth_factor = 35
set feedforward_jitter_factor = 4
set feedforward_boost = 18
set simplified_pids_mode = OFF
set simplified_dterm_filter_multiplier = 200

rateprofile 2

# rateprofile 2
set rates_type = BETAFLIGHT
set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 90
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70

# end the command batch
batch end

DMA SHOW 
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 2
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

TIMER SHOW

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2 : CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

KarateBrot, 2022-06-06:
@Kal-69 Can you go to the CLI and type

```
set dshot_burst = ON
save
```

and see if that helps?

limonspb, 2022-06-06:
> @Kal-69 Can you go to the CLI and type
> 
> ```
> set dshot_burst = ON
> save
> ```
> 
> and see if that helps for BF 4.3?

ok we figured it out. He just flashed F7V2 target on on TMOTORF7 (not V2). so disregard these comments.

daleckystepan, 2022-06-06:
@Guidus93 Can you please provide us at least `diff` and `resource show all` using board `TMOTORF7V2` from BF 4.3? And also please reassure that all 4 motors are working for you without any issue.

Guidus93, 2022-06-07:
> @Guidus93 (as a maintainer of this unified-targets) Can you please provide us at least `diff` and `resource show all` using board `TMOTORF7V2` from BF 4.3? And also please reassure that all 4 motors are working for you without any issue.

@daleckystepan I have the FC sample on the way, will get it tomorrow morning and do all tests.

Guidus93, 2022-06-07:
@ItsBlunty can you please ask photo of the FC that the guy is using? TMTR create a bit of confusion with FCs naming and target, I want to double check

ItsBlunty, 2022-06-07:
> 

He is not available to provide that. As I said I have provided all information available.

Guidus93, 2022-06-07:
I got a sample and I also investigate on this FC. They told me that the Tmotor F7 V2 Target apply only to the TMTR FCs with BLE or WIFI onboard chip. So the one I have or the one with the 8 motor connector in the front [(like this one)](https://www.getfpv.com/t-motor-f7-pro-flight-controller-w-wifi-bluetooth.html) or the one i have:
[image]

I flashed BF 4.3 RC7 and i test all 4 motor output from this FC and are all fine with stock setting, as you can see from the diff i enabled only the DSHOT600 (Before I did the test also with Bidirectional and all 4 motors were spinning)

**diff:** 

```
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Jun  1 2022 / 06:23:23 (4fa2dc1) MSP API: 1.44
# config: manufacturer_id: TMTR, board_name: TMOTORF7V2, version: 637194cf, date: 2021-04-06T04:23:26Z

# start the command batch
batch start

board_name TMOTORF7V2
manufacturer_id TMTR

# master
set motor_pwm_protocol = DSHOT600

profile 0

rateprofile 0

# end the command batch
batch end

# 
```

**Resource:** 
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: MOTOR 1
B01: MOTOR 2
B02: SPI_MOSI 3
B03: CAMERA_CONTROL
B04: MOTOR 3
B05: MOTOR 4
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: PINIO 1
C02: ADC_BATT
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: BEEPER
C14: LED 1
C15: PINIO 2
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2 : CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 2
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

Here the Dshot signal: (sorry the quality but i found a cheapycheap scope where I am now)
[image]

And here BlHeli32 Suite:

[image]

Let me know guys if you need me to do more test, I asked TMTR to send also the 8 motor version of the FC but I rembember that we tested the 8 motors with 4.2.11 before release the FC and was fine

Guidus93, 2022-06-07:
> @Guidus93 Thank you for very quick debugging. So I think we can close the issue as everything seems to be working on BF 4.3. Right?

If we want to wait till Thursday I will try the same tests on the 8 motor version , that FC it’s still on the way, but pinout it’s the same so I expect the same results

daleckystepan, 2022-06-07:
@Guidus93 No problem. Let's wait to be 100 % sure.

sugaarK, 2022-06-08:
> . The user is not interested in helping at all further.

Then….. please do not waste our teams time…

ItsBlunty, 2022-06-08:
> > . The user is not interested in helping at all further.
> 
> Then….. please do not waste our teams time…

I was asked to post this with the lack of info I has by @KarateBrot

KarateBrot, 2022-06-08:
@sugaarK Yeah I encouraged Blunty to post it so that we at least have it on the record. He did what he could. Thanks Blunty!