betaflight/betaflight #11637: Crossfire Nano RX not recognized in Betaflight (v4.3 RC7)

relentlesstech73, 2022-06-06:
### Describe the bug

New Crossfire Nano RX, wired correctly to Kakute H7 Mini stack, and bound - There is no stick input in Betaflight, however when flight controller was flashed with INAV 5.0 RC2 stick inputs worked fine.

### To Reproduce

Install Crossfire Nano RX, attempt to see stick input in Betaflight

### Expected behavior

I expected to see my stick movements in Betaflight with RX bound to transmitter

### Flight controller configuration

```
 # 

# diff

# version
# Betaflight / STM32H743 (SH74) 4.3.0 Jun  1 2022 / 06:25:02 (4fa2dc1) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEH7MINI, version: f8deabe8, date: 2022-04-24T10:13:52Z

# start the command batch
batch start

board_name KAKUTEH7MINI
manufacturer_id HBRO

# name: Kakute H7 Mini

# feature
feature -RX_PARALLEL_PWM

# serial
serial 3 64 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# beeper
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1300 2100 0 0
aux 1 13 2 1300 2100 0 0
aux 2 35 3 1300 2100 0 0
aux 3 36 1 1600 2100 0 0

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = -133,76,-449,1
set mag_hardware = NONE
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set osd_units = IMPERIAL
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_cap_alarm = 2100
set osd_alt_alarm = 1100
set osd_rssi_pos = 2480
set osd_link_quality_pos = 2472
set osd_link_tx_power_pos = 2114
set osd_rssi_dbm_pos = 2103
set osd_tim_2_pos = 2092
set osd_vtx_channel_pos = 2131
set osd_craft_name_pos = 2440
set osd_altitude_pos = 2082
set osd_avg_cell_voltage_pos = 2486
set osd_disarmed_pos = 2346
set osd_core_temp_pos = 2466
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_min_rssi_dbm = ON
set name = Kakute H7 Mini

profile 0

rateprofile 0

# end the command batch
batch end

# 
```
```
  resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FLASH_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: MOTOR 3
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: MOTOR 4
B11: PINIO 1
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: LED 1
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: BEEPER
C14: FREE
C15: FREE
D00: SERIAL_RX 4
D01: SERIAL_TX 4
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: CAMERA_CONTROL
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : CAMERA_CONTROL
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: DSHOT_BITBANG 2
DMA1 Stream 1: SPI_MISO 1
DMA1 Stream 2: SPI_MOSI 2
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 4
DMA1 Stream 5: SPI_MISO 4
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Flight controller

Holybro Kakute H7 Mini (Purchased from RaceDayQuads)

### Other components

Crossfire Nano RX (Firmware v1.16)

haslinghuis, 2022-06-07:
From: http://www.holybro.com/product/kakute-h7-mini/

Not sure if it makes a difference as Holybro suggests using UART 6 for TBS Crossfire.

[image]

Just tested Crossfire and works fine with tbs firmware 6.17 on latest betaflight version and configurator.

If using TAER this command in cli might solve it:
```
map TAER1234
save
```

relentlesstech73, 2022-06-07:
Firmware has been at v6.17, and I upgraded a few times just to check

closed 2022-07-15 by github-actions[bot]