betaflight/betaflight #11638: betaflight lua not working

Dteksolutions, 2022-06-07:
### Describe the bug

on all of the 4.3 rc's ive not been able to use the betaflight lua. it is stuck on initializing. As soon as i revert to to 4.2.11 it works excellently.  FC is iflight succex-e f722. esc is the 45amp iflight flashed to blujay. rx elrs 2,4 betafpv nano - Though ive tried with 915 and other 2.4 same with the modules betafpv 2.4 micro 1w, betafpv 900 micro frsky2019 with elrs. radio tx16s edge vers 2.6.0. hopefully its some setting that ive overlooked. it also does not work on skystars f722 mini. It doesnt have a usb port any longer so i cant provide a diff.

### To Reproduce

use any of the 4.3 rc and try to use betaflight lua script.

### Expected behavior

the script to work. and be able to interact with betaflight via tx

### Flight controller configuration

```
  # # 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Jun  1 2022 / 06:23:23 (4fa2dc1) MSP API: 1.44
# config: manufacturer_id: IFRC, board_name: IFLIGHT_SUCCEX_E_F7, version: c2b8c0d3, date: 2021-06-18T03:42:42Z

# start the command batch
batch start

board_name IFLIGHT_SUCCEX_E_F7
manufacturer_id IFRC

# name: Nazgul5 V2 FRME

# feature
feature GPS
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR

# serial
serial 3 2 115200 115200 0 115200
serial 4 1024 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# led
led 0 7,12::L:0

# aux
aux 0 0 0 900 1300 1 0
aux 1 1 2 900 1300 0 0
aux 2 2 2 1300 1700 0 0
aux 3 27 4 900 1300 0 0
aux 4 46 4 1300 1700 0 0
aux 5 13 1 1300 2100 0 0
aux 6 19 5 1275 1675 0 0
aux 7 26 0 900 1300 0 0
aux 8 35 1 1700 2100 0 0
aux 9 39 5 900 1300 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BAND_A   A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BAND_B   B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BAND_E   E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 AIRWAVE  F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWRACE  L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 4
vtxtable powervalues 14 26 29 32
vtxtable powerlabels 25 400 800 MAX

# rxfail
rxfail 8 s 1500

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 0
set dyn_notch_count = 2
set dyn_notch_min_hz = 125
set dyn_notch_max_hz = 650
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_calibration = 23,-57,-7,1
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_auto_factor = 175
set rc_smoothing_auto_factor_throttle = 100
set rc_smoothing_setpoint_cutoff = 12
set rc_smoothing_feedforward_cutoff = 12
set rc_smoothing_throttle_cutoff = 20
set fpv_mix_degrees = 21
set serialrx_provider = CRSF
set blackbox_sample_rate = 1/1
set dshot_bidir = ON
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set vbat_min_cell_voltage = 340
set vbat_display_lpf_period = 10
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_angle = 37
set gps_rescue_throttle_max = 1679
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set simplified_gyro_filter = OFF
set osd_warn_batt_not_full = OFF
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_rssi_alarm = 98
set osd_cap_alarm = 494
set osd_alt_alarm = 0
set osd_vbat_pos = 14421
set osd_rssi_pos = 2128
set osd_link_quality_pos = 14411
set osd_link_tx_power_pos = 14625
set osd_rssi_dbm_pos = 14405
set osd_tim_1_pos = 246
set osd_flymode_pos = 6561
set osd_vtx_channel_pos = 6305
set osd_crosshairs_pos = 14606
set osd_current_pos = 14550
set osd_mah_drawn_pos = 14520
set osd_gps_speed_pos = 6465
set osd_gps_lon_pos = 14801
set osd_gps_lat_pos = 14769
set osd_gps_sats_pos = 6178
set osd_home_dir_pos = 6286
set osd_home_dist_pos = 6497
set osd_flight_dist_pos = 6529
set osd_compass_bar_pos = 6634
set osd_warnings_pos = 14378
set osd_avg_cell_voltage_pos = 14453
set osd_disarmed_pos = 6593
set osd_rtc_date_time_pos = 2443
set osd_efficiency_pos = 212
set osd_stat_min_link_quality = ON
set osd_stat_total_flights = ON
set osd_stat_min_rssi_dbm = ON
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 4
set vtx_freq = 5865
set vcd_video_system = PAL
set gyro_1_align_yaw = 1800
set rpm_filter_harmonics = 2
set name = Nazgul5 V2 FRME

profile 1

# profile 1
set dterm_lpf1_dyn_min_hz = 90
set dterm_lpf1_dyn_max_hz = 180
set dterm_lpf1_static_hz = 90
set dterm_lpf2_static_hz = 180
set anti_gravity_gain = 8000
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set yaw_lowpass_hz = 0
set p_pitch = 69
set i_pitch = 124
set d_pitch = 45
set f_pitch = 219
set p_roll = 66
set i_roll = 118
set f_roll = 210
set p_yaw = 66
set i_yaw = 118
set f_yaw = 210
set d_min_roll = 40
set d_min_pitch = 45
set thrust_linear = 20
set feedforward_smooth_factor = 35
set feedforward_jitter_factor = 12
set dyn_idle_min_rpm = 35
set simplified_master_multiplier = 135
set simplified_pi_gain = 110
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 130
set simplified_dterm_filter_multiplier = 120

rateprofile 0

# rateprofile 0
set roll_rc_rate = 15
set pitch_rc_rate = 15
set yaw_rc_rate = 20
set roll_expo = 76
set pitch_expo = 76
set yaw_expo = 64
set roll_srate = 91
set pitch_srate = 91
set yaw_srate = 80

# end the command batch
batch end

# 
```
```
  # # 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: LED_STRIP
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: GYRO_EXTI
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: GYRO_CS 1
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: PINIO 3
B09: FLASH_CS
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: FREE
C04: LED 1
C05: FREE
C06: MOTOR 2
C07: MOTOR 4
C08: MOTOR 1
C09: MOTOR 3
C10: SERIAL_TX 4
C11: SERIAL_RX 4
C12: FREE
C13: PINIO 1
C14: PINIO 2
C15: BEEPER
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2 : LED_STRIP
TIM3: FREE
TIM4: FREE
TIM5:
    CH1 : CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: LED_STRIP
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Flight controller

iflight succex-e f722     RDQ

### Other components

elrs betafpv nano 2.4 latest frmwre, rush tank solo, 

### How are the different components wired up

plugged in via jst

daleckystepan, 2022-06-07:
Please use the latest LUA nightly with 4.3.

https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases

Dteksolutions, 2022-06-07:
It does not work.

AQJay01, 2022-06-08:
With 4.3.0 builds, only Nighly builds of Lua Scripts would work, these new lua scripts even support slider tuning, its awesome..

Here you go;

https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases/tag/v20220602.6544

closed 2022-07-16 by github-actions[bot]