betaflight/betaflight #11642: github is my issue

DroneFliesTheWorld, 2022-06-08:
### Describe the bug

any time i click anything betflight related all i get is this text nonsense that does aboslutely nothing.  I seriously hate when someone thinks they can take over for a software and make it only usable on one operating system....

[image]

### To Reproduce

i can only reproduce angry text as myself and many friends have the same issue.

### Expected behavior

gee i dont know... Make It so I can actually use betaflight and fix my drones that are in DFU mode...

This GITHUB thing is absolute garbage.

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```
I HAVE NO OUTPUT WITH YOUR STUPID SITE

### Flight controller

flysky

KarateBrot, 2022-06-08:
What do you mean? You can use the BF configurator on Linux, macOS and Windows, so what are you blabbering about?

sugaarK, 2022-06-08:
Who the hell is using flysky in 2022 😭

howels, 2022-06-08:
Assuming that you are not a troll, please download https://github.com/betaflight/betaflight-configurator/releases and use it to configure your quad. You can use the CLI tab to generate the output which is mentioned in this form.

michey, 2022-06-12:
Ok. Probably I can translate from normies to geeks language.
> I just need a big green "Download" button which leads me directly to configurator executable.

KarateBrot, 2022-06-12:
[image]

closed 2022-06-12 by KarateBrot