betaflight/betaflight #11646: The quad goes crazy even if the receiver is getting the correct values

SGr33n, 2022-06-11:
### Describe the bug

CRSF + TAER the quad goes crazy even if the receiver is getting the correct values.

### To Reproduce

Update to BF 4.3 rc *
Configure CRSF.

### Expected behavior

The quad is not moving.

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

CLRacing F7 v3 AIO 35A

### Other components

CRSF Nano Receiver

aowi7280, 2022-06-13:
Did you use the latest nightly configurator? RC3 and RC4 had some bugs. RC6 has been really stable for me. Erase all when you flash and accept the suggested configuration.

SGr33n, 2022-06-13:
Hi,
This happened on the latest two or maybe 3 firmware versions.
The configurator is 10.08.0-debug-04b4c74, will try downloading the latest version.
Thanks!

haslinghuis, 2022-06-13:
Fixed in https://github.com/betaflight/betaflight-configurator/pull/2842

closed 2022-06-13 by haslinghuis