betaflight/betaflight #11649: HDzero issues....

sugaarK, 2022-06-13:
### Describe the bug

when first plug in, the quad will boot up but then it will boot up second time (as you can see from the video). If it only boots up once, then it won't arm. I will then have to replug and wait until it boots up twice. After it boots up twice, I can then arm but as soon as i disarm, the motor still spins for a bit and makes that tone (see video). After all this, then it acts normally (can arm and disarm as per normal)

also 

### To Reproduce

upgrade your HDzero hardware to 11062022 
i tried 22032022 (one rev before) and seemed to have the issue as well. My baseline version one that I have been using with no issue before was like 18122021 (from Dec 2021)

diff  from bf 

# Betaflight / STM32H743 (SH74) 4.3.0 Apr  9 2022 / 14:35:19 (norevision) MSP API: 1.44
# config: manufacturer_id: JHEF, board_name: JHEH743_AIO, version: d33b629a, date: 2022-03-09T17:23:56Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name JHEH743_AIO
manufacturer_id JHEF
mcu_id 004600243330510f36363234
signature 

# name: CGO-UL2

# feature
feature -RX_PARALLEL_PWM
feature TELEMETRY
feature LED_STRIP

# serial
serial 1 64 115200 57600 0 115200
serial 5 1 115200 57600 0 115200

# beacon
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1800 2100 0 0
aux 1 13 2 1800 2100 0 0
aux 2 15 2 900 1200 0 0
aux 3 26 2 1300 1700 0 0
aux 4 35 1 1800 2100 0 0
aux 5 49 2 1300 1700 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable powerlevels 3

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 300
set dyn_notch_count = 1
set dyn_notch_q = 450
set dyn_notch_min_hz = 125
set dyn_notch_max_hz = 650
set gyro_lpf1_dyn_min_hz = 500
set gyro_lpf1_dyn_max_hz = 1000
set acc_calibration = -46,-17,27,1
set baro_hardware = NONE
set rc_smoothing_auto_factor = 25
set rc_smoothing_auto_factor_throttle = 25
[... 487 lines cut ...]
set yaw_srate = 70

rateprofile 5

# rateprofile 5
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70

# restore original rateprofile selection
rateprofile 1

# save configuration
save
# 

https://user-images.githubusercontent.com/83344205/173345362-4a93de71-576d-46a5-8475-5e926af2fe60.mp4

### Flight controller

JHEH743_AIO

sugaarK, 2022-06-13:
This is a slightly different issue to the dfu issue

haslinghuis, 2022-06-13:
If build `18122021` was working it's definite a HDZero firmware issue

matyas-brenner, 2022-06-15:
This is a bad name for an issue IMHO. I suggest something like "FC won't arm with HDZero VTX".

matyas-brenner, 2022-06-15:
I'm using HDZero Freestyle VTX with firmware rev. 22032022 and don't have this issue.
I'm using Mamba H743 FC. It probably won't be as simple as having the VTX on a specific FW version.

sugaarK, 2022-06-15:
> This is a bad name for an issue IMHO. I suggest something like "FC won't arm with HDZero VTX".

Why it arms but it’s not behaving as it should.. it’s likey flight controller sensitive. Just because you don’t have the issue doesn’t mean it doesn’t exist

bootdsc, 2022-07-23:
You can solve this by adding a resistor inline with the RX and another with the TX, 220ohm should do the trick. 

To validate before doing the mod just disconnect the VTX TX and RX then boot up and it should be normal otherwise you have a different cause.

As to why this fixes it well it's more to do with poor design of the FC and not the VTX.

sugaarK, 2022-07-23:
The dfu mode issue happens on all fcs I have.. tmotor and foxeer f7s. The foxeers are not poorly designed frankly

drobot, 2023-04-03:
Hey SugaarK I just started having this issue after I switched to HDZero using an Foxeer H7 board.  I have tried multiple firmwares of HDZ and none of them fixed the problem. I am currently on the HDZ Beta and still having the arming issue.  Has this been resolved?

NOTE: I found a may to mitigate this problem by simply turning off and on the MSP Display port, and the UART HDZ is tied too.  While this is a quick fix, by changing any pid or filter related settings, it starts happening again, which is annoying while trying to tune.

sugaarK, 2023-04-03:
I think i heard it was a missing resistor on the VTX uart not sure how we fix it on our side. You can try a higher uart as it seems to be worse on the lower uarts

drobot, 2023-04-03:
Currently using UART 7 which is the highest on FoxH7, but I will try another and see what happens

sugaarK, 2023-04-03:
Are you on the hdzero discord?

drobot, 2023-04-03:
Yes
Drobot-Racer#9798

sugaarK, 2023-04-03:
Wana ask over there as we have no idea how to fix it on our end

haslinghuis, 2023-10-23:
@sugaarK as I cannot confirm myself without having HDZero hw - I think this could be closed?

Prajwal-Koirala, 2023-12-26:
The issue reported in this Github thread is that a flight controller (FC) won't arm with an HDZero VTX. The FC boots up twice before arming, and the motors spin for a bit after disarming. This issue seems to be related to the HDZero firmware version, as upgrading to 11062022 fixed the problem for one user.

Here's a more detailed breakdown of the discussion:

* **Symptoms:**
    * Flight controller doesn't arm with HDZero VTX.
    * FC boots up twice before arming.
    * Motors spin for a bit after disarming.
* **Possible causes:**
    * HDZero firmware version: Upgrading to 11062022 might fix the issue.
    * UART configuration: Using a higher UART might help.
    * Hardware issue: Adding a resistor inline with the RX and TX could be a solution.
* **Current status:**
    * The issue is still being discussed on Github.
    * No official solution has been found yet.

If you're experiencing this issue, you can try upgrading the HDZero firmware to 11062022 or using a higher UART. You can also add a resistor inline with the RX and TX, but this should be done as a last resort, as it requires soldering.

I hope this helps! Let me know if you have any other questions.

sugaarK, 2023-12-27:
@Prajwal-Koirala  thanks for your summery but I think this is fixed now

haslinghuis, 2024-05-12:
Fixed according to https://github.com/betaflight/betaflight/issues/11649#issuecomment-1869957783

closed 2024-05-12 by haslinghuis