betaflight/betaflight #11654: SDcard cannot be recognized by F4 in SDIO mode and mass storage mode cannot be activated

tianqiraf, 2022-06-14:
### Describe the bug

I am using the `STM32F407ZG` development board as my flight controller. The development board connectes sdcard with `SDIO`. I found that another board `WORMFC(F405RG)` is also using SDIO mode, and the GPIO used is the same.
# First Problem
1. *The problem is*: WORMFC firmware compiled with **4.2.9** and earlier versions can be burned into my F407ZG, and sdcard is recognized(`SD card: Manufacturer 0x3, 1931264kB, 06/2002, v8.0, 'SU02G' Filesystem: Ready`); but WORMFC firmware compiled with **4.2.10** and later versions can not recognize my sdcard 
2. *My solution*: I guessed the problem is the `dmaAllocate` used by the `SD_Initialize_LL` function in `drivers/sdio_f4xx.c` in version 4.2.10, so I replaced it with the `SD_Initialize_LL` function in version 4.2.9, and add the definition of `dmaInit` function in `drivers/dma_stm32f4xx.c` and `drivers/dma.c`. 
Then I compile the WORMFC firmware, and the sdcard can be recognized!
what's more, I made the same change in **4.3.0** and it still works well.
The following is the modified function of `SD_Initialize_LL` in 4.2.10 (or 4.3.0):
```c
bool SD_Initialize_LL(DMA_Stream_TypeDef *dma)
{
    //const dmaIdentifier_e dmaIdentifier = dmaGetIdentifier((dmaResource_t *)dmaStream);
    /*if (!(dma == DMA2_Stream3 || dma == DMA2_Stream6) || !dmaAllocate(dmaIdentifier, OWNER_SDCARD, 0)) {
        return false;
    }
	*/
    // Reset SDIO Module
    RCC->APB2RSTR |=  RCC_APB2RSTR_SDIORST;
    delay(1);
    RCC->APB2RSTR &= ~RCC_APB2RSTR_SDIORST;
    delay(1);
	......
	......
	......
    dmaStream = dma;
    RCC->AHB1ENR |= RCC_AHB1ENR_DMA2EN;
	if ((uint32_t)dmaStream == (uint32_t)DMA2_Stream3) {
        // Initialize DMA2 channel 3
        DMA2_Stream3->CR   = 0;                                                 // Reset DMA Stream control register
        DMA2_Stream3->PAR  = (uint32_t)&SDIO->FIFO;
        DMA2->LIFCR        = IFCR_CLEAR_MASK_STREAM3;                           // Clear all interrupt flags
        DMA2_Stream3->CR   = (DMA_CHANNEL_4        | DMA_SxCR_PFCTRL        |   // Prepare the DMA Stream configuration
                              DMA_MINC_ENABLE      | DMA_PDATAALIGN_WORD    |   // And write to DMA Stream CR register
                              DMA_MDATAALIGN_WORD  | DMA_PRIORITY_VERY_HIGH |
                              DMA_MBURST_INC4      | DMA_PBURST_INC4        |
                              DMA_MEMORY_TO_PERIPH);
        DMA2_Stream3->FCR  = (DMA_SxFCR_DMDIS | DMA_SxFCR_FTH);                 // Configuration FIFO control register
        dmaInit(dmaGetIdentifier((dmaResource_t *)DMA2_Stream3), OWNER_SDCARD, 0);
        dmaSetHandler(dmaGetIdentifier((dmaResource_t *)DMA2_Stream3), SDIO_DMA_ST3_IRQHandler, 1, 0);
    } else {
        // Initialize DMA2 channel 6
        DMA2_Stream6->CR   = 0;                                                 // Reset DMA Stream control register
        DMA2_Stream6->PAR  = (uint32_t)&SDIO->FIFO;
        DMA2->HIFCR        = IFCR_CLEAR_MASK_STREAM6;                           // Clear all interrupt flags
        DMA2_Stream6->CR   = (DMA_CHANNEL_4        | DMA_SxCR_PFCTRL        |   // Prepare the DMA Stream configuration
                              DMA_MINC_ENABLE      | DMA_PDATAALIGN_WORD    |   // And write to DMA Stream CR register
                              DMA_MDATAALIGN_WORD  | DMA_PRIORITY_VERY_HIGH |
                              DMA_MBURST_INC4      | DMA_PBURST_INC4        |
                              DMA_MEMORY_TO_PERIPH);
        DMA2_Stream6->FCR  = (DMA_SxFCR_DMDIS | DMA_SxFCR_FTH);                 // Configuration FIFO control register
        dmaInit(dmaGetIdentifier((dmaResource_t *)DMA2_Stream6), OWNER_SDCARD, 0);
        dmaSetHandler(dmaGetIdentifier((dmaResource_t *)DMA2_Stream6), SDIO_DMA_ST6_IRQHandler, 1, 0);
    }

    return true;
}
```

# Second Problem
1. *The problem is*: After solve the problem 1, I found that the MSC cannot be used. There is only an empty disk symbol in the explorer but it cannot be opened, and it will prompt "Please insert a removable disk *" when clicked. I tried version **4.2.10** (with the changes in problem 1) to compile the WORMFC firmware, and it can recognize sdcard and reboot into MSC, but in **4.3.0**, it can only recognize sdcard and **cannot reboot into MSC**.
2. *My solution*: A piece of code is commented out at about 600 lines in `fc/init.c` in 4.3.0, and then it works well!
```c
#ifdef TARGET_BUS_INIT
    targetBusInit();

#else

    // Depending on compilation options SPI/QSPI initialisation may already be done.
    if (!(initFlags & SPI_AND_QSPI_INIT_ATTEMPTED)) {
[... 312 lines cut ...]
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SDCARD
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

WORMFC and STM32F4DISCOVERY

VMaximI, 2022-06-14:
I confirm the problem with MSC. When switching the Black Box to removable disk mode, the connection with the flight controller is lost, the removable disk does not appear in the explorer. This issue is not present in version 4.2.11 and earlier. Target MatekF405SE.

mituritsyn, 2022-06-29:
@tianqiraf i think it could goes faster if you'll open pull request with those changes

tianqiraf, 2022-07-01:
@mituritsyn I didn't delve into what the code I deleted did and what would happen if it was deleted. This issue is just to locate the problematic code and the idea of debug.

closed 2022-08-09 by github-actions[bot]